libremesh/lime-packages #162: Include model in hostname

ilario, 2017-06-17:
Hostname can include router model for making easier the understanding of network representation in web ui.

ilario, 2017-06-19:
Adding the router model in the hostname would make the identification of devices easier, for example in this BMX6 graph where I added manually the models.

[image]

FreifunkUFO, 2017-07-03:
so these hostnames contains spaces and will be very long?

ilario, 2017-07-04:
It was mainly a concept, I can rephrase the issue title in "Include more node characteristics in default hostname".
The sanitized and truncated router model name is an option.

ilario, 2017-09-22:
How can one get the router model from inside LEDE?
I mean something like `tl-wdr3600`.

ilario, 2017-09-22:
This info can be obtained from `/tmp/sysinfo/model`.

dangowrt, 2017-09-22:
speaking in shell, the ubus API allows you to

```
jsonfilter -s "$(ubus call system board)" -e "@.board_name"
```

obviously you can use other languages to call ubus and process the JSON output.

ilario, 2017-09-22:
Nice!!
I have no idea why but on my router (Sagem F@ST2704N) there's info in `/tmp/sysinfo/board_name` but there's no board_name field in `ubus call system board` (there's a model field, but it's not as much compact).
Do you think it's incorrect to stick to 
Anyway, what's missing is just implement this in `utils.lua` as a function `applyBoardTemplate`, use it in `system.lua` and document it in `lime-example`

ilario, 2018-12-08:
I perceive a non-interest on this topic, that's ok, closing.

closed 2018-12-08 by ilario