libremesh/lime-packages #164: unify releases at repo.libremesh.org

aparcar, 2017-06-21:
The 16.07 releases is missing a `targets` seperation
http://repo.libremesh.org/lime-16.07/
http://repo.libremesh.org/lime-17.04/targets/

Is that on purpose?

ilario, 2017-06-21:
The whole path and organization is different... I have no idea why.
For example, for the same router and the two versions the paths are:
`http://repo.libremesh.org/lime-16.07/ar71xx/generic/openwrt-15.05.1-ar71xx-generic-tl-wdr3600-v1-squashfs-sysupgrade.bin`
and
`http://repo.libremesh.org/lime-17.04/targets/ar71xx/generic/tl-wdr3600-v1/lime_default/lede-17.01.2-libremesh-ar71xx-generic-tl-wdr3600-v1-squashfs-sysupgrade.bin`

dangowrt, 2017-06-21:
I reckon this is due to `cooker` being introduced and several flavors for each device now being generated... Also note that 16.07 was still based on OpenWrt 15.05 which didn't have per-architecture packages, it used to build all packages for each target/subtarget rather than only the target-specific packages.

p4u, 2017-06-29:
> I reckon this is due to cooker being introduced and several flavors for each device now being generated... Also note that 16.07 was still based on OpenWrt 15.05 which didn't have per-architecture [...]

Right.

The community_chaos (lime-16.07) binaries were manually copied to repo.libremesh.org (which is the new server using  _cooker_) for legacy purposes only. The new directory structure is based on the LEDE directory structure [1].

[1] http://downloads.lede-project.org/releases/17.01.2/

aparcar, 2017-06-29:
all right, thank you both for the clarification! I think this is solved

closed 2017-06-29 by aparcar