libremesh/lime-packages #166: Avoid copy of /etc/config on /etc/lime-config-firstboot

p4u, 2017-06-29:
On first boot the directory /etc/config is fully copied to /etc/lime-config-firstboot

Is it really needed? I'm not able to find a use case.

Related commit: https://github.com/libremesh/lime-packages/commit/de31181ed4e4a0c9c2249848cf3a5418295dcc70

dangowrt, 2017-06-29:
On Thu, Jun 29, 2017 at 01:21:46AM -0700, Pau wrote:
> On first boot the directory /etc/config is fully copied to /etc/lime-config-firstboot
> 
> Is it really needed? I'm not able to find a use case.
> 
> Related commit: https://github.com/libremesh/lime-packages/commit/de31181ed4e4a0c9c2249848cf3a5418295dcc70

I believe this is needed for lime-config to be able to re-run in case
/etc/config/lime or /etc/config/lime-default has been changed.
Esp. /etc/config/system, /etc/config/network and /etc/config/wireless
have already been re-written by lime-config.

This is the exact reason why I believe that using /etc/board.json in a
way similar to config_generate should be used to generate those files
instead of re-writing them (which obviously cannot work another time
if the files have already been rewritten, hence the need for a copy)

p4u, 2017-06-29:
> I believe this is needed for lime-config to be able to re-run in case
> /etc/config/lime or /etc/config/lime-default has been changed.
> Esp. /etc/config/system, /etc/config/network and /etc/config/wireless
> have already been re-written by lime-config.

It seems to be only used this way by lime-hwd-openwrt-wan. However it is not needed to save all the UCI config files, maybe just _network_ is enough. Also this should be handled by lime-hwd-openwrt-wan package instead of lime-system.

> This is the exact reason why I believe that using /etc/board.json in a
> way similar to config_generate should be used to generate those files
> instead of re-writing them (which obviously cannot work another time
> if the files have already been rewritten, hence the need for a copy)

Agree. Related issue: https://github.com/libremesh/lime-packages/issues/135

ilario, 2017-08-01:
Pau, can you fix this the way you commented above?

aparcar, 2018-01-24:
So fixing means only copying the *network* file for now?

dangowrt, 2018-01-24:
Imho this is unneeded as config_generate can be used at any time to
re-generate the default configuration of the device.
lime-config should use the data in /etc/board.json just like
config_generate does instead of re-writing the generated config.

commit cbaa9d90 (2018-01-30): fix #166
closed 2018-01-30 by G10h4ck