libremesh/lime-packages #169: Replace ebtables for something else more efficient

p4u, 2017-06-30:
Ebtables is used by anyGW to avoid the propagation of packets sent to anyMAC, but it is very expensive for the performance since all packets which are passing throgugh the bridge must  be inspected for matching (or not) the ebtables rules. A better approach might be used.

@dangowrt wrote https://github.com/libremesh/lime-packages/pull/153
> The Linux bridge code supports VLAN filtering without having to use (the rather costly) ebtables, however, BRIDGE_VLAN_FILTERING isn't enabled in OpenWrt/LEDE (probably for space reasons).

Also TC (linux traffic control) or iptables with some layer2 module might be used ¿?

FreifunkUFO, 2017-07-03:
even freifunk-gluon-firmware (which loves to replace common tools, packages and commands) is using ebtables

dangowrt, 2017-07-04:
`ebtables` (no matter which netlink tools you use to set it up) eats quite a lot of performance. We simply shouldn't use it if we want throughput.
Gluon uses ebtables, however, it uses it to filter broadcast/multicast traffic from the vAP to leak into the mesh -- hence each node only has to deal with it's own vAP's traffic ebtables overhead, there is no ebtables in the forwarding plance.

closed 2019-03-14 by gmarcos87

ilario, 2019-03-15:
@gmarcos87 is this issue solved? [ebtables looks still used in lime-packages](https://github.com/libremesh/lime-packages/search?q=ebtables).

gmarcos87, 2019-03-15:
No one took this torch for two years, I think this enhancement is not valued as necessary. If someone wants to reopen this feature they can do it via a pull-request. Having it open doesn't contribute to its development by itself. No one even performed metrics to analyze the cost-benefit of the effort required.

ilario, 2023-02-01:
With OpenWrt 22.03, both ebtables and iptables have been replaced by nftables.