libremesh/lime-packages #170: firmware upgrade via webui simple mode: no keep-no-settings

FreifunkUFO, 2017-07-03:
trying firmware upgrade via "simple mode" gives me no chance to say "keep no settings".
its mentioned on the webui, but there is no checkbox!

unchecking the "keep settings" checkbox is very important, because libremesh is an autoconfig mesh firmware.. (so keeping settings is not needed for most of the devices. To uncheck that option is mandatory for testing)

[image]

ilario, 2017-07-09:
This happens because `packages/lime-webui/luasrc/view/flashops.htm` introduced in commit 430deae lacks the option, no idea why. @nicopace can you fix this?

nicopace, 2017-07-09:
Shouldn't be keeping it. don't have a device to test next to me. can you test @FreifunkUFO ?

p4u, 2017-07-10:
The libremesh web upgrade uses the command line tool "lime-sysupgrade"
which is a wrapper of standard sysupgrade. It saves only few
configuration files such as /etc/config/lime, /etc/passwd /etc/dropbear
and so on. But none of the standard UCI /etc/config files is saved so it
would not be a problem.

So IMO this behavior is correct. If you don't want to save any file (not
even /etc/config/lime) just use standard "sysupgrade -n" from command line.

ilario, 2017-07-10:
Ok, so it's just a matter of removing the ` Check "Keep settings" to retain the current configuration (requires a compatible firmware image).` text from [here](/libremesh/lime-packages/blob/develop/packages/lime-webui/luasrc/view/flashops.htm#L18).

aparcar, 2017-07-11:
hopefully fixed in #180

commit facc9b8e (2017-07-14): Merge pull request #180 from aparcar/patch-1

fix #170
closed 2017-07-14 by ilario