libremesh/lime-packages #171: webui simple mode: "switch" tab appears, even when there is no switch

FreifunkUFO, 2017-07-03:
even when there is no switch
the "switch" tab appears atm
(webinterface, on "status" mode, without any login)

FreifunkUFO, 2017-07-03:
with "administration" login the behaviour seems to be normal:
on "network" tab there is no "switch" link to click..

p.s. i tried that with tplink-710-v.2.1 (a.k.a. tplink-710-v1) where wan is eth0 and lan is eth1

p.p.s. its a very small bug, but please even take care of it!

FreifunkUFO, 2019-03-29:
ping

nicopace, 2019-05-29:
hehe... pinging doesn't make any progress on the issue... the stalebot has a bug :)
many of the devs (me included) are putting the ui efforts in lime-app, and phasing out the use of luci.
PR welcome, but no maintenance will happen, at least it is extremely low priority

ilario, 2019-10-04:
I copy @nicopace: help on this issue is welcome but as by default we'll not include LuCI in the 19.something release, this issue has low priority.
Closing.

closed 2019-10-04 by ilario