libremesh/lime-packages #172: dhcp-client: do NOT get IPs from other LibreMesh nodes DHCP-servers

FreifunkUFO, 2017-07-03:
with very old libremesh firmware the behaviour was correct:
when doing WAN-LAN-ccoupling within libremesh-firmware routers 
the WAN side DIDNT  got any IPs from another-routers LAN !!!

so, there might be an ebtables-confi(?) to make sure not getting an IP when connected (via cat6/cat7 cable) to another libremesh node..
maybe the next node MAC alias could be used for that=?

p.s. i even dont know how that very old firmware does it that correctly, but it was working very well!?

ilario, 2017-07-10:
Related discussion going on in lime-users ML in [this](https://lists.libremesh.org/pipermail/lime-users/2017-July/000751.html) thread.

p4u, 2017-08-01:
There is not such a ebtables rule. However if more than one DHCP servers are listening at the same Ethernet domain, the first replying will be the one used by the client.

ilario, 2017-08-02:
@p4u have you read the email I linked above? Do you think is another issue or not an issue? Why the "invalid" label?

p4u, 2017-08-03:
Yes I read it but still I think this is not an issue. In a physical ethernet domain there is no way, from the libremesh router, to avoid the announcements of other devices connected to the same domain (switch). So, if more than one DHCP servers are runing, the first replying will be the one used by the client. There is nothing else we can do.

nikksno, 2017-08-21:
Thanks @ilario for following up on this. I don't know if my issue is directly related, but it is the "first LiMe node" that lets DHCP "broadcasts" [terminology might be incorrect] travel from inside the LAN-to-LAN LiMe "hyperlan" outside through the wan port and into the ISP router. The only "requirements" for such error are only two in my tests:
1] multiple [two are sufficient] LiMe nodes connected LAN-to-LAN
2] a "vulnerable" ISP router to which one of the LiMe nodes is connected
[by vulnerable i mean an ISP router that lets DHCP "broadcasts" from its LAN port influence its decisions on how to connect to the ISP "backbone" on its WAN side]
I'm running stock LiMe and still have this issue.
Thank you

ilario, 2018-12-08:
@FreifunkUFO I supposed that the node connected via WAN port on another node on LAN port would get an IP via DHCP (if both of the routers are LibreMesh, that's a broken broken broken setup), if this does not happen, I don't know why as I don't think there's any mechanism for discovering which kind of router is on the other side of the cable (see #367). There is some discussion about a mitigation for these cases in #332.

@nordurljosahvida what you write sounds like a very different issue, and for solving that I suspect we need to know which packet type is causing the mess in order to filter it.

nordurljosahvida, 2018-12-09:
@ilario ok thank you, i'll open a separate issue for this as soon as I can gather enough data to provide.

Thanks!

FreifunkUFO, 2018-12-14:
@ilario but, on older libremesh firmware there was such a mechanism for that discovery.

yes, i am talking about one cable between two libremesh-nodes (WAN-Port on Node1 to LAN-Port on Node2). For us, it is still a standard-requirement for mesh-networks, for instance when using two nanostation-m at one location (so Node1 will power Node2, throught the Poe-power-passthrough-option).
yes, it will lead to a broken libremesh. With a small mesh there is also a chance to miss that problem (!), because impact is just on a fifth or sixt libremesh-node (near to these two nodes) and will not happen when the "normal" internet-gateway is near.

"my" problem is mentioned in more than one ticket (also on https://github.com/libremesh/lime-packages/issues/331 ) but many of these are still not closed. i will do another live-test some times.

ilario, 2019-10-04:
The original question from @FreifunkUFO was about LibreMesh1 WAN to LibreMesh2 LAN cable connection. This is an absolute don't-do. Maybe more documentation is missing saying that you should not add an unnecessary NAT in the middle of a LibreMesh network? Anyway I see no easy way for a LibreMesh WAN port to recognize that the neighbour is another LibreMesh router.

The issue observed by @nordurljosahvida seems that could be avoided simply connecting the ISP router to a LibreMesh WAN port instead than to a LAN port. If you really want to connect it to a LAN port but avoid having DHCP offers on that port, maybe you'll have to add some custom firewall rule?

Closing this.

closed 2019-10-04 by ilario