libremesh/lime-packages #175: remove lime-hwd-openwrt-wan as dependency of lime-basic

aparcar, 2017-07-08:
creating a new flavor with `-lime-hwd-openwrt-wan`does not work anymore. 
I want to create an image for routers with low power not needing any gateway capabilities.

FreifunkUFO, 2017-07-09:
please try it with deactivated lime-proto-wan

see flavour "freifunk_zero_minus_opkg_wan" at
https://github.com/freifunk-leipzig/libremesh-lime-sdk/blob/master/flavors.conf

aparcar, 2017-07-09:
my current flavor looks like this and it does not remove the `-lime-hwd-openwrt-wan` packages:

    lime_mini_client="-lime-hwd-openwrt-wan -lime-proto-wan $lime_mini"

In `/usr/lib/opkg/status` it says both "removed" packages are a dependencie of `lime-basic`

```
Package: lime-basic                                                                                                                                                                                                                            
Version: 2017-06-14-1497464299                                                                                                                                                                                                                 
Depends: libc, lime-system, lime-proto-bmx6, lime-proto-batadv, lime-proto-anygw, lime-proto-wan, dnsmasq-lease-share, dnsmasq-distributed-hosts, lime-webui, lime-hwd-openwrt-wan, [...]
```

p4u, 2017-07-10:
You are right, as lime-hwd-openwrt-wan is part of the lime-basic collection it cannot be removed if you base your flavor on lime-basic. However lime-hwd-openwrt-wan does not has a big impact on the firmware size since it only adds a dependency on lime-proto-wan which do not has any special dependency (such as PPP). So we are talking just about few bytes of LUA code.

In any case, if you really want to delete lime-proto-wan from your routers you might create your own flavor NOT based on lime-basic but on the individual packages instead.

`lime_whatever="lime-system lime-proto-bmx6 lime-proto-batadv lime-proto-anygw lime-proto-wan dnsmasq-lease-share dnsmasq-distributed-hosts lime-webui lime-hwd-openwrt-wan lime-docs-minimal bmx6-auto-gw-mode smonit"`

Hope it helps.

dangowrt, 2017-07-10:
Why is it even an extra package then if not having it is unintended?
I reckon modularity is a diverse argument and not only about saving
flash space. The modular approach would be to remove the dependency
and include lime-hwd-openwrt-wan in all the flavors, which makes it
much easier to have flavors which come without it.

ilario, 2017-08-02:
PR #200 together with libremesh/lime-sdk#20 should solve this.
(please tell me if you think we should do the same for other packages, for example lime-hwd-ground-routing)

ilario, 2017-08-02:
@aparcar those PR are not merged yet :)

p4u, 2017-08-02:
I don't see the point for doing this. This package has almost no impact on the flash size and it is useful for autodetecting the WAN ports of the routers (if not enabled, everything is configured as LAN).

@aparcar is it really necessary to free these few bytes? I don't think it makes a difference.

aparcar, 2017-08-03:
@p4u I'll just deactivate the auto detecting, so I'm fine with whatever option.

ilario, 2017-08-21:
Any other opinion?
@G10h4ck @nicopace @nicoechaniz @altergui 

I think that all this thing is not very important and we could just keep things unchanged, as @p4u writes, but it would be *slightly better* to remove `lime-hwd-openwrt-wan`, `lime-proto-wan`, `lime-webui` (this would make useless the lime-basic-no-ui package), `lime-hwd-ground-routing`, `dnsmasq-distributed-hosts` and `bmx6-auto-gw-mode` from `lime-basic` package and add them to `lime-full` package and to `lime-zero` (except `lime-webui`) and `lime-mini` flavors.

ilario, 2017-09-22:
Inactivity for more than one month, is there still interest on this or shall we close it?

aparcar, 2017-09-22:
nope

closed 2017-09-22 by ilario

FreifunkUFO, 2017-11-14:
yes, im still very interested on this "feature". 
this package has huge impact on WAN-Ports, especially when the dhcp-client will get an IP-Adress (f.i. from another libremesh-router)

https://github.com/libremesh/lime-packages/issues/172

FreifunkUFO, 2017-11-14:
btw @p4u said "if you really want to delete lime-proto-wan from your routers you might create your own flavor".
but i was forbidden by him to create such flavor :-( https://github.com/libremesh/lime-sdk/commit/9f56f3e6460f0264171b3b5b7513a1b8ec6b660f#diff-d5b5948fa2c7f423643e50f7c31968c7

nicopace, 2017-11-14:
Now you can do it creating a network profile: https://github.com/libremesh/lime-sdk#communities

dangowrt, 2017-11-14:
The problem is that in my network profile I may definitely want to include `lime-basic` and that currently depends on all the wan hacks I may not want. It'd be great if network profiles (including the default profile) which actually **do** want that feature are free to include it or in some other way make sure it no longer is a hard non-removable dependency.