libremesh/lime-packages #178: Remove lime-full and lime-basic packages, just use lime-sdk flavors

ilario, 2017-07-10:
Widening the discussion started on #175:
now that we have collections in the new lime-sdk called [flavors](/libremesh/lime-sdk/blob/master/flavors.conf) why do we still need lime-basic and lime-full collections?
Wouldn't be more direct to just use one of the two ways for selecting and deselecting a list of packages?

p4u, 2017-07-10:
Having collections of packages is useful, not only to have everything
more organized (if lime-basic does not exist, the flavor must contain a
long list of packages) but also when using opkg to from a plain LEDE
router to install libremesh (opkg install lime-full).

ilario, 2017-07-11:
Ok, like in #179? Nice!
Anyway I would disentangle lime-sdk from lime-packages collections.
I mean: let's keep the collections but let's avoid their use in lime-sdk.
This prevents situations like #175

ilario, 2017-08-01:
Are there more opinions before closing this?
@altergui @G10h4ck @nicopace @nicoechaniz

nicopace, 2017-08-01:
i think it is clear to close it

p4u, 2017-08-01:
I don't see a strong point for removing the collections from lime-sdk. So IMO it is better if we use the collections. The users of lime-sdk can define their own flavors without using any collection if they have a point for doing so.

closed 2017-08-02 by ilario

ilario, 2020-01-07:
lime-basic has been removed in #674