libremesh/lime-packages #181: Unify LibreMesh release information

aparcar, 2017-07-11:
On standard LEDE I'm using `ubus call system board` and check the `release` section. On Libremesh a look at /etc/lime_release provides the information. Is it possible to modify the `ubus call system board` output to give the same information as lime_release does?

The most simple solution could be to modify the `/etc/openwrt_release` file or create a symlink from `lime_release` to `openwrt_release`

This issue is very important for the [attended sysupgrade](https://github.com/aparcar/gsoc17-attended-sysupgrade) I'm trying to implement. 

(This was send on `lime-dev` mailing list before)

nicopace, 2017-07-12:
how would you suggest to solve it?

aparcar, 2017-07-12:
> The most simple solution could be to modify the /etc/openwrt_release file or create a symlink from lime_release to openwrt_release

I could create a PR which auto overrides the `openwrt_release` file with LiMe data.

nicopace, 2017-07-13:
Please do :)

commit 26bfc42d (2017-08-13): Merge pull request #209 from aparcar/openwrt_release

luci-system: fix #181
commit 83bcf1b4 (2017-08-13): luci-system: fix #181

this PR changes the LIME_RELEASE value (in /etc/lime_release) from
0.0 to the current branch aka release number (17.06).

the data of /etc/lime_release will then replace the data of
/etc/openwrt_release, only changing concerning values.

this behaviour was requested (by me) in a issue [1] to make LiMe
 work well with my GSoC project "attended sysupgrades" [2].

root@LiMe-ac6b63:/# cat /etc/openwrt_release
DISTRIB_ID='LiMe'
DISTRIB_RELEASE='17.06'
DISTRIB_REVISION='13a91e0'
DISTRIB_CODENAME='development'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='L [... cut]
closed 2017-08-13 by p4u