libremesh/lime-packages #182: Error compailing lime-docs in fresh install lime-sdk 17.01.2

gmarcos87, 2017-07-12:
```
lime-sdk git:(master) ✗ J=4 V=s ./cooker -b x86/generic --package=lime-docs
-> Building 17.01.2/x86/generic/sdk
Updating feed 'base' from '/home/okupa/libremesh/lime-sdk/feeds/base' ...
Create index file './feeds/base.index' 
Updating feed 'packages' from '/home/okupa/libremesh/lime-sdk/feeds/packages' ...
Create index file './feeds/packages.index' 
Updating feed 'luci' from '/home/okupa/libremesh/lime-sdk/feeds/luci' ...
Create index file './feeds/luci.index' 
Updating feed 'routing' from '/home/okupa/libremesh/lime-sdk/feeds/routing' ...
Create index file './feeds/routing.index' 
Updating feed 'telephony' from '/home/okupa/libremesh/lime-sdk/feeds/telephony' ...
Create index file './feeds/telephony.index' 
Updating feed 'libremesh' from '/home/okupa/libremesh/lime-sdk/feeds/libremesh' ...
Create index file './feeds/libremesh.index' 
Updating feed 'libremap' from '/home/okupa/libremesh/lime-sdk/feeds/libremap' ...
Create index file './feeds/libremap.index' 
Updating feed 'limeui' from '/home/okupa/libremesh/lime-sdk/feeds/limeui' ...
Create index file './feeds/limeui.index' 
Installing all packages from feed libremesh.
Installing all packages from feed libremap.
Installing all packages from feed limeui.
-> Building package lime-docs on sdk 17.01.2/x86/generic/sdk
make: se entra en el directorio '/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk'
make[1]: Entering directory '/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk'
make[2]: Entering directory '/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk'
make[2]: Leaving directory '/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk'
make[2]: Entering directory '/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk'
make[2]: Leaving directory '/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk'
make[1]: Leaving directory '/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk'
tmp/.config-package.in:7:warning: ignoring type redefinition of 'PACKAGE_busybox' from 'boolean' to 'tristate'
tmp/.config-package.in:27:warning: ignoring type redefinition of 'PACKAGE_dnsmasq' from 'boolean' to 'tristate'
tmp/.config-package.in:101:warning: ignoring type redefinition of 'PACKAGE_firewall' from 'boolean' to 'tristate'
tmp/.config-package.in:122:warning: ignoring type redefinition of 'PACKAGE_jsonfilter' from 'boolean' to 'tristate'
tmp/.config-package.in:166:warning: ignoring type redefinition of 'PACKAGE_libc' from 'boolean' to 'tristate'
tmp/.config-package.in:194:warning: ignoring type redefinition of 'PACKAGE_libgcc' from 'boolean' to 'tristate'
tmp/.config-package.in:251:warning: ignoring type redefinition of 'PACKAGE_libpthread' from 'boolean' to 'tristate'
tmp/.config-package.in:279:warning: ignoring type redefinition of 'PACKAGE_librt' from 'boolean' to 'tristate'
tmp/.config-package.in:373:warning: ignoring type redefinition of 'PACKAGE_rpcd' from 'boolean' to 'tristate'
tmp/.config-package.in:436:warning: ignoring type redefinition of 'PACKAGE_ubus' from 'boolean' to 'tristate'
tmp/.config-package.in:451:warning: ignoring type redefinition of 'PACKAGE_ubusd' from 'boolean' to 'tristate'
tmp/.config-package.in:465:warning: ignoring type redefinition of 'PACKAGE_uci' from 'boolean' to 'tristate'
tmp/.config-package.in:1075:warning: ignoring type redefinition of 'PACKAGE_r8169-firmware' from 'boolean' to 'tristate'
tmp/.config-package.in:2743:warning: ignoring type redefinition of 'PACKAGE_libiwinfo-lua' from 'boolean' to 'tristate'
tmp/.config-package.in:2758:warning: ignoring type redefinition of 'PACKAGE_lua' from 'boolean' to 'tristate'
tmp/.config-package.in:2863:warning: ignoring type redefinition of 'PACKAGE_libip4tc' from 'boolean' to 'tristate'
tmp/.config-package.in:2877:warning: ignoring type redefinition of 'PACKAGE_libip6tc' from 'boolean' to 'tristate'
tmp/.config-package.in:2907:warning: ignoring type redefinition of 'PACKAGE_libxtables' from 'boolean' to 'tristate'
tmp/.config-package.in:2976:warning: ignoring type redefinition of 'PACKAGE_libblobmsg-json' from 'boolean' to 'tristate'
tmp/.config-package.in:3077:warning: ignoring type redefinition of 'PACKAGE_libiwinfo' from 'boolean' to 'tristate'
tmp/.config-package.in:3092:warning: ignoring type redefinition of 'PACKAGE_libjson-c' from 'boolean' to 'tristate'
tmp/.config-package.in:3105:warning: ignoring type redefinition of 'PACKAGE_liblua' from 'boolean' to 'tristate'
tmp/.config-package.in:3221:warning: ignoring type redefinition of 'PACKAGE_libnl-tiny' from 'boolean' to 'tristate'
tmp/.config-package.in:3334:warning: ignoring type redefinition of 'PACKAGE_libubox' from 'boolean' to 'tristate'
tmp/.config-package.in:3360:warning: ignoring type redefinition of 'PACKAGE_libubus' from 'boolean' to 'tristate'
tmp/.config-package.in:3373:warning: ignoring type redefinition of 'PACKAGE_libubus-lua' from 'boolean' to 'tristate'
tmp/.config-package.in:3387:warning: ignoring type redefinition of 'PACKAGE_libuci' from 'boolean' to 'tristate'
tmp/.config-package.in:3400:warning: ignoring type redefinition of 'PACKAGE_libuci-lua' from 'boolean' to 'tristate'
tmp/.config-package.in:4340:warning: ignoring type redefinition of 'PACKAGE_luci-base' from 'boolean' to 'tristate'
tmp/.config-package.in:4366:warning: ignoring type redefinition of 'LUCI_LANG_hu' from 'boolean' to 'tristate'
tmp/.config-package.in:4369:warning: ignoring type redefinition of 'LUCI_LANG_pt' from 'boolean' to 'tristate'
tmp/.config-package.in:4372:warning: ignoring type redefinition of 'LUCI_LANG_sk' from 'boolean' to 'tristate'
tmp/.config-package.in:4375:warning: ignoring type redefinition of 'LUCI_LANG_no' from 'boolean' to 'tristate'
tmp/.config-package.in:4378:warning: ignoring type redefinition of 'LUCI_LANG_en' from 'boolean' to 'tristate'
tmp/.config-package.in:4381:warning: ignoring type redefinition of 'LUCI_LANG_pl' from 'boolean' to 'tristate'
tmp/.config-package.in:4384:warning: ignoring type redefinition of 'LUCI_LANG_uk' from 'boolean' to 'tristate'
tmp/.config-package.in:4387:warning: ignoring type redefinition of 'LUCI_LANG_ru' from 'boolean' to 'tristate'
tmp/.config-package.in:4390:warning: ignoring type redefinition of 'LUCI_LANG_vi' from 'boolean' to 'tristate'
tmp/.config-package.in:4393:warning: ignoring type redefinition of 'LUCI_LANG_he' from 'boolean' to 'tristate'
tmp/.config-package.in:4396:warning: ignoring type redefinition of 'LUCI_LANG_ko' from 'boolean' to 'tristate'
[... 187 lines cut ...]
install -m0755 /home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk/build_dir/target-i386_pentium4_musl-1.1.16/lime-docs-2017-07-08-1499545999/en_config.txt /home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk/build_dir/target-i386_pentium4_musl-1.1.16/lime-docs-2017-07-08-1499545999/ipkg-i386_pentium4/lime-docs-minimal/www/docs/
find /home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk/build_dir/target-i386_pentium4_musl-1.1.16/lime-docs-2017-07-08-1499545999/ipkg-i386_pentium4/lime-docs-minimal -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="i486-openwrt-linux-musl-"   ; NM="i486-openwrt-linux-musl-nm" STRIP="/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk/staging_dir/host/bin/sstrip" STRIP_KMOD="/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk/scripts/strip-kmod.sh" PATCHELF="/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk/staging_dir/host/bin/patchelf" /home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk/scripts/rstrip.sh /home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk/build_dir/target-i386_pentium4_musl-1.1.16/lime-docs-2017-07-08-1499545999/ipkg-i386_pentium4/lime-docs-minimal
(cd /home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk/build_dir/target-i386_pentium4_musl-1.1.16/lime-docs-2017-07-08-1499545999/ipkg-i386_pentium4/lime-docs-minimal/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -x "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -x "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk/bin/packages/i386_pentium4/libremesh
[... cut ...]
make[1]: Leaving directory '/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk'
/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk/include/toplevel.mk:191: fallo en las instrucciones para el objetivo 'package/lime-docs/compile'
make: *** [package/lime-docs/compile] Error 2
make: se sale del directorio '/home/okupa/libremesh/lime-sdk/17.01.2/x86/generic/sdk'
-> Build error, see full output using V=s J=1 ./cooker <options>
```

If i do ```./cooker -b x86/generic``` fail in the same package.

gmarcos87, 2017-07-12:
I found the bug in this line https://github.com/libremesh/lime-packages/blob/develop/packages/lime-docs/Makefile#L66 but i don know how fix it.

ilario, 2017-07-12:
Thank you very much for the debugging, I didn't expect that line to be broken, it should be fixed by #183

ilario, 2017-07-12:
There's another proposed solution on #184 (but I still prefer #183)

ilario, 2017-07-13:
Should be finally fixed now (sorry for the huge mess) with the merge of #184 and #185.

closed 2017-07-13 by ilario