libremesh/lime-packages #189: "received packet with own address" batman-adv error on first boot

ilario, 2017-07-25:
I observed this old error (see [this](https://lists.open-mesh.org/pipermail/b.a.t.m.a.n/2014-March/011839.html) and [this](https://lists.libremesh.org/pipermail/lime-dev/2014-March/000308.html) old emails, the problem was solved here: ae0672b) happening again but only on first boot.

Only on the first boot, bat0 interface doesn't take the dummy0 interface as main one, so the resulting bat0 MAC address so is taken from an actual interface in bridge with bat0 and results in this error.

I think also @nicoechaniz saw this.

ilario, 2017-08-22:
I saw this on TP-Link WDR3600, but I didn't have time to confirm recently.

altergui, 2017-09-12:
I've seen this happen in random occasions, (not particularly on first boot)
it looks like a race condition to me, i.e. at each boot, there's a small chance that the first interface added to bat0 is not dummy0, so the "own address" error will be logged every 10 secs until reboot.

Maybe this small chance is bigger (or even equal to 1) on first boot, as you report. But i think it's irrelevant since i've seen this happen in non-first boots as well. So i would close this as "wontfix" (i.e. "can't really fix")

ilario, 2017-09-13:
Ok, let's close. I hope that we will find a way to fix this eventually (isn't there a way to delay the up of eth0? Something like a sleep 1` in ifup`?)

ilario, 2017-09-23:
In the router I'm using now the message in dmesg was always present.

I tried to fix this inserting a sleep in `/etc/hotplug.d/iface/00-netstate` for all interfaces but dummy0 (it was something like 
```
echo "$INTERFACE" | grep -q dummy
[ "$?" = 1 ] && {
	sleep 1
}
```
but this didn't change the order of activation of the interfaces, just made the process slower.

So I solved in another unclean way, setting in /etc/config/network
```
config interface 'lm_net_eth0_2_batadv_if'
	option auto '0'
```
and enabling the interface later adding in `/etc/rc.local` a line with `ifup lm_net_eth0_2_batadv_if`.
This is a dirty solution, a clean one would be to understand how netifd dependencies work (I couldn't find enough documentation).

nicoechaniz, 2017-09-30:
if you see a non stop stream of these messages you may be experiencing the "Joker node" sindrome.

TP-Link WDR3500 and 3600 have been known to enter this state after an electrical problem due to lightning or other source.

When a router is in this state, u can usually see one or more LAN leds glowing even when there's no cable connected. It can be solved by cutting the lines in the board running from the compromised port.

ilario, 2017-10-01:
In my case, and very often in many other routers, it's **just** netifd which brings up ethernet interfaces before dummy0.
As you can read in my previous comment, changing this (I did that with an ugly workaround) solved for me.
Is there any way for introducing dependencies in netifd, something like ethernet depends on dummy0?

ilario, 2018-12-08:
I think it was fixed at some point on Batman-adv side:
https://lore.kernel.org/lkml/1265569063-3973-1-git-send-email-chris.nicholson@cnick.org.uk/

Now the messages that I see are:

The newly added mac address already exists on: wlan0-mesh
batman_adv: It is strongly recommended to keep mac addresses unique to avoid problems!

Maybe we could even remove the dummy0 workaround completely as it has a race condition.

closed 2018-12-08 by ilario