libremesh/lime-packages #194: LibreMesh Zero built fails on the luci password setup

nicopace, 2017-08-01:
This video shows it: https://youtu.be/YupbScEzB-A

The image used was: http://repo.libremesh.org/current/targets/ar71xx/generic/tl-mr3020-v1/lime_mini/lede-17.01.2-lime-mini-ar71xx-generic-tl-mr3020-v1-squashfs-factory.bin

ilario, 2017-08-01:
Can you confirm that happens also with release candidate for 17.06?

nicopace, 2017-08-01:
It was on the first boot of the mini image.
Will try to reproduce it.
Will test it with the last version.

nicopace, 2017-08-01:
Also... was the first boot after flashing from stock firmware

nicopace, 2017-08-01:
It seems that it is failing creating the image.
This is the last part of the output, it says that it is too big by 158546 bytes:
```
	root (0)
4801+1 records in
4801+1 records out
2458274 bytes (2.5 MB, 2.3 MiB) copied, 0.0076228 s, 322 MB/s
[mktplinkfw] kernel length aligned to 1369776
[mktplinkfw] *** error: images are too big by 158546 bytes
cp: cannot stat '/home/nicopace/projects/redlibre/altermundi/librestack/lime-sdk/lede/ar71xx/generic/ib/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/lede-lime-mini-ar71xx-generic-tl-mr3020-v1-squashfs-sysupgrade.bin': No such file or directory
4801+1 records in
4801+1 records out
2458274 bytes (2.5 MB, 2.3 MiB) copied, 0.008946 s, 275 MB/s
[mktplinkfw] kernel length aligned to 1369776
[mktplinkfw] *** error: images are too big by 158546 bytes
cp: cannot stat '/home/nicopace/projects/redlibre/altermundi/librestack/lime-sdk/lede/ar71xx/generic/ib/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/lede-lime-mini-ar71xx-generic-tl-mr3020-v1-squashfs-factory.bin': No such file or directory
padding image to 00260000
padding image to 00259000
padding image to 0025a000
padding image to 0025c000
padding image to 00260000
padding image to 00280000
20+1 records in
21+0 records out
2752512 bytes (2.8 MB, 2.6 MiB) copied, 0.00204403 s, 1.3 GB/s

Calculating checksums...
make: Leaving directory '/home/nicopace/projects/redlibre/altermundi/librestack/lime-sdk/lede/ar71xx/generic/ib'

-> Firmware for target ar71xx/generic, profile PROFILE=tl-mr3020-v1 and flavor lime_mini cooked!
Find the binaries in /home/nicopace/projects/redlibre/altermundi/librestack/lime-sdk/output/ar71xx/generic/tl-mr3020-v1/lime_mini directory
```

Also, the output is misleading, as it says it was successful when it was not.

nicopace, 2017-08-01:
We rebooted again and it is still happening, so with the current image it still happens.

panosnethood, 2017-08-01:
Actually the problem is related to a small switch "3G/4G; WISP; AP". When the 3G/4G is selected the bug appears but not when "AP" is selected.

nicopace, 2017-08-01:
The lime_zero flavor works

p4u, 2017-08-01:
As lime-mini does not include luci-mod-admin-full it is normal that the page for changing password does not exist.

p4u, 2017-08-01:
About the cooking problem, are you including some extra package to the image? Can you post the command used?

p4u, 2017-08-01:
If you are using the community of the issue https://github.com/libremesh/network-profiles/issues/9 it is normal that the firmware is too big.

nicopace, 2017-08-01:
so... we should add a page to change the password @p4u ? does that make sense?
The image building you are right... was not paying attention to the size of the added files.

nicopace, 2017-08-01:
now the image is built... thanks @p4u .
would be nice to catch the size error so the command stops saying 'cook successful' when there was a size problem.
I can file a new bug for that if you want (so we can keep this issue just about adding the change password page to lime-packages).

p4u, 2017-08-01:
> so... we should add a page to change the password @p4u ? does that make sense?

Yes!

> would be nice to catch the size error so the command stops saying 'cook successful' when there was a size problem.
LEDE ImageBuilder does not show an error if the image is too big, so IMO it is hard to catch this error. However when the firmware is too big the "sysupgrade" file does not exist, so it is an easy way to realize of the problem.

nicopace, 2017-08-01:
Do we add a page for changing the password in luci?
A custom package that just does that?
Now there is no way to administer an image through the web interface, and if it is the stock image there is no way to get through ssh.

nicopace, 2017-08-01:
@p4u
The current message:
```
-> Firmware for target ar71xx/generic, profile PROFILE=tl-mr3020-v1 and flavor lime_zero cooked!
Find the binaries in /home/nicopace/projects/redlibre/altermundi/librestack/lime-sdk/output/ar71xx/generic/tl-mr3020-v1/lime_zero directory
```
Is misleading, because it suggests that the process finished successfully.
If we don't check the error situation, we could change the message to something like
```
Process finished.
You may find the binaries here: /home/nicopace/projects/redlibre/altermundi/librestack/lime-sdk/output/ar71xx/generic/tl-mr3020-v1/lime_zero
If not, check the log to find the issue that caused it.
```

or something like that (not good for this messages, sorry :( ).

nicopace, 2017-08-01:
@p4u this also should be moved to another issue, isn't it?

closed 2017-08-01 by nicopace

p4u, 2017-08-01:
> Do we add a page for changing the password in luci?
> A custom package that just does that?

I would prefer this page to be part of lime-webui and only enable it in case there is not luci-mod-admin-full installed (via /etc/uci-default script)

> Now there is no way to administer an image through the web interface, and if it is the stock image there is no way to get through ssh.
> I will add a new issue for this.

If there is no password and no keys configured on dropbear, ssh just works without password, so the stock images are working.

nicopace, 2017-08-03:
Issues found here reported in other issues.
Thanks.