libremesh/lime-packages #195: Webui: add link to offline documentation in /www/docs

ilario, 2017-08-01:
lime-docs packages includes documentation for offline use in LibreMesh firmware images.
The documentation is present in /www/docs directory (with a symbolic link from /docs), so it's accessible by the http server.
A link has to be added in the webui for accessing this documentation directory.

pedro-nonfree, 2018-02-18:
I don't know what's the syntax being used (asciidoc?) but I think that in the website should render that format in something more web-based

ilario, 2018-02-18:
AsciiDoc renderer is written in Python, AsciiDoctor renderer is written in Ruby. I believe that for obtaining HTML pages from AsciiDoc files, one of these two is needed but none can fit in a router.

If we really really want this, the only option I see is to download HTML pages from the website during lime-docs compilation.

nicopace, 2018-03-22:
I would go for a progressive enhancement approach... a link for now would suffice.
Also, asciidoc was ment to be read in ascii, so we shouldn't bother much about that.
@ilario is that something u can do? (A menu option in luci for accessing the docs).

ilario, 2018-12-08:
@nicopace I'm not knowledgeable with the web interface, maybe @gmarcos87?

ilario, 2019-10-04:
@gmarcos87 do you think it's a good or a bad idea to have a documentation tab in the lime-app?

nicopace, 2019-10-08:
I wouldn't add it like it is in the lime-app as it would create noise to the lime-app users without adding value.
The technical capacity traning part of lime-app needs to get thought through as a project in itself.
I wouldn't add it on the web, would just add a comment on the /etc/banner saying that there is documentation on the router, as most of the doc are more for a technical audience than they are for the average user.

ilario, 2019-10-08:
> I wouldn't add it on the web, would just add a comment on the /etc/banner saying that there is documentation on the router

Ok, makes sense. New issue for this here: #616 
So now it does not make any sense that the docs are placed in /www/docs and linked in /docs, we can change also that.

ilario, 2020-01-09:
> I wouldn't add it on the web

Ok, so in #675 I removed it from /www

closed 2020-01-09 by ilario