libremesh/lime-packages #197: Password config in lime_mini flavor webui is not available

nicopace, 2017-08-01:
This conversation comes from #194 
When you use the default lime_mini flavor (downloadable from the repo.libremesh.org site) that comes with a webui, there is no way to setup the admin password.
Without it, you can't get to the simple config screens, neither to have a root password to ssh to the device.

nicopace, 2017-08-01:
I would suggest to add a small luci package that mimics this functionality when we don't include the rest of the package.
@p4u comments that makes sense to have it in the same package. ok :+1:

p4u, 2017-08-01:
If there is no password configured and no keys in dropbear, SSH does not ask for password and the web-ui works without any password. So the images from repo.libremesh.org should work as expected.

However I agree that lime-webui should has its own password configuration page.

nicopace, 2017-08-02:
Have digged into the luci files and couldn't find which one is doing the auth thing.
This is the output:
```
/usr/lib/lua/luci/dispatcher.lua:327: Access Violation
The page at 'admin/system/admin/' has no parent node so the access to this location has been denied.
This is a software bug, please report this message at https://github.com/openwrt/luci/issues
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:327: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```

Any suggestion how to deal with this?

closed 2019-04-27 by stale[bot]

ilario, 2019-04-28:
I suppose this has been implemented in lime-app? @gmarcos87