libremesh/lime-packages #201: Set a better default distance parameter on the WiFi conifguration

p4u, 2017-08-03:
Currently distance is set to 10000 (10Km). Reducing it to 1000 (1Km) would increase the network performance for most of the libremesh deployments. For those which are using long distance links, the distance should be set manually to ensure the best performance.

In addition to reduce the default distance, there should be a way to easy configure this parameter from the libremesh webui.

ilario, 2017-08-03:
What about a different distance setting for 2.4 GHz (usually clients) and for 5 GHz (usually mesh nodes)?
For example 200 m for 2.4 GHz and 5 km for 5 GHz. 

This is how it would be set in `/etc/config/lime-defaults`:
```
config lime wifi
	option distance_2ghz '200'
	option distance_5ghz '5000'
```

p4u, 2017-08-03:
I don't think this is possible on the WiFI configuration of LEDE. Actually, if I don't remember wrong, this parameter affects mainly to the 5GHz band.

ilario, 2017-08-04:
It should be possible, it has been implemented in LibreMesh [here](/libremesh/lime-packages/blob/develop/packages/lime-system/files/usr/lib/lua/lime/wireless.lua#L106), and in LEDE the configuration of different radios (2.4 GHz and 5 GHz) is separated, so it's the distance setting in LEDE for the two radios.
Even if nowadays maybe drivers for 2.4 GHz are not affected by this parameter, this could change.

ilario, 2017-08-04:
Closing as #202 has been merged.

p4u, 2017-08-05:
You are right, it can be set independently for the radios. It might be implemented as you proposed.

nicoechaniz, 2017-08-06:
When you are using dual band devices such as tl-wdr3500 the distance on each radio is the same. This is the most common scenario in the networks in our region. So a different value for 5Ghz and 2.4Ghz would not be our default.

ilario, 2017-08-20:
@nicoechaniz so in your scenario both 2.4 and 5 GHz are used for node to node connections? Isn't 2.4 GHz radio involved more heavily in node to client connections?

digitigrafo, 2023-05-19:
>
> Let's reopen this discussion.
> @digitigrafo <https://github.com/digitigrafo> reported that plenty of 
> communities had problem with 1000 m as maximum default distance.
> I propose to set it to 5 km.
I reported it in 2019.. when I was in brasil and I find 2 communities 
that abandoned to use Lime because can't update from lime17 to lime19..

and was the same problem that we had.. lime 17 was working on long 
distance, so the feedback when you upgraded to 19 was:"libremesh is not 
more working" and there was not changelog to know what happen in the 
last release, and also the design of the project has an hole in that 
point (no config needed), but after speaking with the developers was 
easy to fix also in GUI using LUCI.. and our community survived.. but 
sadness.. for the others was too late. not possible without human 
communications and a bit of mutual aid..

I reported again in 2022 because last year also I meet 2 german 
communities that told me they tried to use Lime on ubnt m5 antennas 
already on the roof, passing from AirOS to Lime, but doesn't worked 
because the antennas lost connections with the others.. so yes the same 
problem of the distance. So out of that ecatombe of users in 2018-19... 
I think that still is a problem. because in the design the issue is 
still there. An in other side also the problem of communications between 
developers and communities is still there, when we change something that 
need to be re-configured by users is not clear that happen and where we 
announced. Github comments and matrix are not stable point of 
documentation. same when we introduced network profiles.. a lot of 
communities was not informed..

igni

ilario, 2023-05-23:
Please notice that this discussion forked on the mailing list, read the recent comments here: https://lists.autistici.org/thread/20230519.102407.1e38994a.en.html

ilario, 2023-08-16:
With @a-gave we made some measurements for understanding a bit better what happens when the distance parameter is much larger than the actual distance.
Clearly, the influence of this parameter will be strongly dependent on the quality of the link.
Anyway, on this thread we are sharing the results: 
https://lists.autistici.org/message/20230815.142621.071b5fc0.en.html

And on this repository you can find the code for repeating the measurement on your links (of for optimizing other parameters):
https://github.com/ilario/wifi-distance-setting-exploration

PR #1047 (2023-09-15): Default distance setting: increase 10x
Fix #201

Goes against the decisions implemented in #202 and #324

See the discussion on #201 and on this email thread: https://lists.autistici.org/thread/20230815.142621.071b5fc0.en.html#i20230815.142621.071b5fc0

Here, the default distance setting for 5 GHz radios is being increased from 1 km to 10 km.
And the default distance setting for 2.4 GHz from 100 m to 1 km.
closed 2023-09-15 by ilario