libremesh/lime-packages #203: Make safe-reboot aware if there is a backup or not

nicopace, 2017-08-03:
If there is no backup, safe-reboot shouldn't remove the /etc overlay

ilario, 2017-08-30:
Looking at the code [here](/libremesh/lime-packages/blob/develop/packages/safe-reboot/files/usr/sbin/safe-reboot#L105) seems that it already checks for the existence of a backup...

```
    if [ -s $file_etc_lastgood ] ; then
        safe-reboot discard
    fi
```

So is this check not working?

nicopace, 2017-09-17:
Aparently...this was a bug I transcribed from the core team.
If you can't reproduce, just tag it and close it :)

nicoechaniz, 2017-09-17:
I think it was @altergui who observed this when safe-reboot failed him in node nicojesigioia in QuintanaLibre. @altergui can you explain this further?

altergui, 2017-09-17:
> I think it was @altergui <https://github.com/altergui> who observed this
> when safe-reboot failed him in node nicojesigioia in QuintanaLibre.
> @altergui <https://github.com/altergui> can you explain this further?

steps to reproduce:

do "safe-reboot discard" in a node (without having done "safe-reboot" first)

expected:
no action, error reported

actual result:
node dies

ilario, 2017-09-21:
Ok, so what's missing is a check before the `rm -rf $dir_etc` line in safe-reboot.

commit ab32ad6d (2017-11-09): Merge pull request #251 from svalo/issue/203

Checks for $file_etc_lastgood existance before removal. Fixes #203
closed 2017-11-09 by nicopace