libremesh/lime-packages #208: How to configure an encrypted mesh based on ieee80211s

netzcoop, 2017-08-10:
hello there,
(we also asked this on lime-users@lists.libremesh.org on 09.08.17 23:29)

we try to setup different customized lime installations for a camp.

our routers:  tplink 4300.

1) the following configuration is working for ap encryption but not for
the mesh:

* cooking in branch stable (we also tried develop with the latest
working commit):

 V=s J=1 ./cooker -c ar71xx/generic --flavor=lime_camp
--profile=tl-wdr4300-v1 --community=camp/commoniee80211s

* flavors.config

lime_camp="lime-full -dnsmasq -wpad-mini wpad authsae"

* etc/config/lime_defaults:

config lime system
    option hostname 'CA-%M4%M5%M6'
    option domain 'mesh.camp.local'
    option keep_on_upgrade 'libremesh base-files-essential
/etc/sysupgrade.conf'

config lime network
    option primary_interface eth0
    option main_ipv4_address '10.5.0.0/21'
    option main_ipv6_address '2801:01e8:2::/64'
    option bmx6_mtu '1500'
    list protocols ieee80211s
    list protocols lan
    list protocols anygw
    list protocols batadv:%N1
    list protocols bmx6:13
    list protocols olsr:14
    list protocols olsr6:15
    list protocols olsr2:16
    list resolvers 4.2.2.2   # b.resolvers.Level3.net
    list resolvers 141.1.1.1 # cns1.cw.net
    list resolvers 2001:470:20::2 # ordns.he.net
    option bmx6_over_batman false
    option bmx6_pref_gw none
    option bmx7_over_batman false
    option anygw_mac "aa:aa:aa:%N1:%N2:aa"

config lime wifi
    option channel_2ghz '11'
    option channel_5ghz '48'
    option htmode_5ghz 'HT40'
    list modes 'ap'
    list modes 'apname'
    list modes 'ieee80211s'
    option ap_ssid 'Unsecure'
    option apname_ssid 'Unsecure | %H'
    option adhoc_ssid 'Mesh | Camp'
    option adhoc_bssid '16:27:F6:27:57:14'
    option adhoc_mcast_rate_2ghz '24000'
    option adhoc_mcast_rate_5ghz '6000'
    option ieee80211s_mesh_fwding '0'
    option ieee80211s_mesh_id 'Mesh_Camp'
    option ap_encryption 'psk2+ccmp'
    option ap_key 'somepassword'
    option apname_encryption 'psk2+ccmp'
    option apname_key 'somepassword'

2) if we add to the "config lime wifi"

   option ieee80211s_encryption 'psk2+ccmp'
   option ieee80211s_key 'someotherpassword'

the mesh disappears.

are there other packages we still need to get it working besides "wpad"
& "authsae" or is there anything else we are missing?

thank you!

dangowrt, 2017-08-11:
authsae is deprecated in favour of wpad-mesh or wpa_supplicant-mesh.
(wpad-mesh = wpa_supplicant-mesh + hostapd)

netzcoop, 2017-08-11:
thank you for the information.

i tested it with wpad-mesh and the wpa-supplicant-mesh with the above config,  still no encrypted mesh. after first boot the normal ap's appear for short time than disappears, the mesh also appears (unencrypted) from time to time.

in the error logs i can see nothing which gives me a hint:
Fri Aug 11 22:47:00 2017 cron.info crond[2216]: USER root pid 2940 cmd /usr/sbin/libremap-agent
Fri Aug 11 22:47:01 2017 user.warn libremap-agent[2941]: unable to execute plugin "bmx6"; ?:0: attempt to index a nil value
Fri Aug 11 22:47:01 2017 daemon.warn dnsmasq[2709]: nameserver 192.168.1.3 refused to do a recursive query
Fri Aug 11 22:47:01 2017 user.err libremap-agent[2941]: submission to http://libremap.net/api/ failed
Fri Aug 11 22:47:01 2017 user.err libremap-agent[2941]: error creating/updating router document at URL http://libremap.net/api//router/; {"error":"forbidden","reason":"key lat expected"}
Fri Aug 11 22:47:01 2017 user.warn libremap-agent[2941]: unable to execute plugin "bmx6"; ?:0: attempt to index a nil value
Fri Aug 11 22:47:01 2017 user.err libremap-agent[2941]: submission to http://libremap.berlin.freifunk.net/api failed
Fri Aug 11 22:47:01 2017 user.err libremap-agent[2941]: error creating/updating router document at URL http://libremap.berlin.freifunk.net/api/router/; -2
Fri Aug 11 22:47:02 2017 daemon.notice hostapd: handle_probe_req: send failed

p4u, 2017-08-12:
> thank you for the information.
>
> i tested it with wpad-mesh and the wpa-supplicant-mesh with the above
> config, still no encrypted mesh. after first boot the normal ap's
> appear for short time than disappears, the mesh also appears
> (unencrypted) from time to time.

It looks like an error in the wireless configuration, netifd tries to
set WiFi up but there is something wrong and it goes down again.

Can you check that wpad and authsae is really installed on your routers?

opkg list| egrep 'wpa|auth'

Acording to the example here [1] you might try 'psk2+aes' as encryption
type for ieee80211s. So the wifi section configuration of
/etc/config/lime-defaults should be something like:

option ieee80211s_encryption 'psk2+aes'
option ieee80211s_key 'someotherpassword'

For ap and apname you might also try psk2+aes or just use "psk2" (which
is the most common configuration).

After running "lime-config" you can apply the options with "lime-apply"
and or "wifi down ; wifi up".

I'm not sure why it is not working, but try different chipers might help
to find the problem.

Please, if you manage to make it work, let us know, so we can add an
entry for the documentation about "how to encrypt the mesh".

Cheers.

[1] https://wiki.openwrt.org/doc/howto/mesh.80211s

netzcoop, 2017-08-12:
thank you for the information!

i will try further....

i found out that the encrypted mesh on the TP Link WDR 4300 is only working  if the ap and the mesh are on different are sepereated.

lime_camp="lime-full   -dnsmasq -wpad-mini wpad-mesh"
my hotfix is:
lime-sdk/communities/camp/community-iee80211s-hotfix-tp/etc/uci-defaults/99-nchotfix-wireless
#!/bin/sh
sed -i "/config wifi-iface 'lm_wlan1_ap_radio1'/a \ \ \ \ option disabled '1'" /etc/config/wireless
sed -i "/config wifi-iface 'lm_wlan0_mesh_radio0'/a \ \ \ \ option disabled '1'" /etc/config/wireless
wifi down 
wifi up".

so until i find a better solution i will put mesh on 5ghz and the ap on 2ghz.

dangowrt, 2017-08-13:
On Sat, Aug 12, 2017 at 06:00:27PM +0000, Pau wrote:
> On 08/12/17 00:51, netzcoop wrote:
>
>
> It looks like an error in the wireless configuration, netifd tries to
> set WiFi up but there is something wrong and it goes down again.
>
> Can you check that wpad [...]

You need *either* wpa_supplicant-mesh (or wpad-mesh) *or* authsae.
authsae is the initial early prototype, later code for SAE was merged
into wpa_supplicant.

netzcoop, 2017-08-13:
thank you for the clarification!

so if i understand right, this seems to be one of the correct ways; to remove wpad-mini and add for example wpad-mesh:
lime_camp="lime-full -dnsmasq -wpad-mini wpad-mesh"

p4u, 2017-08-13:
> thank you for the information!
>
> i will try further....
>
> i found out that the encrypted mesh on the TP Link WDR 4300 is only
> working if the ap and the mesh are on different are sepereated. [...]

You can make this configuration permanent on your network-profile
modifying /etc/config/lime-default

Here [1] an example for configuring the wifi radios independently. Look
for "WiFi interface specific options (override general option), multiple
allowed".

[1] http://libremesh.org/docs/en_config.html

Cheers

ilario, 2017-08-21:
Have you tried with
```
option ieee80211s_encryption 'psk2/aes'
```
instead of `psk2+aes` as explained in [OpenWrt wiki](https://wiki.openwrt.org/doc/howto/mesh.80211s)?
Maybe there's no difference eh...

netzcoop, 2017-08-21:
yes we have tried id but until now it was the best option

to use the hotfix we described above: lime-sdk/communities/camp/community-iee80211s-hotfix-tp/etc/uci-defaults/99-nchotfix-wireless

netzcoop, 2017-08-26:
a small info for you:
for one week we are using this hotfix with 10  TP Link WDR 4300 routers successfully.

nicopace, 2017-08-26:
Great News!!! Send photos ;)

nicopace, 2017-08-29:
@netzcoop thanks for the report.
Would be good to be able to turn this into a good documentation for the project.
Do you think you can crunch your experience into an article... or even better, an update of http://libremesh.org/docs/en_changing_network_behavior.html (here: https://github.com/libremesh/lime-web/blob/master/docs/en_changing_network_behavior.txt)?

ilario, 2017-08-29:
In my opinion we could include wpad-mesh by default in lime_default flavor, it's 120 KB bigger (shouldn't be a problem for routers with 8 MB flash) but allows this kind of encryption without needing to recompile.

dangowrt, 2017-08-29:
the thing is that it pulls in openssl...

ilario, 2017-08-29:
I compiled with default flavor and @netzcoop's flavor, the result is that with wpad-mesh and its dependencies the sysupgrade.bin file is 704 kB bigger. Still the image fits without any problem in 8 MB routers (for WDR3600, sysupgrade.bin file increased from 4544 kB to 5248 kB).

p4u, 2017-09-14:
I'd not include this into the default flavor. Instead I would create a network profile, let's say _libremesh/encryptmesh_ with this set of packages and the proper lime-defaults configuration file.

This would require to add a new functionallity to cooker for specify a list of packages attached to a community. It might be done by a special file named _libremesh/encryptmesh/PACKAGES_ which include the set of packages that will be added to the firmware when the community is selected. I could implement that.

p4u, 2017-09-14:
Added on commit https://github.com/libremesh/lime-sdk/commit/60e6109b20c9bacd453f12dc55a3e182424305c7

So now, if a PACKAGES file is placed on the root of a community/device folder, the list of packages are added to the cooked firmware. Example here: https://github.com/libremesh/network-profiles/tree/master/libremesh/encrypt-11s I'm not sure if this setup of 11s+encryption works (cause I've not tested it) but I hope it is useful as example. If something must be added or changed to make it work, contributions would be welcome.

nicopace, 2017-09-15:
The PACKAGES files is an amazing feature @p4u , Thanks!!
Also, good example with the ecrypt-11s repo!

nicopace, 2017-09-15:
@p4u what @netzcoop told me is that the most concerning problem was that the radios couldn't work at the same time as 80211s encrypted, and AP, so they ran the script in https://github.com/libremesh/lime-packages/issues/208#issuecomment-322003540 to remove the AP from 5ghz, and the mesh from 2Ghz.

ilario, 2017-09-22:
We should document this stuff, at least in lime-example. Can someone confirm the problem of AP+enc-mesh on the same radio?

netzcoop, 2017-09-22:
for your info: we had only tplink 4300 devices ....

nicopace, 2017-09-22:
It happened to me also

ilario, 2017-09-22:
So, do you think we should include a similar script in encrypt-11s flavor or just explain this in documentation?

dangowrt, 2017-09-23:
I reckon a bug should filed on LEDE flyspray and discuss on mailing lists e.g.
lede-dev@lists.infradead.org
hostap@lists.infradead.org

ilario, 2017-09-26:
Ok, so the scope of this issue is just to document that flavor *encrypt-11s* has to be used.
@nicopace @netzcoop can you report the stuff as @dangowrt suggested?

nicopace, 2017-09-30:
I will try to do a simple setup with this to reproduce... but this has a very low priority for me now.
If @netzcoop can document it, would be much better :)

ps790, 2017-10-21:
Maybe it's the same issue as described here:
[Link](https://bugs.lede-project.org/index.php?do=details&task_id=550)

> It can be fixed by removing following lines in wpa_supplicant.c

`
if (iface->hostapd_ctrl) {
		char *cmd = "STOP_AP";
		char buf[256];
		int len = sizeof(buf);

		wpa_s->hostapd = wpa_ctrl_open(iface->hostapd_ctrl);
		if (!wpa_s->hostapd) {
			wpa_printf(MSG_ERROR, "\nFailed to connect to hostapd\n");
			return -1;
		}
		if (hostapd_stop(wpa_s) < 0)
			return -1;
	}
`

Could anybody try this and confirm if it solves the problem?

dangowrt, 2018-01-31:
I made a proposal to update the hostapd sources.
Can you give http://patchwork.ozlabs.org/patch/864478/ a try and give feedback?

aparcar, 2018-01-31:
@dangowrt at first glance you this looks wrong:

    -PKG_RELEASE:=6
    +PKG_RELEASE:=1

aparcar, 2018-02-13:
@dangowrt updates on this?

aparcar, 2018-04-28:
Thanks to @dangowrt various merges (starting with https://github.com/openwrt/openwrt/commit/d88934aa5aa9b61ec66a6f7ae0fb61d1ef889884) it's now eventually supported!
You're required to install `wpad-mesh` as mentioned before.
The patches are tested and already in use, they work very well! The connection process between nodes will take longer then before.

The patch brings also support for [DFS](http://wifi-insider.com/wlan/dfs.htm) limited frequencies which (currently?) forces you to separate AP and mesh on different radio. If that's desired have a look at the packge `lime-smart-wifi`

closed 2018-04-28 by aparcar