libremesh/lime-packages #210: travis after_success.sh asks for password

aparcar, 2017-08-12:
My current [PR ci build](https://travis-ci.org/libremesh/lime-packages/builds/263925135?utm_source=github_status&utm_medium=notification) seems never to finish as `after_success.sh` asks for a password:
```
iv undefined
iv undefined
Warning: Permanently added the ECDSA host key for IP address '138.255.88.52' to the list of known hosts.
Enter passphrase for key '/home/travis/build/libremesh/lime-packages/.ci/travisci_rsa': 
```

gmarcos87, 2017-08-12:
This error occurs when the encrypted key is created with a password. @nicopace in this commit you created the key (5729b781fa3a0f9019cedcd733ff87ca390842de), did you use password?

nicopace, 2017-08-12:
No, and this worked for the other commits... I will try later to see what is happening

nicopace, 2017-08-15:
For sure it is related wirh the fact that the branch is not from the libremesh repository...
I will modify the script to be able to finish the test without doing the privileged tasks like uploading the firmware or commenting the image.
Regards

gmarcos87, 2017-08-21:
I propouse a fix in #214

ilario, 2017-09-10:
Why not to test for the presence of `$encrypted_4ddb7c1a1584_key` `$encrypted_4ddb7c1a1584_iv` variables at the beginning of `.ci/after_success.sh`?
@gmarcos87 @nicopace do you think it would be enough? If so, should I do a PR with this or you do this?
By the way I don't get the check for the presence of `$TRAVIS_BUILD_DIR` in `.ci/after_success.sh` added in dab1d57, I would remove it.

ilario, 2017-09-18:
This change was implemented in 8f017dc as part of #214 which closes this issue :D
I still miss the meaning of
```
if [ $TRAVIS_BUILD_DIR ]; then
```
but doesn't matter, it doesn't hurt.

closed 2017-09-18 by ilario

gmarcos87, 2017-09-18:
@ilario In the hurry to restore continuous integration I forgot that line. Now I see it and I can not find the meaning. Surely this must have a motive and is not doing it. I think @nicopace should know the intent of that.