libremesh/lime-packages #213: Resolve the licensing issue

gmarcos87, 2017-08-20:
I was analyzing the licensing situation in the lime-packages repository. There is no file that indicates the global license, there is no CLA (Contributor License Agreements) and there are copyright headers set in some files inconsistently.
Talking with some of the developers we consider it appropriate to establish a license (such as GLPv3 or AGPLv3). This would affect future contributions, but it would be necessary for contributors to accept that their already written code is also licensed.
To solve this situation I propose:
* Set a license
* Clarify that the contributions made in this repository respect that license.
* Change existing headers and add missing ones.
* Create a pull request with all these changes and approve the merge among all contributors as a form of acceptance.

ilario, 2017-08-21:
+1
Can you write on lime-dev mailing list proposing the license which better fits in your opinion?

p4u, 2017-08-22:
I would add a default LICENCE file on the root of the repository which will act as default LICENCE in case there is not a specific licence inside the specific package.

I vote for AGPLv3

ilario, 2017-09-07:
I'm quite ignorant on this but AGPLv3 seems good also for me.
In my opinion it's important to have this sorted out as soon as possible.
I read devs discussing about using a custom license/modifying heavily an existing one, I'm against this.

p4u, 2017-09-12:
https://github.com/libremesh/lime-packages/pull/226

gmarcos87, 2017-09-12:
If you run:
``` git log --pretty="%an %ae%n%cn %ce" ./ | sort | uniq ``` 

This is the list of contributors:
* Daniel Golle daniel@makrotopia.org
* Frank95 franciscojim95@gmail.com
* FreifunkUFO ufo@rund.freifunk.net
* gabri94 gabrihacker@gmail.com
* Gioacchino gmazzurco89@gmail.com
* Gioacchino Mazzurco gio@eigenlab.org
* Gio gio@eigenlab.org
* GitHub noreply@github.com
* Guido Iribarren guidoiribarren@buenosaireslibre.org
* Gui Iribarren gui@altermundi.net
* Humz humzini-sweg@hotmail.com
* Ilario Gelmetti gugolimeil@gmail.com
* Ilario Gelmetti ilario@eigenlab.org
* Ilario Gelmetti iochesonome@gmail.com
* Juli juli@deltalibre.org.ar
* leonaard leone@inventati.org
* Marcos Gutierrez gmarcos87@gmail.com
* Micha Stöcker micha@stoecker.me
* nicoechaniz nicoechaniz@altermundi.net
* Nicolás Pace nico@libre.ws
* Nicolas Pace nicopace@gmail.com
* Nicolás Pace nicopace@gmail.com
* p4u p4u@dabax.net
* Pau Escrich p4u@dabax.net
* Paul Spooren paul@spooren.de
* Paul Spooren spooren@informatik.uni-leipzig.de
* Pau pau@dabax.net
* root root@gabriel-laptop.gabriel.fi.nnx
* San Piccinini spiccinini@altermundi.net
* Santiago Piccinini spiccinini@altermundi.net

There are several duplicate users, filtering in total are between 15 and 20 people. Do we need the approval of all of them?

altergui, 2017-09-12:
I trust your collective decision, and agree about urgency

p4u, 2017-09-12:
It would take too long to wait for al contributors to answer.

But there are few which contriuted most of the code, so I would wait to have the approval of this 5-6 people. If some of the "minor contributors" is not speaking now and he has problems with AGPLv3, we will manage it in the future (in the worst case we can rewrite the code, but I don't think it will be necessary).

https://github.com/libremesh/lime-packages/graphs/contributors

I vote for accepting this PR as soon as possible... it is worst to have no license than having one too free.

p4u, 2017-09-12:
@altergui can you aprove the PR so your opinion will be officialy recorded

closed 2017-09-13 by ilario