libremesh/lime-packages #221: Smonit error during compilation with lime-sdk

ilario, 2017-09-07:
During compilation with lime-sdk seems that the init script of smonit gets executed, and fails.

```
$ ./cooker --flavor=lime_default -c brcm63xx/smp --profile=FAST2704N --update-feeds --community=tgnu-tarragonahacklab/mesh
[...]
Activating init scripts
Enabling smonit to crontab
grep: /etc/crontabs/root: No such file or directory
/home/ilario/software/lime-sdk2/17.01.2/brcm63xx/smp/ib/build_dir/target-mips_mips32_musl-1.1.16/root-brcm63xx/etc/init.d/smonit: line 10: /etc/crontabs/root: No such file or directory
/home/ilario/software/lime-sdk2/17.01.2/brcm63xx/smp/ib/build_dir/target-mips_mips32_musl-1.1.16/root-brcm63xx/etc/init.d/smonit: line 11: /etc/init.d/cron: No such file or directory
```

p4u, 2017-09-12:
Right, smonit should have cron as dependency. I thought it was always enabled by default since CONFIG_BUSYBOX_DEFAULT_CROND is usualy set to yes, but looks like it is not in your router profile.

p4u, 2017-09-12:
Can you try this patch on lime-packages?

``
diff --git a/packages/smonit/Makefile b/packages/smonit/Makefile
index 35a150d..f8f0bdb 100644
--- a/packages/smonit/Makefile
+++ b/packages/smonit/Makefile
@@ -20,6 +20,12 @@ define Package/$(PKG_NAME)
   DEPENDS:=
 endef
 
+define Package/$(PKG_NAME)/config
+  select CONFIG_BUSYBOX_CONFIG_CROND
+  select CONFIG_BUSYBOX_CONFIG_CRONTAB
+endef
+
+
 define Package/$(PKG_NAME)/description
   Small modular daemon for monitoring system processes using hooks
 endef
``

ilario, 2017-09-12:
#224 is not related to this issue. The `/etc/init.d/smonit enable` was run on my laptop, during compilation. Not on the router. So the error is in [init.d/smonit](/libremesh/lime-packages/blob/develop/packages/smonit/files/etc/init.d/smonit) file.

p4u, 2017-09-13:
I would not say during compilation but after compilation and installation of packages on the destination folder which will be packed into the firmware image. The process run "/etc/init.d/<daemon> enable" and it crashes if /etc/crontab does not exist (cause CRONTAB is not installed). This was my guest!

Have you checked it and  #224 does not solve this issue?

ilario, 2017-09-13:
I checked and does not solve.
The script targets the `/etc/crontabs/root` on my laptop, not on the router, this is the problem (and fails because on my laptop neither crontab nor cron is installed).

G10h4ck, 2017-09-18:
I can reproduce it

`Configuring ppp-mod-pppoe.
rm -f /home/gio/Builds/lime-sdk/17.01.2/brcm63xx/smp/ib/build_dir/target-mips_mips32_musl-1.1.16/root-brcm63xx/usr/lib/opkg/lists/*
make[2]: Leaving directory '/home/gio/Builds/lime-sdk/17.01.2/brcm63xx/smp/ib'
make package_postinst
make[2]: Entering directory '/home/gio/Builds/lime-sdk/17.01.2/brcm63xx/smp/ib'

Cleaning up

Activating init scripts
Enabling smonit to crontab
grep: /etc/crontabs/root: No such file or directory
/home/gio/Builds/lime-sdk/17.01.2/brcm63xx/smp/ib/build_dir/target-mips_mips32_musl-1.1.16/root-brcm63xx/etc/init.d/smonit: line 10: /etc/crontabs/root: No such file or directory
/home/gio/Builds/lime-sdk/17.01.2/brcm63xx/smp/ib/build_dir/target-mips_mips32_musl-1.1.16/root-brcm63xx/etc/init.d/smonit: line 11: /etc/init.d/cron: No such file or directory
rm -f /home/gio/Builds/lime-sdk/17.01.2/brcm63xx/smp/ib/build_dir/target-mips_mips32_musl-1.1.16/root-brcm63xx/usr/lib/opkg/info/*.postinst
make[2]: Leaving directory '/home/gio/Builds/lime-sdk/17.01.2/brcm63xx/smp/ib'
make build_image
make[2]: Entering directory '/home/gio/Builds/lime-sdk/17.01.2/brcm63xx/smp/ib'

Building images...
make V=ss -C target/linux/brcm63xx/image install TARGET_BUILD=1 IB=1 EXTRA_IMAGE_NAME="lime_default" \
        PROFILE="DEVICE_FAST2704N"
make[3]: Entering directory '/home/gio/Builds/lime-sdk/17.01.2/brcm63xx/smp/ib/target/linux/brcm63xx/image'
mkdir -p /home/gio/Builds/lime-sdk/output/brcm63xx/smp/FAST2704N/lime_default /home/gio/Builds/lime-sdk/17.01.2/brcm63xx/smp/ib/build_dir/target-mips_mips32_musl-1.1.16/linux-brcm63xx_smp/tmp`

altergui, 2017-09-18:
fixed in https://github.com/libremesh/lime-packages/pull/233

PR #224 (2017-09-12): Add CRON as dependency for smonit
Required for some profiles which do not have CRON enabled by default
Fix #221
commit a0cf213d (2017-09-12): Add CRON as dependency for smonit

Required for some profiles which do not have CRON enabled by default
Fix #221
closed 2017-09-18 by ilario