libremesh/lime-packages #227: Add automation for publishing packages to repo.libremesh.org

nicopace, 2017-09-12:
Right now, packages are being published manually by @p4u .
Would be great to automate the process as a travis task.
This is what is being run right now:
https://github.com/libremesh/lime-sdk/blob/master/snippets/build_and_cook_all.sh

Create a travis task that runs this for each new release.
Define how a release branch is differenciated from any other branch.... probably because is a [0-9][0-9]\.[0-9][0-9]\(\.[0-9][0-9]\) label.

ilario, 2018-12-08:
Any update on this? @aparcar do you have idea of the status of this?

ilario, 2019-10-04:
This has been solved long time ago using continuous integration publishing compiled packages here:
https://snapshots.libremesh.org/packages/
Closing.

closed 2019-10-04 by ilario