libremesh/lime-packages #236: IPv6 default route announced even when there's no IPv6 gateway

ilario, 2017-09-22:
Connecting to a LibreMesh node I received a default route for IPv6, even if there was no IPv6 gateway.
```
# ip -6 r
2a00:1508:a2e:1300::/64 dev enp0s25 proto kernel metric 202 mtu 1500 pref medium
fe80::/64 dev enp0s25 proto kernel metric 256 pref medium
default via fe80::a8aa:aaff:fe2e:13aa dev enp0s25 metric 202 mtu 1500 pref medium
```

This wouldn't be a problem if we used ISP DNS as they don't answer for AAAA queries, but the custom ones set in lime-defaults (4.2.2.2, 141.1.1.1) does.
Most of modern applications fallback on IPv4 after realizing that IPv6 doesn't work, but many still don't.
This result in some applications not being able to reach the internet, for example ping.

In my opinion we should disable IPv6 default route announcement by default and let the user or some automatic stuff like watchping6 from #71 activate it.

ilario, 2017-09-22:
I think @altergui also observed this.

ilario, 2017-09-25:
Can anyone point me to the line in the code where this is set?

G10h4ck, 2017-12-04:
> Most of modern applications fallback on IPv4 after realizing that IPv6 doesn't work, but many still don't.

ilario, 2017-12-04:
> reasonable linux distributions handle ICMP messages properly

My experience of this problem was while running Linux

ilario, 2018-01-21:
> reasonable linux distributions handle ICMP messages properly

* Arch Linux does not
* LEDE 17.01.4 does not (I was trying to "opkg update" on a plain LEDE connected to a LibreMesh gateway and failed using IPv6)

Maybe we should consider this issue more seriously? @G10h4ck

ilario, 2018-01-22:
Related to #71

altergui, 2018-01-22:
> I think @altergui <https://github.com/altergui> also observed this.

NAK, the default gw should always be announced, regardless of v6
internet connectivity, the same way a default ipv4 gw is announced,
regardless of v4 internet connectivity

(else, there would be no ipv6 way of crossing over to another cloud, in
a set of libremesh clouds connected between them , but not to the internet)

dangowrt, 2018-01-22:
Hi!

On Mon, Jan 22, 2018 at 02:03:03PM +0000, Gui wrote:
> On 22/09/17 11:28, Ilario Gelmetti wrote:
> > I think @altergui <https://github.com/altergui> also observed this.
> 
> NAK, the default gw should always be announced, regardless of v6
> internet connectivity, the same way a default ipv4 gw is announced,
> regardless of v4 internet connectivity

I disagree, or at least only a ULA prefix should be announced.
Truely public v6 addresses should appear only for gateway tunnels.

> 
> (else, there would be no ipv6 way of crossing over to another cloud, in
> a set of libremesh clouds connected between them , but not to the internet)

That could also be achieved using an ULA.
Our currenty way of having an publicly routed default IPv6 prefix
(literally **everybody** uses 2a00:1508:0a42:2400::/64) as the default
prefix for LiMe nodes is wrong in a couple of ways:
 - this prefix can impossibly be routed back to all nodes using it
 - using addresses within that prefix as a source address will hence
   only work in a single-gateway scenario **having exactly this prefix
   delegated from upstream**
 - RAs with that public prefix are sent out and hence clients pick
   (unreachable form outside) addresses, and as the default gateway
   is also within the 2000::/4 range it will make clients believe
   that we got access to other public IPv6 destinations which isn't
   true in most cases -- this will slow down surfing as browsers
   sometimes try reaching IPv6 addresses and fail.

I suggest to move the default main_ipv6_address into an
network-name-hash-based ULA prefix, e.g. fd%N1:%N2%N3:%N4%N5::/48 or
something like that to decrease the overall confusion...

Cheers

Daniel

ilario, 2019-10-05:
Annoying @stalebot closed this, but in my opinion is a possible-to-solve issue.
I mean, this happens because the clients receive a default IPv6 route, even if on the router there is none, we have to understand why this happens.

On an isolated LibreMesh router (not connected to other nodes), the IPv6 routes are like this:

```
# ip -6 route show
2a00:1508:af3:3400::/64 dev br-lan proto kernel metric 256 pref medium
2a00:1508:af3:3400::/64 dev anygw proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev anygw proto kernel metric 256 pref medium
fe80::/64 dev eth0-1_17 proto kernel metric 256 pref medium
fe80::/64 dev eth0.2 proto kernel metric 256 pref medium
fe80::/64 dev eth0-2_17 proto kernel metric 256 pref medium
fe80::/64 dev eth0-2_259 proto kernel metric 256 pref medium
fe80::/64 dev eth0_17 proto kernel metric 256 pref medium
fe80::/64 dev dummy0 proto kernel metric 256 pref medium
fe80::/64 dev br-lan proto kernel metric 256 pref medium
fe80::/64 dev wlan0-ap proto kernel metric 256 pref medium
fe80::/64 dev wlan0-apname proto kernel metric 256 pref medium
fe80::/64 dev wlan0-mesh proto kernel metric 256 pref medium
fe80::/64 dev wlan0-mesh_17 proto kernel metric 256 pref medium
fe80::/64 dev wlan0-mesh_259 proto kernel metric 256 pref medium
fe80::/64 dev wlan1-ap proto kernel metric 256 pref medium
fe80::/64 dev wlan1-apname proto kernel metric 256 pref medium
fe80::/64 dev wlan1-mesh proto kernel metric 256 pref medium
fe80::/64 dev wlan1-mesh_17 proto kernel metric 256 pref medium
fe80::/64 dev wlan1-mesh_259 proto kernel metric 256 pref medium
```
as you can see, there is no default route.

But on my laptop, connected via wireless, I have:

```
$ ip -6 route show
::1 dev lo proto kernel metric 256 pref medium
2a00:1508:af3:3400::/64 dev wlp3s0 proto ra metric 600 pref medium
fe80::/64 dev wlp3s0 proto kernel metric 600 pref medium
default proto ra metric 20600 
	nexthop via fe80::c6e9:84ff:fef9:c237 dev wlp3s0 weight 1 
	nexthop via fe80::a8aa:aaff:fef3:34aa dev wlp3s0 weight 1 pref medium
```

and on another laptop, connected via cable I have:

```
$ ip -6 route show
2a00:1508:af3:3400::/64 dev enp7s8 proto ra metric 100 pref medium
fe80::/64 dev enp7s8 proto kernel metric 100 pref medium
fe80::/64 dev enp7s8 proto kernel metric 256 pref medium
default via fe80::c6e9:84ff:fef9:c237 dev enp7s8 proto ra metric 100 pref medium
```

What does originate the default route in the clients?

nicopace, 2019-10-08:
Don't take it on @stalebot that is helping us, we love you little stale!
thanks for reopening this relevant issue @ilario , and for the thorough debug.

ilario, 2019-11-28:
@dangowrt suggestion is being implemented in #656 and #657 

It does not remove the default route

```
ip -6 r
::1 dev lo proto kernel metric 256 pref medium
fd0d:fe46:8ce8::/64 dev wlp3s0 proto ra metric 600 pref medium
fe80::/64 dev wlp3s0 proto kernel metric 600 pref medium
default proto ra metric 20600 
	nexthop via fe80::c24a:ff:fedd:6800 dev wlp3s0 weight 1 
	nexthop via fe80::de9f:dbff:fe37:28a9 dev wlp3s0 weight 1 
	nexthop via fe80::a8aa:aaff:fe0d:feaa dev wlp3s0 weight 1 pref medium
```

but the fact that the IPv6 is a private one makes the various software to not use it for reaching IPv6 internet, which fixes this issue!

ilario, 2019-12-12:
Thanks to @dangowrt suggestion, @spiccinini fixed this in #657.

closed 2019-12-12 by ilario