libremesh/lime-packages #237: safe-reboot should not use sleep for grace time

ilario, 2017-09-22:
Can happen that with a bad configuration the router goes into a **reboot loop**.
I didn't hit this problem while using safe-reboot, but I think that in such a case using `sleep $fallback_timeout` is not going to work.
An alternative is to compare the current time (I suppose routers clock usually works, right?) with the creation timestamp of `$file_etc_lastgood`.

G10h4ck, 2017-12-04:
Clock in routers is not reliable at all because most of them doesn't have a battery to keep it coherent accross reboots, so when your router boot it's clock is foobar once ntp manage to sync it

ilario, 2018-12-08:
So what's the way to detect a reboot loop? Maybe we could add a check before the sleep step: checking the existence of a file, if the file is already present means that we're in a reboot loop, then create the file and sync.

ilario, 2019-10-05:
If we decide to put more effort in safe-reboot, let's remember also of the issue killed by the annoying @stalebot #238

nicopace, 2019-10-08:
Can you share the steps to reproduce the issue? it seems like a no-issue for now.

ilario, 2019-10-09:
I did not actually test it, as it would require to induce a reboot loop with some very bad configuration.

Let's suppose that you use the software like this (taken from the safe-reboot help):

https://github.com/libremesh/lime-packages/blob/13b5abda61dadebfa8197767f6a7c3c0a1a554d5/packages/safe-reboot/files/usr/sbin/safe-reboot#L31-L34

so that you launch safe-reboot, do some very bad configuration, reboot for testing it and the router enters in a reboot loop or a kernel panic due to the new configuration.

The mechanism for reverting the old configuration, here:

https://github.com/libremesh/lime-packages/blob/13b5abda61dadebfa8197767f6a7c3c0a1a554d5/packages/safe-reboot/files/usr/sbin/safe-reboot#L104-L106

would not activate, as the reboot or kernel panic happens before the `$fallback_timeout` time has passed.

To fix this seems complex (maybe adding a +1 to a file every time the router boots, so that we can detect if too many reboots happened?), maybe it's better to just consider this as a very disastrous case that safe-reboot cannot manage. If you agree, we can close this issue as a "won't fix".

nicopace, 2019-10-09:
i agree.

ilario, 2019-10-09:
Ok, closing.

closed 2019-10-09 by ilario