libremesh/lime-packages #238: safe-reboot add reminder to banner and PS1

ilario, 2017-09-23:
When safe-reboot is activated:

* it should add some warning to be displayed on next boot, for example in `/etc/banner.notes` to remind the user to issue either `safe-reboot cancel` or `safe-reboot discard`;
* it could change the prompt of the shell ($PS1 variable) to remind that the system is going to reboot (e.g. *root@LiMe-aaaaaa safe-reboot scheduled :~#*) and change it after reboot to remind to exec the cancel or discard command (e.g. *root@LiMe-aaaaaa config reset scheduled :~#*).

These warnings should be deleted when safe-reboot is stopped (with safe-reboot cancel or discard commands).

amuuza, 2017-10-04:
What is safe-reboot?

ilario, 2017-10-04:
In case you don't know what `safe-reboot` is:
is [this](https://github.com/libremesh/lime-packages/blob/develop/packages/safe-reboot) thing: 
```
When you run $(PKG_NAME) specifying a grace time, it makes a backup of /etc at that point,
and after the grace time reboots the system. At the next boot, it waits again for a timeout;
if you can't login and delete the backup, it will restore it and reboot.
This mechanism makes it possible to run $(PKG_NAME), try risky operations (like restarting
the network), and if something goes wrong, just wait for the fallback timeout.
Even more, you can make actual changes to /etc config and reboot. If it comes back up right,
you can login and delete the backup. If something went wrong with the new /etc, just wait
for the revert timeout.
```

nicopace, 2018-02-02:
If safe-reboot is running it is hard, even for the same user that is using it, to know when it will reboot the system.

It would be great if it can notify all users connected that it will reboot the system and how to prevent it or react to it.

I think it could be added here:
https://github.com/libremesh/lime-packages/blob/feb3b8996fe8e008ff391dd6feef9a6f11225dfa/packages/safe-reboot/files/usr/sbin/safe-reboot#L100

This is the function that it could use to notify:
```bash
function message_to_all_loggedin_users {
  message="$1";
  for pts in `ls /dev/pts/`; do
    echo "$message" > /dev/pts/$pts;
  done;
}
```