libremesh/lime-packages #242: smonit should run also as in the init process

ilario, 2017-09-24:
It's happening frequently on my router that dnsmasq-dhcpv6 doesn't start with the system so that I have to wait smonit to run (every 5 minutes) for having a DHCP server.
I propose to add smonit also as last process in init so that it's run just after the boot.

ilario, 2017-10-29:
this issue is about creating a small init script running after all other init scripts and executing smonit

G10h4ck, 2017-12-04:
I believe this should not be done, we risk to cause race conditions, as services being tested could be not yet ready and so running the *_fixes at that time could make the openwrt/LEDE init system crazy

closed 2017-12-04 by G10h4ck

ilario, 2017-12-04:
Ok, any other solution?
Maybe we could decrease the interval time from 5 min to 2-3 min?
Or just check if 17.01.4 uses a good version of dnsmasq and move to that as soon as possible?
Anyway I reproduced this dnsmasq problem also running LibreMesh in a virtual machine, and dnsmasq never starts by itself.

G10h4ck, 2017-12-04:
Please open or reference another issue for the specific dnsmasq bug