libremesh/lime-packages #243: smonit longfix should also run quickfix

ilario, 2017-09-24:
On my router seems that dnsmasq-dhcpv6 dies quite frequently, so that smonit reached the limit for `hook_quickfix` executions without finding dnsmasq running at the next occasion so it started executing `hook_longfix` which does nothing and the problem persists.
So, in my opinion when smonit reaches the point to execute `hook_longfix` should also execute `hook_quickfix` in addition.

ilario, 2017-10-29:
https://github.com/libremesh/lime-packages/blob/develop/packages/smonit/files/usr/sbin/smonit#L77

p4u, 2017-11-01:
IMO longfix should only execute quickfix if the hook programmer wants explicity to do so. So longfix() function can call quickfix() and/or do anything else.

If you have a proposal for longfix on this hook, please send the PR

ilario, 2017-12-29:
@modante if you want to address this, the changes should go in [this file](/libremesh/lime-packages/blob/develop/packages/smonit/files/etc/smonit/dnsmasq) and the last part of the file should change from this:

```
function hook_longfix()
{
	log="Doing nothing for longfix"
	true
}
```

to something more similar to this (I did not test):

```
function hook_longfix()
{
	log="Repeating quickfix for longfix"
	hook_quickfix
}
```

(and I don't get why there's a `true` in the last line).

modante, 2017-12-29:
Then, I have updated the dnsmasq package. Should I change this lines in the /etc/smonit/dnsmasq too?
Thanks :-)

ilario, 2017-12-29:
Nope, on your router let's keep the smonit stuff as it is, so we can check if upgrading the dnsmasq worked. I was referring to a pull request to this code repository, for addressing the issue regardless of the dnsmasq version :) You should fork the lime-packages repository, do the modification there and do a pull request for having your fix merged here. Then we can test on the pull request before merging.

ilario, 2018-05-28:
@modante did you try to do the pull request on this?

modante, 2018-06-04:
@ilario what do you mean? Sorry but I am not familiar with this terms and I don't understand what you mean with "pull request".

PR #417 (2018-10-22): smonit dnsmasq repeat quickfix as longfix
Fixes #243
closed 2018-10-22 by nicopace