libremesh/lime-packages #244: prioritize internet traffic from local connections over traffic from the mesh

ilario, 2017-09-25:
For sharing the internet connection without worries of an eventual decrease of the original bandwidth for personal use, QoS/TC (quality of service, traffic control) can be used.
This setup should prioritize data coming from hosts directly connected to the gateway, while giving lower priority to internet traffic originated from other hosts of the mesh.

amuuza, 2017-10-04:
Nice idea! I do not know what would be the best by-default setting for this, but I think it would be good if it could be easily configurable.

ilario, 2017-10-04:
@amuuza I tagged this with epic 'cause I think it's quite epic to do this ;)
In my opinion this could fit a bachelor thesis or a GSoC.

panosnethood, 2017-12-02:
Perhaps it is easier to allow an admin to set a "minimum" guaranteed bandwidth for the directly attached hosts? And perhaps make sure that this is not too high? :-) I guess there are already many already available tools to do this, no? 

For a dynamic prioritization I am sure there are also tools, but it is more complicated and not always easy to "guarantee". So, the configurable minimum guaranteed rate is a good idea, I think.

(sorry and please ignore  if this has been already discussed)

G10h4ck, 2017-12-04:
This proposal hasn't been implemented yet because while it can seems a simple thing at first glance it is not in our network model we have to answer some questions in an elegant way

What we consider directly attached host in a libre-mesh network, and how we detect them, or should the one that share the connection input a list of MAC/IP addresses of prioritized devices?
What is the bandwidth of the upstream connection (we need to know it to setup any working traffic prioritization policy), do we have some way to measure it or the one who share it should input this information?

closed 2019-04-27 by stale[bot]

ilario, 2019-04-28:
Even if stale bot closed this, I still think users would love to have such feature. Still I recognize that, as @G10h4ck wrote, this is verry complex to implement.

ilario, 2019-10-05:
A related discussion will go on on libremesh/pirania#3, libremesh/pirania#37, and #261