libremesh/lime-packages #245: Remove useless line in travis/after_success.sh

ilario, 2017-09-30:
As noted in #210, there's a strange check in [after_success.sh](/libremesh/lime-packages/blob/develop/.ci/after_success.sh)
```
    if [ $TRAVIS_BUILD_DIR ]; then
        [...]
    else
        echo "not a pull request branch, skip copying binaries."
fi
```
In my opinion it's useless, @nicopace can that line be deleted?

nicopace, 2017-10-01:
From what I remember, the check of the existence of TRAVIS_BUILD_DIR is necessary because travis is not only executed on pull requests but also on merges and other situations.
This ensures that we only have builds copied to our server when those are new branches and not merges.

ilario, 2017-10-01:
Wasn't this check done by
```
if [ ! -z "$encrypted_4ddb7c1a1584_key" ]; then
```
implemented in #214 because the other check didn't work?
And please note that the TRAVIS_BUILD_DIR variable is used also in the code **before** the check I'm questioning.

nicopace, 2017-10-02:
> Wasn't this check done by
> if [ ! -z "$encrypted_4ddb7c1a1584_key" ]; then
> implemented in #214 because the other check didn't work?
> And please note that the TRAVIS_BUILD_DIR variable is used also in
> the code before the check I'm questioning.

You can try and see then!
Don't remember well... I guess the idea was not to push to the images
server an image that does not belong to a merge request.
If that happens... then it is ok :)

ilario, 2018-10-22:
the mentioned line does not exist anymore, closing

closed 2018-10-22 by ilario