libremesh/lime-packages #253: 802.11r (fast BSS transition (FT) - fast roaming)?

ilario, 2017-11-07:
Is it needed to improve the roaming offered by Batman and Anygw?
In this case, we can look to 802.11r

802.11r on Wikipedia:
https://en.wikipedia.org/wiki/IEEE_802.11r-2008

Usage on LibreMesh as tested by Nk:
https://lists.libremesh.org/pipermail/lime-users/2017-July/000781.html

ilario, 2018-02-04:
Options for 802.11r in OpenWrt:
https://lede-project.org/docs/user-guide/wifi_configuration#fast_bss_transition_options

G10h4ck, 2019-04-24:
AFAIR this is already enabled by default

aparcar, 2019-04-24:
cool. could you verify this via docs? then we could close this

ilario, 2019-04-24:
@nordurljosahvida can you have a look?

nordurljosahvida, 2019-04-24:
@G10h4ck what do you mean enabled by default? If nothing has changed you need to specify details of every AP in the network, and therefore we would need a provisioning system. How could it be already working?

nordurljosahvida, 2019-04-24:
@ilario thanks for following through with this

spiccinini, 2019-04-24:
To add more light of the 802.11k/v/r extensions here is useful info https://community.ubnt.com/t5/UniFi-Wireless/Wireless-LAN-Roaming-FAQ/m-p/1825118. 
Also here is a simple explanation of the implementation of 802.11v/k/r for iOS devices (so from a client perspective) https://support.apple.com/en-us/HT202628 and the iOS devices support. 
And here a Cisco documentation on the subject https://www.cisco.com/c/en/us/td/docs/wireless/controller/8-1/Enterprise-Mobility-8-1-Design-Guide/Enterprise_Mobility_8-1_Deployment_Guide/Chapter-11.pdf

It's not very clear to me what part of these features are supported by hostapd / wpa_supplicant (in the case of authenticated networks). Some info on this is here https://blog.freifunk.net/2018/08/13/dawn-final-post/

nordurljosahvida, 2019-04-25:
@spiccinini thanks for the info. @G10h4ck as far as I can tell what I wrote in the ML back in the day is still valid: https://lists.libremesh.org/pipermail/lime-users/2017-July/000781.html

altergui, 2019-04-25:
personally, i'm disinclined to put time into features that i don't understand the final purpose. probably happens to other devs as well.
i do understand that 802.11r only matters when you're using WPA2-Enterprise i.e. a RADIUS auth server.

nick, why are you using WPA2-Enterprise?

nordurljosahvida, 2019-04-26:
@altergui it's been some time, but i don't remember the WPA2-E restriction. are you sure?

ilario, 2019-10-04:
Skimming through https://www.networkcomputing.com/wireless-infrastructure/wifi-fast-roaming-simplified also seems that this option is designed for working with WPA2 Enterprise.
I cannot find if it can be useful for open networks.
@nordurljosahvida can you provide some link about 802.11r on open networks?
In the meanwhile I'll close this.

closed 2019-10-04 by ilario

ilario, 2019-12-11:
The [FreeMesh router](https://freemeshwireless.com/) designers are using 802.11r together with WPA2-psk.
You can see their code here:
https://gitlab.com/slthomason/freemesh/tree/master/package/freemesh/cfg-router