libremesh/lime-packages #258: Adding an easier-to-use console text editor in lime-full

ilario, 2017-12-02:
Up to now vim is the default LEDE console text editor, we all love vim but it is not user friendly at all.
For routers with more than 4 MB we can consider to include `nano` also.

ilario, 2017-12-02:
Adding `nano` to a normal ar71xx wdr3600 sysupgrade image with lime-default flavor increased its size by 64 kB.

nicopace, 2017-12-02:
The users that get to use a console through ssh I believe that should be capable enough to install their own editor if they want to.
What would be the usecase for changing the default behavior of having vi installed?

ilario, 2017-12-02:
> The users that get to use a console through ssh I believe that should be capable enough to install their own editor if they want to.

IF they have internet connectivity.

> What would be the usecase for changing the default behavior of having vi installed?

What do you mean?

nicopace, 2017-12-02:
A usecase is a user that needs the functionality, and a case for it to use it that way.
In this case, the user could be: a use that is turning into a power user but it is not yet.
The usecase could be: The user needs a simpler editor because it get's stuck inside vi (?).

Any user can always use lime-sdk to do his own choosing of packages (or chef), so we are talking about changing the default behavior.

ilario, 2017-12-02:
>  In this case, the user could be: a use that is turning into a power user but it is not yet.

Ok, yep, that's the usecase: a user that following [our instructions](http://libremesh.org/docs/en_quick_starting_guide.html) can connect via SSH but can't edit configuration files with `vim` as its usage it's not as trivial and as documented on our website as SSH usage.

> so we are talking about changing the default behavior

Yep.
Maybe there are also other commands that the user is used to and expects to find and are not included in LibreMesh by default, like `iwconfig`, but in my opinion `vim` is the most problematic bit.

This issue is mainly a question: do you think it's worth to add 64 kB to the image for having `nano`?

nicopace, 2017-12-02:
I feel this is a no-issue... no user via mailing-list or matrix, that is willing to do ssh, is not willing to understand vi.

For sure a user that gets to the end of [our instructions](http://libremesh.org/docs/en_quick_starting_guide.html) has already got through a lot of a lot of technicalities: ssh, ips, scp.
Also it is an optional step.

Also, Pau has been doing a great job to shrink the image, I feel we for sure have a lot of uses for 64kb than adding a second editor.

ilario, 2017-12-02:
> I feel this is a no-issue... no user via mailing-list or matrix, that is willing to do ssh, is not willing to understand vi. 
> For sure a user that gets to the end of our instructions has already got through a lot of a lot of technicalities: ssh, ips, scp.

Yep, that's why it was a question, could be that this is not a real problem. Let's see if anyone else comments...

> Also, Pau has been doing a great job to shrink the image, I feel we for sure have a lot of uses for 64kb than adding a second editor.

That's why I didn't propose to add it in lime-basic package (base for lime_zero and lime_mini flavors), which is where it makes sense to keep the image as small as possible.

panosnethood, 2017-12-02:
I think that if there are not severe space constraints an additional easier-to-use editor makes sense. But for the only-vi version, I would include somewhere visible "the smallest Vi tutorial in the world" or something like this :-)

Here is my take and I challenge to make it shorter :-)

"Vi is an extremely powerful console editor based on a very clever idea (the only thing you need to understand before using it yourself): there are two "modes" of operation. One in which you type as you do always (editing or Insert more) and one in which every keyboard key is transformed to a useful editing action (Command mode).  To switch from the command mode to the insert mode you press "i" (to type where the cursor is) or "o" (to create a new line). To switch from the Insert mode to the Command mode you press <Escape>. At the command mode you can navigate the text with the letters "j", "k", "l", "h" and exit with ":wq" (to save the current version) or ":q!" (to quit without saving). The rest is optional and you can discover by yourself! And don't forget, there are two modes: command mode and insert mode. That's all!"

ilario, 2017-12-02:
@panosnethood thaaat's not a bad idea!
Such a guide should be included in https://github.com/libremesh/lime-web/tree/master/docs as `en_vim-guide.txt` so that it gets installed in the routers in the `/docs` directory (but just on lime-full installations for routers with >4 MB flash memory).
Can you open a new issue on this on lime-web repository or directly a pull request to lime-web (adding a bit of asciidoc markup)?

aparcar, 2017-12-22:
Can be closed due to merge?

ilario, 2017-12-23:
What's still missing is linking this guide from every instance of "editing a configuration file" in the rest of the documentation, for example [here](https://github.com/libremesh/lime-web/blob/1864ec72e88867d33c06374501669cf688e458bf/docs/en_quick_starting_guide.txt#L172) or [here](https://github.com/libremesh/lime-web/blob/1864ec72e88867d33c06374501669cf688e458bf/docs/en_connecting_nodes.txt#L16).

ilario, 2018-02-04:
The issue is not related anymore to lime-packages but to lime-web. Closing here, opening on libremesh/lime-web#51

closed 2018-02-04 by ilario