libremesh/lime-packages #260: Missing PPPoE Luci App in WDR3500

modante, 2017-12-06:
I think is a package that is usefull to be included in the images. At least in the **lede-17.01.2-lime-default-ar71xx-generic-tl-wdr3500-v1-squashfs-factory.bin** is not included by default (although is easily installed).
Regards :-)

peter-tank, 2017-12-06:
unsubscrib

aparcar, 2018-01-23:
@modante can you compare how big the package is? An image with and without the package installed? Maybe it has some additional dependencies bloating the image over 8MB

dangowrt, 2018-01-23:
I also noticed the missing luci-proto-ppp (which is installed on
vanilla LEDE releases) on TP-LINK W8970B (Lantiq VRV200 xDSL router).
I guess it should become part of the lime-default flavour, because it
really isn't that big (3.7kB). Maybe just adding the 'luci'
meta-package instead of just luci-mod-admin-full and
luci-theme-bootstrap to the lime-default package-set would be the most
appropriate solution imho.

@p4u @G10h4ck @altergui

commit 170f33e9 (2018-01-23): Merge pull request #286 from libremesh/aparcar-patch-3

replace luci{-mod-admin-full,} to fix #260
commit 2a4db8c5 (2018-01-23): replace luci{-mod-admin-full,} to fix #260

Install full luci instead of "only" the luci-mod-admin-full. The package luci-proto-ppp is additionally installed which result's in a squashed file size increase of < 1KB.
closed 2018-01-23 by p4u