libremesh/lime-packages #261: How to setup QoS for a guest WiFi AP with limited internet (gateway) access?

modante, 2017-12-11:
Hello.
We are a little neighborhood with 3 libremesh routers (wdr3500, wdr3600 and wdr4100) sharing the (expensive) cost of a WISP (Wireless Internet Service Provider) in order to have a basic internet connection in the rural area where we live. By now we are encrypting the AP connections but we would like to provide a minimum connection for guests but limited in order to disturb as less as possible to the users who pay for the provider, and also to allow them to use the rest of the service that the free network could offer like sharing files between users, a chat service and others.
Is because we would like know how to configure QoS (or whatever tool needed for) in order to allow a unencrypted LibreMesh AP added to each router of the mesh with the only restriction of a limited bandwith connection to internet (or with the lowest priority over the rest of the users who support the costs).
I have installed QoS in one of the routers and I can limit the bandwith for a specific users adding their IPs, but I cannot get to limit it for all the users connected to an specific AP Station.

Thank you very much for your support :-)

ilario, 2017-12-29:
Related #244

modante, 2017-12-31:
QoS looks like difficult for my skills.
Another question: is there an easy way to create a captive portal where users register the firs time and after that and depending of the group they belong can access to full or limited internet?
Thanks and regards :-)

ilario, 2018-01-01:
There's [Pitbull](https://github.com/libremesh/pitbull) being developed by @nicopace but I don't know if it's ready to use.

modante, 2018-06-10:
I have talked with the neighbors about the creation of a Free WiFi Spot for the neighborhood but with limited internet access and their agree that would be great for the ones that have no other way to connect to internet for economical reasons. Then I am searching a way to do it that fits in my times, lack of technical knowledge and that doesn't breaks what is working fine 6 months ago.

Pitbull is still alpha and I don't need the captive portal.
QoS is too complex for my knowledge. Is not an option.

What about to use wshaper? I found this https://openwrt.org/docs/guide-user/network/wifi/guestwifi/guest-wlan (https://openwrt.org/docs/guide-user/network/wifi/guestwifi/guest-wlan) but is quite openwrt oriented.
Anybody could help me step by step to get it work on LiMe?
I have created a new AP but I am not sure about which kind of network interface should I create in order to attach the wireless AP. Also I am not sure if the wiki steps about DHCP would be the same to apply them in LiMe.
If I get it working, I can write a wiki (like the openwrt one) for LiMe explain it step by step.
Thanks in regards :-)

nicopace, 2018-06-20:
Pitbull does not have an initial plan for bandwith rating for now, but may have in the future.
It is also relevant what was said in the mailing list, shaping does not get you too far, cause the bandwidth was already used, you are just stopping it from getting to the client (and adding lots of buffers in the process).

Would be great if you create an article about it if you make it (and also, the user experience that you get out of it).

dangowrt, 2018-06-20:
> Pitbull does not have an initial plan for bandwith rating for now, but may have in the future.
> It is also relevant what was said in the mailing list, shaping does not get you too far, cause the [...]

TCP window scaling works perfectly well for incoming TCP traffic,
most UDP protocols which use significant amounts of bandwidth
(eg. BitTorrent uTP) can also very effectively control their
consumption. Large RTP streams (say: WebRTC video) are also
usually driven by some kind of bandwidth adaptation, so that also
shouldn't lead to a lot of packages being dropped at the wrong
end of the bottlekneck.

> Would be great if you create an article about it if you make it (and also, the user experience that you get out of it).

I suppose a few iptables rules to flag streams with TC handles and then
use an appropriate shaper hierachy to manage both incoming and outgoing
bandwidth shouldn't be very hard.

aparcar, 2018-06-21:
maybe related https://github.com/openwrt/packages/pull/5926

ilario, 2019-06-12:
Annoying @stalebot  is annoying.
This issue is not stale as it is closely related to these: https://github.com/libremesh/pirania/issues/3 and https://github.com/libremesh/pirania/issues/37

ilario, 2020-06-02:
Can we remove stale bot please? Its job can be done much better by any non-lazy developer.