libremesh/lime-packages #262: Advices for an optimal setup of the bands for Dual Band Routers

modante, 2017-12-11:
Hello.
We are a little neighborhood with 3 libremesh Dual Band Routers (TP-Link wdr3500, wdr3600 and wdr4100) that can work simultaneously in 2.4 and 5 GHz.
One of the routers have an internet connection to a WISP through the 2,4 GHz band using a NanoStaion Loco device and connected to the WAN port and this connection is shared through the rest of the routers of the mesh.
I saw that the default config of the LibreMesh clones the 3 WiFi Stations (LibreMesh.org, LibreMesh.org/LiMe-xxxxxx and LiMe) in all the radios availables on the device and I suppose that for the mesh (LiMe AdHoc) uses the best connection of them or both (please, confirm me the behavior).
I thought that in order to increase the performance (decreasing the interferences and dividing the bandwidth between the 2 radios) would be better to use the 5 GHz radios to run the mesh (LiMe), removing the APs from this band and exclusively the 2,4 Ghz to run the connection with the devices (mobiles, PCs, etc.), removing the mesh connection (LiMe) of this band.
What do you think is a better setup in this case?
Thank you and regards :-)

p4u, 2017-12-12:
> I saw that the default config of the LibreMesh clones the 3 WiFi Stations (LibreMesh.org, LibreMesh.org/LiMe-xxxxxx and LiMe) in all the radios availables on the device and I suppose that for the mesh (LiMe AdHoc) uses the best connection of them or both (please, confirm me the behavior).

It's a bit more complex than this but yes, the routing protocols decide for each packet which is the best next hope to a destination.

> I thought that in order to increase the performance (decreasing the interferences and dividing the bandwidth between the 2 radios) would be better to use the 5 GHz radios to run the mesh (LiMe), [...]

I would say it does not have a big impact because in general the 5G AP is not used and Mesh at 2.4GHz is usually not used either (5GHz is usually the best path). However disable it won't hurt.

What might have a big impact is to change the 2.4GHz AP channels, by default it's always 11 but you can set it to different channels such as 1 or 6. 5GHz radios must be always the same freq (to not breack the mesh cloud) but 2.4GHz can be different.

The APname (ssid like LibreMesh.org/LiMe-foo) should not have any negative impact either, but you can disable it to avoid seeing too much wifi networks.

In any case to tune the WiFi devices you might want to modify the configuration file /etc/config/lime and then run "lime-config && lime-apply".

Here an example of radio1 -> only-mesh and radio0 -> only-AP

```
config wifi 'radio1'
	list modes 'ieee80211s'
	option ieee80211s_mesh_id 'LiMe'
	option distance '1000'
	option ieee80211s_mesh_fwding '0'
	option htmode_5ghz 'HT40'
       option channel_5ghz '48'

config wifi 'radio0'
	list modes 'ap'
	option ap_ssid 'LibreMesh.org'
	option distance '1000'
	option apname_ssid 'LibreMesh.org/%H'
	option htmode_5ghz 'HT40'
	option channel_2ghz '11'
	option channel_5ghz '48'
```

It's important to keep the same "ap_ssid" on all your nodes cause it is used to calculate the Mesh Cloud Identifier (batman-adv VLAN tag).

Hope it helps.

modante, 2017-12-12:
> I would say it does not have a big impact because in general the 5G AP is not used and Mesh at 2.4GHz is usually not used either (5GHz is usually the best path). However disable it won't hurt.
Great!!! I didn't know that! Then maybe better to leave it by default :-)

> What might have a big impact is to change the 2.4GHz AP channels, by default it's always 11 but you can set it to different channels such as 1 or 6. 5GHz radios must be always the same freq (to not breack the mesh cloud) but 2.4GHz can be different.
I understad and agree that would be better this way.

It helped me a lot! Thank you very much :-)

I'll make some test and report the results.

ilario, 2019-10-04:
For some reason, lime-smart-wifi is not being included in the next release.
In my opinion we could even include something like this (I have no idea if it is already possible, @G10h4ck @spiccinini @nicopace @gmarcos87 ?) in the lime-defaults-factory:

```
config lime wifi
	list modes_2ghz 'ap'
	list modes_2ghz 'apname'
	list modes_2ghz 'ieee80211s' 
	list modes_5ghz 'ieee80211s' 
```

Which means: use 2.4 GHz radios both for AP and for mesh but keep 5 GHz radios just for mesh.
It is not "smart" but it is the generic configuration I would always suggest.

G10h4ck, 2019-10-04:
are you sure it is not possible already?

ilario, 2019-10-04:
Yesss, if you check the line I linked in wireless.lua, this one
https://github.com/libremesh/lime-packages/blob/18c8aa59ec14bf8a2e82f72b0b761194d525213c/packages/lime-system/files/usr/lib/lua/lime/wireless.lua#L101

you'll notice that the modes are gathered from a list named "modes" without the possibility to have modes_2ghz or modes_5ghz.
In my opinion it is worth to expand that line to use the two band-specific lists (as happens for distance and htmode) and the generic one as a fallback.

gmarcos87, 2019-10-04:
Currently it is possible to whit diferents radios.
You can have the 2ghz radios in adhoc and the 5ghz radios in mesh. I usually do it to "hunt" nodes that were misconfigured in our network.

ilario, 2019-10-04:
@gmarcos87 the proposal was to include such configuration in the lime-defaults-factory file, without using interface specific configuration, in order to have a more rational setup and higher performances.
I suppose that what you do, you do it with interface specific configuration?

ilario, 2019-11-26:
@spiccinini just made me realize that this is already possible thanks to #405 with:

```
config lime 'wifi'
    list modes 'ap_2ghz'
    list modes 'apname_2ghz'
    list modes 'ieee80211s'
```

Which means:
AP only on 2.4 GHz radios
APname only on 2.4 GHz radios
IEEE802.11s mesh both on 2.4 and 5 GHz radios

spiccinini, 2019-11-27:
With this change the AP with only 5ghz wont have ap/apname by default, that may be unexpected for the users. I prefer this to the defaults, but I would prefer directly disabling mesh in 2.4ghz but it also has drawbacks and real world use cases vary a lot. In my experience having 2.4ghz enabled in a real mesh for meshing has very bad performance beacause of hidden node problems and sometimes when there is no direct line of sight the only link that works (but very bad) is the 2.4ghz leading to a lot of problems.
So I am not sure what to do about this. I would say that we should recommend using dual boand routers only for a serious mesh networks and use single band routers for very specific tasks or deploys.

ilario, 2020-01-09:
In #669 I added to lime-example these suggestions:

```
list modes 'ap_2ghz'		# You can specify if the mode is applied to all radios or just some particular type, for example apname_2ghz and adhoc_5ghz to have an access point in 2ghz and the adhoc in 5ghz. These suffixes (_2ghz and _5ghz) work for all modes
list modes 'apname_2ghz'		# For networks where only dual band routers are used, AP and APname can be on the 2.4 GHz radio...
list modes 'ieee80211s_5ghz'	# ...while the 5 GHz radio can be reserved for the node-to-node connections
```

closed 2020-01-09 by ilario