libremesh/lime-packages #265: Strange behavior with WAN settings in the gateway router

modante, 2017-12-16:
Something happened yesterday while I was configuring a new router for the mesh.
I am not sure if that was the originator, but I selected in the Simple Config > Advanced > Internet Priorized Gateway (bmx6) > LiMe-WDR3500 and suddenly the internet connection stopped. I thought that was another fault of the WISP.
Revising two hours later I noticed that in the WAN connection of the router that serves like gateway, the protocol changed from PPPoE to DHCP. I put again the settings for PPPoE and it is working till now.
Anybody that noticed something similar?

modante, 2017-12-17:
More about this issue. Yesterday after rebooting all the routers of the mesh I coudn't get internet access in any of them except the one connected to WAN. Watching the BMX6 "Gateways tunnel announcements" in the rest of the routers, none of them had the inet4 tunnel (internet network) listed (there were clouds, publicv6 and nodes).
Then I changed the WAN protocol from PPPoE to DHCP and again switched to PPPoE, put the login/pass of the WISP, connected and then announced to the mesh and got internet connection in all of the routers.
Again this morning I turned on all the routers that yesterday I switched off cause the storm and I could see the same behavior and had to swithc the WAN protocol in order to make it work in the rest of the routers.
Anybody knows why?
Thanks and regards :-)

modante, 2017-12-24:
Definitively now I can recreate the issue. When the router working as gateway (with the WAN connection) is rebooted, this router automatically connects to the WAN but doesn't announce the internet gateway tunnel to the rest of the mesh. Simply reconnecting the WAN connection (in the interfaces page), announces the internet gateway tunnel to the rest of the mesh. But if you don't manually reconnect the WAN connection, the rest of the mesh have no internet connectivity.
Any suggestion?
Thanks and regards :-)

modante, 2017-12-28:
Nothing?
At least anybody could help me with a script in order to disconnect an reconnect the WAN connection 1 minute after boot to avoid to do that manually in case I am not available at home in case of accidental switch off the router with the WAN connection?
Thanks and regards :-)

dangowrt, 2017-12-28:
On Thu, Dec 28, 2017 at 01:44:45PM +0000, Modante wrote:
> Nothing?
> At least anybody could help me with a script in order to disconnect an reconnect the WAN connection 1 minute after boot to avoid to do that manually in case I am not available at home in case of accidental switch off the router with the WAN connection?
> Thanks and regards :-) 

The behaviour you describe is indeed a bit weird. Which type of WAN
connection do you use? After all it sounds like a problem with
watchping...

p4u, 2017-12-28:
Just as a temporay workaround, you can use **smonit** to monitor and execute actions. 

Here an example hook thay you might copy to /etc/smonit/

It checks ping to 192.168.1.1, if does not work it disables and enables the WAN interface.

Smonit is executed by default in cron, but you can run it manually typing "smonit".

```
#!/bin/sh

hook_name="check_gw"
gw_ip="192.168.1.1"

function hook_is_installed() {
  installed=yes
}   

function hook_check() {
  status=error
  ping -c5 $gw_ip 2>&1 1>/dev/null && status=ok 2>&1 1>/dev/null 
}

function hook_quickfix() {
  log="Gateway $gw_ip  unrechable"
  ifdown wan
  sleep 1
  ifup wan
}

function hook_longfix() {
  log="What should I do? Reboot?"
  #reboot
}

```

ilario, 2017-12-29:
> Which type of WAN connection do you use?

@dangowrt seems that @modante is using PPPoE, as reported also here #260

modante, 2017-12-30:
@p4u Thank you for the smonit script. I will test it. Is it 192.168.1.1 the IP that should verify? Not better an internet public IP?
And yes, the WAN connection is a Nanostation connected to a WISP with PPPoE protocol.
Thanks and Regards :-)

modante, 2017-12-31:
I have solved it at "my way" ;-P
I think the problem is that watchping runs when booting and detects no-internet and run /etc/watchping/wan-fail.d/bmx6-gw that removes inet tunnel:
```
#!/bin/sh
logger -t bmx6-auto-gw "Searching Internet access into the mesh"
bmx6 -c tunIn -inet4
bmx6 -c tunOut inet4 /network 0.0.0.0/0 /maxPrefixLen 0
```
Then, what I did was to delay the execution of the watchping script adding sleep 10 at the beginning of /etc/init.d/watchping:
```
#!/bin/sh /etc/rc.common
# Copyright (C) 2012 Gui Iribarren
# Copyright (C) 2017 Daniel Golle
# This is free software, licensed under the GNU General Public License v3.

**sleep 10**
START=99
USE_PROCD=1
PROG=/usr/bin/watchping
[...]
```
Cause it takes a time to connect to WAN via PPPoE, when boots, watchping it takes 10 secs to test internet connectivity and if it detects, announce the gateway tunnel. Without this delay always detects no-internet and removed the tunnels.
What do you thing? I am proud of my first fix!!! hehehehe :-D
Thanks and regards.

ilario, 2018-01-01:
Uh, I always thought it was running in cron!
Instead seems that has its own [loop time parameter](/libremesh/lime-packages/blob/develop/packages/watchping/files/usr/bin/watchping#L6) `pinginterval`.
So maybe fails to start 'cause that interface is not up when watchping is called?
@modante can you check if watchping is present in `ps` output when the `sleep 10` is not present?
Something like this should appear:
```
ps | grep watch
 1806 root      1368 S    {watchping} /bin/sh /usr/bin/watchping eth0.2 120 8.8.8.8 20 wan
23653 root      1356 S    grep watch
```
@p4u @G10h4ck @aparcar @altergui @nicopace What's an elegant fix? The `sleep 10` @modante added? Inserting that in cron?

dangowrt, 2018-01-01:
On Mon, Jan 01, 2018 at 03:46:08PM +0000, Ilario Gelmetti wrote:
> Uh, I always thought it was running in cron!
> Instead seems that has its own [loop time parameter](/libremesh/lime-packages/blob/develop/packages/watchping/files/usr/bin/watchping#L6) `pinginterval`. [...]

Neither cron nor an explicit sleep is the right solution to this
problem, please don't do either of it as it adds an imcomplete
work-around for a problem which should be addressed in watchping's
init-script by adding the appropriate reload trigger and also
making sure it only starts once the corresponding interface is up.
There are many examples for this and I tried implementing it for
watchping:
https://github.com/libremesh/lime-packages/blob/develop/packages/watchping/files/etc/init.d/watchping#L100

Maybe the interfaces are not collected properly from your config?
Can you add some 'echo' calls into the init script to see what's going
on?

modante, 2018-01-01:
> So maybe fails to start 'cause that interface is not up when watchping is called?

I am pretty sure. But from a novice point of view ;-)
I think that when watchping runs for fist time after booting, the WAN interface is still connecting, there are no pongs and then removes the inet tunnel. What I cannot understand is why doesn't announce the inet tunnel automatically when WAN finishes to connect. Without the Sleep 10, when the gateway router boots never announces the inet tunnel except if I manually reconnect WAN, then it announces to the rest of the mesh in a few seconds.

> @modante can you check if watchping is present in ps output when the sleep 10 is not present?

I ran ps before adding the sleep and yes it was running. It is enabled in the init scripts.
Probably is not a solution, but till is found I'll use like workaround.

> Maybe the interfaces are not collected properly from your config? Can you add some 'echo' calls into the init script to see what's going on?
Could you explain me better what should I do? Sorry but I don't understand right.

dangowrt, 2018-01-01:
I had another look at it and think that the problem can be resolved by
changing line 6 'START=99' in
https://github.com/libremesh/lime-packages/blob/develop/packages/watchping/files/etc/init.d/watchping#L6
to be 'START=19' (ie. just before network) instead, so that triggers
are defined before the interface comes up.

modante, 2018-01-01:
Doesn't work for me. Removed sleep, changed start to 19 and after rebooting the scrip was disabled. I enabled it but didn't work till I ran it. Rebooted again, now is enabled but still doesn't annouce the inet tunnel if I don't run it manually.
By now I come back to start 99 and sleep 10 till a better fix.
Thanks and regards :-)

ilario, 2019-10-04:
Up to now, seems that watchping will not be included in the next 19.something release, so I would close this.
Out of curiosity, wouldn't watchping eventually realise that the connection appeared?

closed 2019-10-04 by ilario