libremesh/lime-packages #268: Implement wpad-mesh-nocrypt package

p4u, 2017-12-17:
Using wpa_supplicant over a mesh WiFi interface gives the option to use channel auto selection and DFS restricted channels.

To this end, a modified copy of wpad-mesh might be created as a libremesh package removing the crypto related dependencies.

aparcar, 2018-06-01:
@dangowet works on a smaller version called wpad-mesh-wolftls. We should check the size impact and may move.

aparcar, 2018-12-09:
@p4u please see the package wpad-mesh-wolfssl, wolfssl should already be much smaller then the openssl libary. Can we close this?

ilario, 2019-10-04:
I think that after @aparcar comment, we can move the discussion on whether include wpad-mesh-wolfssl by default, see #613. Closing this.

closed 2019-10-04 by ilario