libremesh/lime-packages #272: Onion Omega 2+ Kernel Panics

kallisti5, 2017-12-27:
Loading libremesh onto the Onion Omega 2 devices works, however they are really unstable.

Onion Omega 2+
```
# [ 1217.789975] CPU 0 Unable to handle kernel paging request at virtual address 074031ac, epc == 802aceb4, ra == 802aced0
[ 1217.800815] Oops[#1]:
[ 1217.803129] CPU: 0 PID: 2634 Comm: dnsmasq Tainted: G        W       4.4.71 #0
[ 1217.810454] task: 86c95928 ti: 87524000 task.ti: 87524000
[ 1217.815927] $ 0   : 00000000 770509ac 00000001 07403000
[ 1217.821245] $ 4   : 802acd98 00000000 87525c4c 00000002
[ 1217.826557] $ 8   : 8774b17c 00002000 00000000 02000000
[ 1217.831868] $12   : 7ff76aa8 77050990 00000000 770fc2c0
[ 1217.837179] $16   : 07402fcc 8037cff0 8037d080 87525dc0
[ 1217.842489] $20   : 00000000 00000000 00000000 00000025
[ 1217.847796] $24   : 00000000 802c0ef8                  
[ 1217.853109] $28   : 87524000 87525c30 87525da0 802aced0
[ 1217.858421] Hi    : 00000002
[ 1217.861337] Lo    : 00000000
[ 1217.864260] epc   : 802aceb4 0x802aceb4
[ 1217.868148] ra    : 802aced0 0x802aced0
[ 1217.872033] Status: 1100e403 KERNEL EXL IE 
[ 1217.876288] Cause : 00800008 (ExcCode 02)
[ 1217.880350] BadVA : 074031ac
[ 1217.883269] PrId  : 00019655 (MIPS 24KEc)
[ 1217.887329] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_fh
[ 1217.978621] Process dnsmasq (pid: 2634, threadinfo=87524000, task=86c95928, tls=770fdd48)
[ 1217.986908] Stack : 00000000 00000002 00000003 80228e84 00000000 8037cff0 8037d4e4 87525dc0
          00000000 0000000e 00000001 00000000 00000005 00020021 86c50b8c 0000003d
          ffffff82 00000025 00000002 000e0001 877d6f00 00000005 fffffff2 00000000
          00000000 87525cfc 87525da0 8774af00 8037cff0 802a4d14 800ba380 00000100
          00000200 00000018 87525dd0 00000008 00000000 87525da0 80380000 8774af00
          ...
[ 1218.023019] Call Trace:[<80228e84>] 0x80228e84
[ 1218.027582] [<802a4d14>] 0x802a4d14
[ 1218.031117] [<800ba380>] 0x800ba380
[ 1218.034682] [<802a4e00>] 0x802a4e00
[ 1218.038232] [<802a4ebc>] 0x802a4ebc
[ 1218.041797] [<802c1488>] 0x802c1488
[ 1218.045360] [<801fdf3c>] 0x801fdf3c
[ 1218.048932] [<8026d3b0>] 0x8026d3b0
[ 1218.052468] [<800ba380>] 0x800ba380
[ 1218.056003] [<80246a00>] 0x80246a00
[ 1218.059560] [<8026ecd8>] 0x8026ecd8
[ 1218.063130] [<801f36a0>] 0x801f36a0
[ 1218.066675] [<801f55d4>] 0x801f55d4
[ 1218.070210] [<802a2c0c>] 0x802a2c0c
[ 1218.073745] [<801f4200>] 0x801f4200
[ 1218.077306] [<800bc3a8>] 0x800bc3a8
[ 1218.080851] [<801f31c8>] 0x801f31c8
[ 1218.084386] [<800c2a64>] 0x800c2a64
[ 1218.087951] [<800b8700>] 0x800b8700
[ 1218.091495] [<8000682c>] 0x8000682c
[ 1218.095034] [<800069e0>] 0x800069e0
[ 1218.098569] [<800a6af4>] 0x800a6af4
[ 1218.102107] 
[ 1218.103614] 
Code: 02202021  1000000e  24030018 <8e0601e0> 10c00005  27a70030  afa20010  27a5001c  0c0ab2f9 
[ 1218.113756] ---[ end trace 2beb8dc92ed9d410 ]---
[ 1218.122704] Fatal exception: panic in 5 seconds
[ 1223.132075] Kernel panic - not syncing: Fatal exception
```

Boot info:
```

   ____       _             ____
  / __ \___  (_)__  ___    / __ \__ _  ___ ___ ____ _
 / /_/ / _ \/ / _ \/ _ \  / /_/ /  ' \/ -_) _ `/ _ `/
 \____/_//_/_/\___/_//_/  \____/_/_/_/\__/\_, /\_,_/
 W H A T  W I L L  Y O U  I N V E N T ? /___/"

Board: Onion Omega2 APSoC DRAM:  128 MB
relocate_code Pointer at: 87f60000
flash manufacture id: c2, device id 20 19
find flash: MX25L25635E
*** Warning - bad CRC, using default environment

============================================ 
Onion Omega2 UBoot Version: 4.3.0.3
-------------------------------------------- 
ASIC 7628_MP (Port5<->None)
DRAM component: 1024 Mbits DDR, width 16
DRAM bus: 16 bit
Total memory: 128 MBytes
Flash component: SPI Flash
Date:Oct 18 2016  Time:17:29:05
============================================ 
icache: sets:512, ways:4, linesz:32 ,total:65536
dcache: sets:256, ways:4, linesz:32 ,total:32768 
CPU freq = 575 MHZ
Estimated memory size = 128 Mbytes
Resetting MT7628 PHY.
Initializing MT7688 GPIO system.

**************************************
* Hold Reset button for more options *
**************************************

Boot Linux from Flash NO RESET PRESSED.
## Booting image at bc050000 ...
   Image Name:   MIPS LEDE Linux-4.4.71
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1229457 Bytes =  1.2 MB
   Load Address: 80000000
   Entry Point:  80000000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK
No initrd
## Transferring control to Linux (at address 80000000) ...
## Giving linux memsize in MB, 128

Starting kernel ...

[    0.000000] Linux version 4.4.71 (buildbot@builds-02.infra.lede-project.org) (gcc version 5.4.0 (LEDE GCC 5.4.0 r3104-41de9a2) ) #0 Wed Jun 7 19:24:41 2017
```

Hopefully can figure out a solution... The Onion Omega 2 devices are perfect mesh nodes in a low-cost ($12 USD each) mesh network.

kallisti5, 2017-12-27:
Also shows a bunch of random kernel oops at boot.:
```
[   26.842230] ------------[ cut here ]------------
[   26.846979] WARNING: CPU: 0 PID: 6 at /build/lede-17.01/slaves/phase1/ramips_mt7688/build/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7688/mt76-2017-01-31-3c8ca)
[   26.867899] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_fh
[   26.959267] CPU: 0 PID: 6 Comm: kworker/u2:0 Not tainted 4.4.71 #0
[   26.965572] Workqueue: phy0 0x8755d08c [mt7603e@87558000+0x6840]
[   26.971663] Stack : 87c03410 87c1eb98 00000088 8004aa60 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[   27.007813] Call Trace:[<8004aa60>] 0x8004aa60
[   27.012374] [<800144bc>] 0x800144bc
[   27.015911] [<800144bc>] 0x800144bc
[   27.019446] [<8002506c>] 0x8002506c
[   27.023015] [<8755d294>] 0x8755d294 [mt7603e@87558000+0x6840]
[   27.028875] [<80025124>] 0x80025124
[   27.032425] [<800431f4>] 0x800431f4
[   27.035989] [<8755d294>] 0x8755d294 [mt7603e@87558000+0x6840]
[   27.041843] [<80036920>] 0x80036920
[   27.045466] [<80037734>] 0x80037734
[   27.049017] [<80037484>] 0x80037484
[   27.052581] [<80037484>] 0x80037484
[   27.056117] [<8003baec>] 0x8003baec
[   27.059690] [<8003ba14>] 0x8003ba14
[   27.063255] [<80004478>] 0x80004478
[   27.066800] 
[   27.068310] ---[ end trace 4b2bf5c28b116a10 ]---
[   27.075460] br-lan: port 3(wlan0-ap) entered forwarding state
[   27.081351] br-lan: port 4(wlan0-apname) entered forwarding state
[   27.892280] ip6_tunnel: bmxmain xmit: Local address not yet configured!
[   28.272273] ip6_tunnel: bmxmain xmit: Local address not yet configured!
[   28.662289] ip6_tunnel: bmxmain xmit: Local address not yet configured!
[   34.062101] ------------[ cut here ]------------
[   34.066849] WARNING: CPU: 0 PID: 10 at /build/lede-17.01/slaves/phase1/ramips_mt7688/build/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7688/mt76-2017-01-31-3c8c)
[   34.087840] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_fh
[   34.179304] CPU: 0 PID: 10 Comm: kworker/u2:2 Tainted: G        W       4.4.71 #0
[   34.186924] Workqueue: phy0 0x8755d08c [mt7603e@87558000+0x6840]
[   34.193023] Stack : 87c03410 87caea18 00000088 8004aa60 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[   34.229180] Call Trace:[<8004aa60>] 0x8004aa60
[   34.233751] [<800144bc>] 0x800144bc
[   34.237289] [<800144bc>] 0x800144bc
[   34.240828] [<8002506c>] 0x8002506c
[   34.244397] [<8755d17c>] 0x8755d17c [mt7603e@87558000+0x6840]
[   34.250259] [<80025124>] 0x80025124
[   34.253814] [<80042d28>] 0x80042d28
[   34.257348] [<800431f4>] 0x800431f4
[   34.260915] [<8755d17c>] 0x8755d17c [mt7603e@87558000+0x6840]
[   34.266789] [<80036920>] 0x80036920
[   34.270399] [<80037734>] 0x80037734
[   34.273962] [<80037484>] 0x80037484
[   34.277513] [<80037484>] 0x80037484
[   34.281050] [<8003baec>] 0x8003baec
[   34.284638] [<8003ba14>] 0x8003ba14
[   34.288189] [<80004478>] 0x80004478
[   34.291727] 
[   34.293250] ---[ end trace 4b2bf5c28b116a11 ]---
[   45.162121] MCU message 40 (seq 4) timed out
[   63.826606] random: nonblocking pool is initialized
[   83.655297] ------------[ cut here ]------------
[   83.660043] WARNING: CPU: 0 PID: 0 at /build/lede-17.01/slaves/phase1/ramips_mt7688/build/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7688/mt76-2017-01-31-3c8ca)
[   83.680846] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_fh
[   83.772235] CPU: 0 PID: 0 Comm: swapper Tainted: G        W       4.4.71 #0
[... 27 lines cut ...]
[   83.996262] CPU: 0 PID: 0 Comm: swapper Tainted: G        W       4.4.71 #0
[   84.003333] Stack : 8032087c 00000000 00000001 80370000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[   84.039472] Call Trace:[<800144bc>] 0x800144bc
[   84.044009] [<800144bc>] 0x800144bc
[   84.047544] [<8002506c>] 0x8002506c
[   84.051107] [<876a591c>] 0x876a591c [mac80211@87680000+0x5ff40]
[   84.057139] [<80025124>] 0x80025124
[   84.060681] [<801fdf3c>] 0x801fdf3c
[   84.064261] [<876a591c>] 0x876a591c [mac80211@87680000+0x5ff40]
[   84.070277] [<8007a3cc>] 0x8007a3cc
[   84.073905] [<875f98ec>] 0x875f98ec [mt76@875f8000+0x3c00]
[   84.079531] [<875f90e8>] 0x875f90e8 [mt76@875f8000+0x3c00]
[   84.085133] [<8020c504>] 0x8020c504
[   84.088702] [<800275ec>] 0x800275ec
[   84.092259] [<8004f1a0>] 0x8004f1a0
[   84.095802] [<8004b6b8>] 0x8004b6b8
[   84.099355] [<800047f4>] 0x800047f4
[   84.102899] 
[   84.104410] ---[ end trace 4b2bf5c28b116a13 ]---
```

kallisti5, 2017-12-27:
Looks like https://github.com/openwrt/mt76/issues/121 ?

p4u, 2017-12-28:
> Looks like openwrt/mt76#121 <https://github.com/openwrt/mt76/issues/121> ?

Yes, this mediatek chip is still in early adoption, so it is quite
unstable. This is not a libremesh thing but a LEDE/OpenWRT.

kallisti5, 2017-12-30:
The latest libremesh snapshot release (when you click advanced in the build system) actually is stable on the Onion Omega 2 and Onion Omega 2+

closed 2017-12-30 by kallisti5