libremesh/lime-packages #273: Suddenly losts of internet connection

modante, 2017-12-29:
Setup: WDR3500 (with WAN connection) + WDR4300 + WDR3600

Last night I left all the mesh working with internet connectivity in all the routers.
This morning I found that in one of the routers there is not internet connection in the WDR4300 but there was in the router acting like gateway (WDR3500). I tried to disconnect and connect the WAN connection in the router acting like gateway in order to refresh the announce through BMX6 but didn't worked. In the BMX6 tunnels, the internet connection was listed and all looked like it should work.
I could only make it work rebooting.
After a while, the same happened to the WDR3600: Suddenly the internet connectivity was lost but it worked in the others 2 routers.
Again, all it looked like it should work but only get to return the internet connectivity rebooting.

This is the system log of the WDR3600 in the range of time when it loosed the internet connection:
`Fri Dec 29 08:50:00 2017 cron.info crond[873]: USER root pid 3735 cmd /usr/sbin/smonit
Fri Dec 29 08:50:00 2017 cron.info crond[873]: USER root pid 3736 cmd ( for file in /etc/alfred/* ; do [ -x $file ] && $file ; done )
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq[1851]: read /etc/hosts - 4 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/dnsmasq-lease-share - 7 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/d-d-h_64:66:b3:fa:a5:94 - 2 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/bmx6 - 12 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq-dhcp[1851]: read /etc/ethers - 0 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq-dhcp[1851]: read /tmp/dhcp.hosts_remote
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq[1851]: read /etc/hosts - 4 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/dnsmasq-lease-share - 7 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/d-d-h_64:66:b3:fa:a5:94 - 2 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/bmx6 - 12 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq-dhcp[1851]: read /etc/ethers - 0 addresses
Fri Dec 29 08:50:00 2017 daemon.info dnsmasq-dhcp[1851]: read /tmp/dhcp.hosts_remote
Fri Dec 29 08:50:02 2017 daemon.info hostapd: wlan0-ap: STA b4:6d:83:7c:5a:88 IEEE 802.11: authenticated
Fri Dec 29 08:50:02 2017 daemon.info hostapd: wlan0-ap: STA b4:6d:83:7c:5a:88 IEEE 802.11: associated (aid 3)
Fri Dec 29 08:50:02 2017 daemon.notice hostapd: wlan0-ap: AP-STA-CONNECTED b4:6d:83:7c:5a:88
Fri Dec 29 08:50:02 2017 daemon.info hostapd: wlan0-ap: STA b4:6d:83:7c:5a:88 WPA: pairwise key handshake completed (RSN)
Fri Dec 29 08:50:04 2017 daemon.info dnsmasq-dhcp[1851]: RTR-SOLICIT(anygw) 
Fri Dec 29 08:50:04 2017 daemon.info dnsmasq-dhcp[1851]: RTR-ADVERT(anygw) 2a00:1508:a0d:fe00::
Fri Dec 29 08:50:04 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3600_anygw' to 'LiMe-WDR3500_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 08:50:04 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3500_anygw' to 'LiMe-WDR4300_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 08:50:05 2017 daemon.info dnsmasq-dhcp[1851]: DHCPDISCOVER(anygw) 10.13.236.98 b4:6d:83:7c:5a:88 
Fri Dec 29 08:50:05 2017 daemon.info dnsmasq-dhcp[1851]: DHCPOFFER(anygw) 10.13.236.98 b4:6d:83:7c:5a:88 
Fri Dec 29 08:50:05 2017 daemon.info dnsmasq-dhcp[1851]: DHCPDISCOVER(anygw) 10.13.236.98 b4:6d:83:7c:5a:88 
Fri Dec 29 08:50:05 2017 daemon.info dnsmasq-dhcp[1851]: DHCPOFFER(anygw) 10.13.236.98 b4:6d:83:7c:5a:88 
Fri Dec 29 08:50:05 2017 daemon.info dnsmasq-dhcp[1851]: DHCPREQUEST(anygw) 10.13.236.98 b4:6d:83:7c:5a:88 
Fri Dec 29 08:50:05 2017 daemon.info dnsmasq-dhcp[1851]: DHCPACK(anygw) 10.13.236.98 b4:6d:83:7c:5a:88 acire-Lenovo-E31-70
Fri Dec 29 08:50:10 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3600_anygw' to 'LiMe-WDR3500_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 08:50:10 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3500_anygw' to 'LiMe-WDR4300_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 08:50:15 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3600_anygw' to 'LiMe-WDR3500_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 08:50:15 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3500_anygw' to 'LiMe-WDR4300_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 08:50:21 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3600_anygw' to 'LiMe-WDR3500_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 08:50:21 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3500_anygw' to 'LiMe-WDR4300_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 08:50:26 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3600_anygw' to 'LiMe-WDR3500_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 08:50:26 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3500_anygw' to 'LiMe-WDR4300_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 08:50:31 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3600_anygw' to 'LiMe-WDR3500_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 08:50:31 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3500_anygw' to 'LiMe-WDR4300_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 08:52:00 2017 cron.info crond[873]: USER root pid 3918 cmd /usr/bin/bmx6hosts_dnsmasq_update
Fri Dec 29 08:52:39 2017 daemon.notice hostapd: wlan0-ap: AP-STA-DISCONNECTED 20:82:c0:e6:dd:5f
Fri Dec 29 08:54:03 2017 daemon.info hostapd: wlan0-ap: STA 20:82:c0:e6:dd:5f IEEE 802.11: authenticated
Fri Dec 29 08:54:03 2017 daemon.info hostapd: wlan0-ap: STA 20:82:c0:e6:dd:5f IEEE 802.11: associated (aid 1)
Fri Dec 29 08:54:03 2017 daemon.notice hostapd: wlan0-ap: AP-STA-CONNECTED 20:82:c0:e6:dd:5f
Fri Dec 29 08:54:03 2017 daemon.info hostapd: wlan0-ap: STA 20:82:c0:e6:dd:5f WPA: pairwise key handshake completed (RSN)
Fri Dec 29 08:54:03 2017 daemon.info dnsmasq-dhcp[1851]: DHCPDISCOVER(anygw) 20:82:c0:e6:dd:5f 
Fri Dec 29 08:54:03 2017 daemon.info dnsmasq-dhcp[1851]: DHCPOFFER(anygw) 10.13.234.191 20:82:c0:e6:dd:5f 
Fri Dec 29 08:54:03 2017 daemon.info dnsmasq-dhcp[1851]: DHCPREQUEST(anygw) 10.13.234.191 20:82:c0:e6:dd:5f 
Fri Dec 29 08:54:03 2017 daemon.info dnsmasq-dhcp[1851]: DHCPACK(anygw) 10.13.234.191 20:82:c0:e6:dd:5f android-fa5690fd432697a5
Fri Dec 29 08:54:04 2017 daemon.info dnsmasq-dhcp[1851]: RTR-SOLICIT(anygw) 20:82:c0:e6:dd:5f
Fri Dec 29 08:54:04 2017 daemon.info dnsmasq-dhcp[1851]: RTR-ADVERT(anygw) 2a00:1508:a0d:fe00::
Fri Dec 29 08:55:00 2017 cron.info crond[873]: USER root pid 3966 cmd /usr/sbin/smonit
Fri Dec 29 08:55:00 2017 cron.info crond[873]: USER root pid 3967 cmd ( for file in /etc/alfred/* ; do [ -x $file ] && $file ; done )
Fri Dec 29 08:55:00 2017 daemon.info dnsmasq[1851]: read /etc/hosts - 4 addresses
Fri Dec 29 08:55:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/dnsmasq-lease-share - 7 addresses
Fri Dec 29 08:55:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/d-d-h_64:66:b3:fa:a5:94 - 2 addresses
Fri Dec 29 08:55:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
Fri Dec 29 08:55:00 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/bmx6 - 12 addresses
[... 39 lines cut ...]
Fri Dec 29 09:00:01 2017 daemon.info dnsmasq[1851]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
Fri Dec 29 09:00:01 2017 daemon.info dnsmasq-dhcp[1851]: read /etc/ethers - 0 addresses
Fri Dec 29 09:00:01 2017 daemon.info dnsmasq-dhcp[1851]: read /tmp/dhcp.hosts_remote
Fri Dec 29 09:00:14 2017 daemon.info dnsmasq-dhcp[1851]: RTR-SOLICIT(anygw) 
Fri Dec 29 09:00:14 2017 daemon.info dnsmasq-dhcp[1851]: RTR-ADVERT(anygw) 2a00:1508:a0d:fe00::
Fri Dec 29 09:01:09 2017 daemon.notice hostapd: wlan0-ap: AP-STA-DISCONNECTED b4:6d:83:73:96:32
Fri Dec 29 09:01:09 2017 daemon.info hostapd: wlan0-apname: STA b4:6d:83:73:96:32 IEEE 802.11: authenticated
[... cut ...]
Fri Dec 29 09:02:51 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3500_anygw' to 'LiMe-WDR4300_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 09:02:57 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3600_anygw' to 'LiMe-WDR3500_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 09:02:57 2017 daemon.err uhttpd[927]: Warning - mac already known (changing name from 'LiMe-WDR3500_anygw' to 'LiMe-WDR4300_anygw'): aa:aa:aa:0d:fe:aa
Fri Dec 29 09:03:00 2017 cron.info crond[873]: USER root pid 4282 cmd /usr/sbin/libremap-agent`

Anybody knows why could happen this issue?

Thanks and regards :-)

modante, 2017-12-29:
Again, I lost the internet connectivity and this is the system log. I hope will be usefull :-)

> Fri Dec 29 10:17:01 2017 daemon.notice hostapd: wlan0-ap: AP-STA-DISCONNECTED 20:82:c0:39:f8:0b
> Fri Dec 29 10:17:03 2017 daemon.info hostapd: wlan0-ap: STA 20:82:c0:39:f8:0b IEEE 802.11: authenticated
> Fri Dec 29 10:17:03 2017 daemon.info hostapd: wlan0-ap: STA 20:82:c0:39:f8:0b IEEE 802.11: associated (aid 1)
> Fri Dec 29 10:17:03 2017 daemon.notice hostapd: wlan0-ap: AP-STA-CONNECTED 20:82:c0:39:f8:0b
> Fri Dec 29 10:17:03 2017 daemon.info hostapd: wlan0-ap: STA 20:82:c0:39:f8:0b WPA: pairwise key handshake completed (RSN)
> Fri Dec 29 10:17:03 2017 daemon.info dnsmasq-dhcp[1883]: DHCPDISCOVER(anygw) 20:82:c0:39:f8:0b 
> Fri Dec 29 10:17:03 2017 daemon.info dnsmasq-dhcp[1883]: DHCPOFFER(anygw) 10.13.119.161 20:82:c0:39:f8:0b 
> Fri Dec 29 10:17:03 2017 daemon.info dnsmasq-dhcp[1883]: DHCPREQUEST(anygw) 10.13.119.161 20:82:c0:39:f8:0b 
> Fri Dec 29 10:17:03 2017 daemon.info dnsmasq-dhcp[1883]: DHCPACK(anygw) 10.13.119.161 20:82:c0:39:f8:0b android-4de80e4b25755a20
> Fri Dec 29 10:17:04 2017 daemon.info dnsmasq-dhcp[1883]: RTR-SOLICIT(anygw) 20:82:c0:39:f8:0b
> Fri Dec 29 10:17:04 2017 daemon.info dnsmasq-dhcp[1883]: RTR-ADVERT(anygw) 2a00:1508:a0d:fe00::
> Fri Dec 29 10:18:31 2017 daemon.info hostapd: wlan0-ap: STA 20:82:c0:39:f8:0b WPA: group key handshake completed (RSN)
> Fri Dec 29 10:18:34 2017 daemon.info hostapd: wlan0-apname: STA b4:6d:83:73:96:32 WPA: group key handshake completed (RSN)
> Fri Dec 29 10:20:00 2017 cron.info crond[1637]: USER root pid 10292 cmd /usr/bin/bmx6hosts_dnsmasq_update
> Fri Dec 29 10:20:00 2017 cron.info crond[1637]: USER root pid 10293 cmd /usr/sbin/smonit
> Fri Dec 29 10:20:00 2017 cron.info crond[1637]: USER root pid 10294 cmd ( for file in /etc/alfred/* ; do [ -x $file ] && $file ; done )
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq[1883]: read /etc/hosts - 4 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dnsmasq-lease-share - 5 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/bmx6 - 0 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_64:70:02:ed:f9:5e - 2 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq-dhcp[1883]: read /etc/ethers - 0 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq-dhcp[1883]: read /tmp/dhcp.hosts_remote
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq[1883]: read /etc/hosts - 4 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dnsmasq-lease-share - 5 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/bmx6 - 1 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_64:70:02:ed:f9:5e - 2 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq-dhcp[1883]: read /etc/ethers - 0 addresses
> Fri Dec 29 10:20:01 2017 daemon.info dnsmasq-dhcp[1883]: read /tmp/dhcp.hosts_remote
> Fri Dec 29 10:23:41 2017 daemon.info dnsmasq-dhcp[1883]: DHCPREQUEST(anygw) 10.13.98.177 b4:6d:83:73:96:32 
> Fri Dec 29 10:23:41 2017 daemon.info dnsmasq-dhcp[1883]: DHCPACK(anygw) 10.13.98.177 b4:6d:83:73:96:32 doctorsky-Lenovo-E31-70
> Fri Dec 29 10:24:00 2017 cron.info crond[1637]: USER root pid 10707 cmd /usr/bin/bmx6hosts_dnsmasq_update
> Fri Dec 29 10:25:00 2017 cron.info crond[1637]: USER root pid 10798 cmd /usr/sbin/smonit
> Fri Dec 29 10:25:00 2017 cron.info crond[1637]: USER root pid 10799 cmd ( for file in /etc/alfred/* ; do [ -x $file ] && $file ; done )
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq[1883]: read /etc/hosts - 4 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dnsmasq-lease-share - 5 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/bmx6 - 12 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_64:70:02:ed:f9:5e - 2 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq-dhcp[1883]: read /etc/ethers - 0 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq-dhcp[1883]: read /tmp/dhcp.hosts_remote
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq[1883]: read /etc/hosts - 4 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dnsmasq-lease-share - 5 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/bmx6 - 12 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_64:70:02:ed:f9:5e - 2 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq-dhcp[1883]: read /etc/ethers - 0 addresses
> Fri Dec 29 10:25:01 2017 daemon.info dnsmasq-dhcp[1883]: read /tmp/dhcp.hosts_remote

modante, 2017-12-29:
And suddenly, the internet connectivity returned without doing anything:

> Fri Dec 29 10:26:10 2017 daemon.info dnsmasq-dhcp[1883]: RTR-SOLICIT(anygw) 
> Fri Dec 29 10:26:10 2017 daemon.info dnsmasq-dhcp[1883]: RTR-ADVERT(anygw) 2a00:1508:a0d:fe00::
> Fri Dec 29 10:27:46 2017 daemon.info dnsmasq-dhcp[1883]: RTR-SOLICIT(anygw) 
> Fri Dec 29 10:27:46 2017 daemon.info dnsmasq-dhcp[1883]: RTR-ADVERT(anygw) 2a00:1508:a0d:fe00::
> Fri Dec 29 10:28:00 2017 cron.info crond[1637]: USER root pid 10911 cmd /usr/bin/bmx6hosts_dnsmasq_update
> Fri Dec 29 10:28:31 2017 daemon.info hostapd: wlan0-ap: STA 20:82:c0:39:f8:0b WPA: group key handshake completed (RSN)
> Fri Dec 29 10:28:33 2017 daemon.info hostapd: wlan0-apname: STA b4:6d:83:73:96:32 IEEE 802.11: authenticated
> Fri Dec 29 10:28:33 2017 daemon.info hostapd: wlan0-apname: STA b4:6d:83:73:96:32 IEEE 802.11: associated (aid 1)
> Fri Dec 29 10:28:33 2017 daemon.notice hostapd: wlan0-apname: AP-STA-CONNECTED b4:6d:83:73:96:32
> Fri Dec 29 10:28:33 2017 daemon.info hostapd: wlan0-apname: STA b4:6d:83:73:96:32 WPA: pairwise key handshake completed (RSN)
> Fri Dec 29 10:28:33 2017 daemon.info dnsmasq-dhcp[1883]: DHCPREQUEST(anygw) 10.13.98.177 b4:6d:83:73:96:32 
> Fri Dec 29 10:28:33 2017 daemon.info dnsmasq-dhcp[1883]: DHCPACK(anygw) 10.13.98.177 b4:6d:83:73:96:32 doctorsky-Lenovo-E31-70
> Fri Dec 29 10:28:34 2017 daemon.info hostapd: wlan0-apname: STA b4:6d:83:73:96:32 WPA: group key handshake completed (RSN)
> Fri Dec 29 10:28:44 2017 daemon.info dnsmasq-dhcp[1883]: RTR-SOLICIT(anygw) 
> Fri Dec 29 10:28:44 2017 daemon.info dnsmasq-dhcp[1883]: RTR-ADVERT(anygw) 2a00:1508:a0d:fe00::
> Fri Dec 29 10:29:20 2017 daemon.err uhttpd[926]: Warning - mac already known (changing name from 'LiMe-WDR3600_anygw' to 'LiMe-WDR3500_anygw'): aa:aa:aa:0d:fe:aa
> Fri Dec 29 10:29:20 2017 daemon.err uhttpd[926]: Warning - mac already known (changing name from 'LiMe-WDR3500_anygw' to 'LiMe-WDR4300_anygw'): aa:aa:aa:0d:fe:aa
> Fri Dec 29 10:30:00 2017 cron.info crond[1637]: USER root pid 11016 cmd /usr/sbin/smonit
> Fri Dec 29 10:30:00 2017 cron.info crond[1637]: USER root pid 11017 cmd ( for file in /etc/alfred/* ; do [ -x $file ] && $file ; done )
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq[1883]: read /etc/hosts - 4 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dnsmasq-lease-share - 5 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/bmx6 - 12 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_64:70:02:ed:f9:5e - 2 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq-dhcp[1883]: read /etc/ethers - 0 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq-dhcp[1883]: read /tmp/dhcp.hosts_remote
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq[1883]: read /etc/hosts - 4 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dnsmasq-lease-share - 7 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/bmx6 - 12 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_64:70:02:ed:f9:5e - 2 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq-dhcp[1883]: read /etc/ethers - 0 addresses
> Fri Dec 29 10:30:01 2017 daemon.info dnsmasq-dhcp[1883]: read /tmp/dhcp.hosts_remote
> Fri Dec 29 10:30:43 2017 daemon.notice hostapd: wlan0-apname: AP-STA-DISCONNECTED b4:6d:83:73:96:32
> Fri Dec 29 10:32:00 2017 cron.info crond[1637]: USER root pid 11230 cmd /usr/bin/bmx6hosts_dnsmasq_update
> Fri Dec 29 10:34:52 2017 daemon.notice hostapd: wlan0-ap: AP-STA-DISCONNECTED 20:82:c0:39:f8:0b
> Fri Dec 29 10:35:00 2017 cron.info crond[1637]: USER root pid 11491 cmd /usr/sbin/smonit
> Fri Dec 29 10:35:00 2017 cron.info crond[1637]: USER root pid 11492 cmd ( for file in /etc/alfred/* ; do [ -x $file ] && $file ; done )
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq[1883]: read /etc/hosts - 4 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dnsmasq-lease-share - 7 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/bmx6 - 12 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_64:70:02:ed:f9:5e - 2 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq-dhcp[1883]: read /etc/ethers - 0 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq-dhcp[1883]: read /tmp/dhcp.hosts_remote
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq[1883]: read /etc/hosts - 4 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dnsmasq-lease-share - 7 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/bmx6 - 12 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_64:70:02:ed:f9:5e - 2 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq[1883]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq-dhcp[1883]: read /etc/ethers - 0 addresses
> Fri Dec 29 10:35:01 2017 daemon.info dnsmasq-dhcp[1883]: read /tmp/dhcp.hosts_remote

If you need another log, just ask me. Sorry for all the lines copied. I am still a novice :-(

ilario, 2017-12-29:
Seems like DNSmasq dying very quickly (in your log there are messages from starting dnsmasq after each run of smonit each 5 minutes), it causes many many troubles with current release, refer to #111 #242 #243.
Can you check smonit log? `/tmp/smonit/smonit.log`
I never got why this happens, must be some dnsmasq bug hopefully fixed in the last version.
Can you try to upgrade dnsmasq on every router? You can do it through the web interface or downloading last version [here](https://downloads.lede-project.org/releases/packages-17.01/mips_24kc/base/dnsmasq-dhcpv6_2.78-4_mips_24kc.ipk) and install it with `opkg` (I linked the last version on LEDE repo 'cause I can't find it on our repos).

ilario, 2017-12-29:
Post here the content also of `/tmp/smonit/smonit.state`

modante, 2017-12-29:
[2017/12/29 09:00] bmx6 -> Status: ok                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                              
[2017/12/29 09:00] dnsmasq -> Executing                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                            
[2017/12/29 09:00] dnsmasq -> Status: ok                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                           
[2017/12/29 09:00] sleepy_radio -> Executing                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                       
[2017/12/29 09:00] sleepy_radio -> Status: error                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                   
[2017/12/29 09:00] sleepy_radio -> Fails:0, executing quick fix                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                    
[2017/12/29 09:05] bmx6 -> Executing                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               
[2017/12/29 09:05] bmx6 -> Status: ok                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                              
[... cut ...]
[2017/12/29 10:55] dnsmasq -> Status: ok                                                                                                                                
[2017/12/29 10:55] sleepy_radio -> Executing                                                                                                                            
[2017/12/29 10:55] sleepy_radio -> Status: error                                                                                                                        
[2017/12/29 10:55] sleepy_radio -> No recovery after longfix, doing nothing else

modante, 2017-12-29:
I guess you are close!!! :-)
I am going to update the package dnsmasq and I will tell you.
Should I update it in all the routers, included the one connected to the WAN?
Thank you very very muuuuch :-) You are great!!!!

ilario, 2017-12-29:
I would say, update every router :)
Looking at the log seems that you hit exactly #243 which is very easy to fix, if you want to try to play with pull requests :D

modante, 2017-12-29:
Ok. Updated dnsmasq and working like a charm for about an hour.
The only issue that still cannot manage is the announcing of the internet gateway (through BMX6 tunnels) when the router connected to WAN is rebooted (or there is a power off in the power supply): [https://github.com/libremesh/lime-packages/issues/265](https://github.com/libremesh/lime-packages/issues/265)

p4u, 2017-12-29:
> The only issue that still cannot manage is the announcing of the internet gateway (through BMX6 tunnels) when the router connected to WAN is rebooted (or there is a power off in the power supply): #265 

Can you give more details about this issue?

modante, 2017-12-29:
> Can you give more details about this issue?
I opened an issue:
[https://github.com/libremesh/lime-packages/issues/265](https://github.com/libremesh/lime-packages/issues/265)

modante, 2017-12-29:
Still persist the issue of losts of internet connectivity after updating the dnsmasq packet in all the routers.
I have been 1 hour outside of the mesh and when returned, neither of the 2 routers had internet connection, only the one with the WAN connection worked.
I have the logs of the system if you want but they are long (because I don't know exactly when it happened)
Should I try to change the lines you said? (/etc/smonit/dnsmasq)?
Or any other suggestion?
Thanks and regards :-)

p4u, 2017-12-29:
Rather than logs it would be more useful if you can execute this set of commands on the gateway and one non-gateway node.

bmx6 -cp
logread | grep watchping
ip r get 1.1.1.1

modante, 2017-12-30:
In the non-gateway node:
```
root@LiMe-WDR4300:~# bmx6 -cp
 plugin                 bmx6_config.so       
 plugin                 bmx6_json.so         
 plugin                 bmx6_sms.so          
 ipVersion              6                    
 syslog                 0                    
 dbgMuteTimeout         1000000              
 dev                    br-lan               
 dev                    wlan0-mesh_13        
    /rateMax            54000                
 dev                    wlan1-mesh_13        
    /rateMax            54000                
 dev                    eth0.2               
 configSync             0                    
 tunOutTimeout          100000               
 tunDev                 main                 
    /tun4Address        10.13.165.148/16     
    /tun6Address        2a00:1508:a0d:fe00::94a5:fa00/64 
 tunOut                 nodes                
    /network            172.16.0.0/12        
 tunOut                 clouds               
    /network            10.0.0.0/8           
 tunOut                 inet4                
    /network            0.0.0.0/0            
    /maxPrefixLen       0                    
 tunOut                 inet6                
    /network            ::/0                 
    /maxPrefixLen       0                    
 tunOut                 publicv6             
    /network            2000::/3             
    /maxPrefixLen       64                   
 tunOut                 inet4p               
    /network            0.0.0.0/0            
    /gwName             LiMe-WDR3500         
    /maxPrefixLen       0                    
    /rating             1000                 
 tunOut                 inet6p               
    /network            ::/0                 
    /gwName             LiMe-WDR3500         
    /maxPrefixLen       0                    
    /rating             1000
```
```
root@LiMe-WDR4300:~# logread | grep watchping
root@LiMe-WDR4300:~# 
```
(returns nothing)
```
root@LiMe-WDR4300:~#  ip r get 1.1.1.1
1.1.1.1 dev bmxOut_LiMe-WD  src 10.13.165.148 
    cache 
```
--------------------
In the gateway node:
```
root@LiMe-WDR3500:~# bmx6 -cp
 plugin                 bmx6_config.so       
 plugin                 bmx6_json.so         
 plugin                 bmx6_sms.so          
 ipVersion              6                    
 syslog                 0                    
 dbgMuteTimeout         1000000              
 dev                    br-lan               
 dev                    wlan0-mesh_13        
    /rateMax            54000                
 dev                    wlan1-mesh_13        
    /rateMax            54000                
 dev                    eth1                 
 configSync             0                    
 tunOutTimeout          100000               
 tunDev                 main                 
    /tun4Address        10.13.42.209/16      
    /tun6Address        2a00:1508:a0d:fe00::d12a:8600/64 
 tunIn                  inet4                
    /network            0.0.0.0/0            
 tunOut                 nodes                
    /network            172.16.0.0/12        
 tunOut                 clouds               
    /network            10.0.0.0/8           
 tunOut                 inet6                
    /network            ::/0                 
    /maxPrefixLen       0                    
 tunOut                 publicv6             
    /network            2000::/3             
    /maxPrefixLen       64  
```
```
root@LiMe-WDR3500:~# logread | grep watchping
root@LiMe-WDR3500:~# 
```
```
root@LiMe-WDR3500:~# ip r get 1.1.1.1
1.1.1.1 via 192.168.104.129 dev pppoe-wan  src 192.168.104.146 
    cache 
```

modante, 2017-12-30:
By the way, I have been reading the smonit log and there is no clues about a dnsmasq failure:
```
[2017/12/29 20:25] dnsmasq -> Executing                                         
[2017/12/29 20:25] dnsmasq -> Status: ok                                        
[2017/12/29 20:25] sleepy_radio -> Executing                                    
[2017/12/29 20:25] sleepy_radio -> Status: error                                
[2017/12/29 20:25] sleepy_radio -> No recovery after longfix, doing nothing else
[2017/12/29 20:30] bmx6 -> Executing                                            
[2017/12/29 20:30] bmx6 -> Status: ok
```
This one repeats each 5 minutes and always dnsmasq -> Status: ok

I stopped dnsmasq following the /etc/smonit/dnsmasq script:
```
/etc/init.d/dnsmasq stop
killall -9 dnsmasq
/etc/init.d/dnsmasq start
```
and after that internet returns :-)
Is possible that smonit is not detecting this issue with dnsmasq and therefore is not restarting the service?
Regards :-)

modante, 2017-12-30:
I have restarted the dnsmasq in the 3th router when internet connectivity dropped but didn't work. I had to reboot it in order to make it work :-(
Sorry, my skills are really low :-/
Regards :-)

modante, 2017-12-30:
More tests. Suddenly the WDR3600 loosed the internet conectivity.
No clues in the smonit: dnsmasq -> Status: ok
Stoped and started dnsmasq without success
Dnsmasq updated: Package: dnsmasq-dhcpv6 Version: 2.78-4
This is the system log at the moment when it happened:
```
Sat Dec 30 11:52:00 2017 cron.info crond[874]: USER root pid 7803 cmd /usr/bin/bmx6hosts_dnsmasq_update
Sat Dec 30 11:55:00 2017 cron.info crond[874]: USER root pid 7852 cmd /usr/sbin/smonit
Sat Dec 30 11:55:00 2017 cron.info crond[874]: USER root pid 7853 cmd ( for file in /etc/alfred/* ; do [ -x $file ] && $file ; done )
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq[1840]: read /etc/hosts - 4 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq[1840]: read /tmp/hosts/dnsmasq-lease-share - 8 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq[1840]: read /tmp/hosts/d-d-h_64:66:b3:fa:a5:94 - 2 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq[1840]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq[1840]: read /tmp/hosts/bmx6 - 12 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq[1840]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq-dhcp[1840]: read /etc/ethers - 0 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq-dhcp[1840]: read /tmp/dhcp.hosts_remote
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq[1840]: read /etc/hosts - 4 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq[1840]: read /tmp/hosts/dnsmasq-lease-share - 8 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq[1840]: read /tmp/hosts/d-d-h_64:66:b3:fa:a5:94 - 2 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq[1840]: read /tmp/hosts/d-d-h_a0:f3:c1:86:2a:d1 - 2 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq[1840]: read /tmp/hosts/bmx6 - 12 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq[1840]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq-dhcp[1840]: read /etc/ethers - 0 addresses
Sat Dec 30 11:55:01 2017 daemon.info dnsmasq-dhcp[1840]: read /tmp/dhcp.hosts_remote
Sat Dec 30 11:55:10 2017 daemon.notice hostapd: wlan0-apname: AP-STA-DISCONNECTED b4:6d:83:73:96:32
Sat Dec 30 11:55:15 2017 daemon.info dnsmasq-dhcp[1840]: RTR-SOLICIT(anygw) 
Sat Dec 30 11:55:15 2017 daemon.info dnsmasq-dhcp[1840]: RTR-ADVERT(anygw) 2a00:1508:a0d:fe00::
Sat Dec 30 11:55:50 2017 daemon.info hostapd: wlan0-apname: STA b4:6d:83:73:96:32 IEEE 802.11: authenticated
Sat Dec 30 11:55:50 2017 daemon.info hostapd: wlan0-apname: STA b4:6d:83:73:96:32 IEEE 802.11: associated (aid 1)
Sat Dec 30 11:55:50 2017 daemon.notice hostapd: wlan0-apname: AP-STA-CONNECTED b4:6d:83:73:96:32
Sat Dec 30 11:55:50 2017 daemon.info hostapd: wlan0-apname: STA b4:6d:83:73:96:32 WPA: pairwise key handshake completed (RSN)
Sat Dec 30 11:55:50 2017 daemon.info dnsmasq-dhcp[1840]: DHCPREQUEST(anygw) 10.13.98.177 b4:6d:83:73:96:32 
Sat Dec 30 11:55:50 2017 daemon.info dnsmasq-dhcp[1840]: DHCPACK(anygw) 10.13.98.177 b4:6d:83:73:96:32 doctorsky-Lenovo-E31-70
Sat Dec 30 11:56:00 2017 daemon.info dnsmasq-dhcp[1840]: RTR-SOLICIT(anygw) 
Sat Dec 30 11:56:00 2017 daemon.info dnsmasq-dhcp[1840]: RTR-ADVERT(anygw) 2a00:1508:a0d:fe00::
Sat Dec 30 11:56:00 2017 cron.info crond[874]: USER root pid 8001 cmd /usr/bin/bmx6hosts_dnsmasq_update
```
Is possible that the issue is beyond dnsmasq problem? Because restarting dnsmasq should make it work again but it is not resolving it.
30 minutes after the loose of internet connectivity still is not working. The only way to make it work again is rebooting the router.
Any ideas?
Regards :-)

p4u, 2017-12-30:
I understand that the information you provided is in the moment that there is not Internet on the non-gateway client. Am I wrong?

I forgot the -i, please once it happens run in the gateway: logread | grep -i watchping

Would be useful to know if the issue is DNS related or IP related. How do you test if there is internet connection? To know it you can run these two test on the non-gateway node-

`ping -c5 8.8.8.8` 
`nslookup libremesh.org`

Also it will be interesting to know if it is a network problem, once you detect it try to ping the gateway from both ipv4 and ipv6

`ping 10.13.42.209`
`ping6 LiMe-WDR3500.mesh`

modante, 2017-12-30:
Sorry if I say something stupid, but comparing the routings of the 2 non-gateway routers, the one wich loosed internet connection listed 12 ARP addresses while the one working showed only 4 addresses:
```
IPv4-Address | MAC-Address | Interface
10.13.98.177 | b4:6d:83:73:96:32 | br-lan
10.13.106.90 | 70:e7:2c:9a:54:a9 | anygw
10.13.42.209 | a0:f3:c1:86:2a:d1 | br-lan
10.13.98.177 | b4:6d:83:73:96:32 | anygw
```
After rebooting the router. internet came back and now shows 4 (instead 12) ARP addresses:

```
IPv4-Address | MAC-Address | Interface
10.13.234.191 | 20:82:c0:e6:dd:5f | anygw
10.13.98.177 | b4:6d:83:73:96:32 | anygw
10.13.98.177 | b4:6d:83:73:96:32 | br-lan
10.13.42.209 | a0:f3:c1:86:2a:d1 | br-lan
```
Also I would say (I will confirm) that in the moment when internet dropped, this line appeared in the system log:
`Sat Dec 30 15:56:00 2017 cron.info crond[1628]: USER root pid 9327 cmd /usr/bin/bmx6hosts_dnsmasq_update`
Maybe is useless, sorry but I am novice trying to help with the only things I can read (not too much).
Thanks and regards :-)

ilario, 2017-12-30:
This is the script you mentioned [bmx6hosts_dnsmasq_update](https://github.com/libremesh/lime-packages/blob/develop/packages/lime-proto-bmx6/src/bmx6hosts_dnsmasq_update).

modante, 2017-12-30:
@p4u I understand what you say. In all the cases, are the non-gateway routers the ones that lost internet connectivity, sometimes the WDR4300, others the WDR3600 and sometimes both.
If I reboot the gateway router, the problem persists, still there is no internet in the non-gateway routers till I reboot them.
Yesterday whilw one of the drops of internet connectivity I tried to ping 8.8.8.8 without success from this router (of course it ponged from the gateway router). Then I think is not a DNS issue.
In the next drop of connectivity I will post logread | grep -i watchping
I was thinking to flash with dev snapshot one of the non-gateway routers because I start to desperate :-/
Thanks and regards :-)

modante, 2017-12-30:
@p4u Also to say that when I am connected to a non-gateway router that drops internet connectivity, I can access to the rest of the routers of the mesh, LuCi, SSH, ect. The only thing wrong is that there is not internet

modante, 2017-12-31:
Now both non-gateway routers failed (less than 30 min after rebooting them!!!!)
```
root@LiMe-WDR4300:~# logread | grep -i watchping
Fri Dec 29 18:52:40 2017 user.info watchping: starting all tasks
Fri Dec 29 18:52:41 2017 user.info watchping: started task (interface=eth0.2;timeout=120;pinghosts=8.8.8.8;pinginterval=20;hookname=wan)
Fri Dec 29 18:52:41 2017 user.info watchping: all tasks started
Fri Dec 29 18:52:41 2017 user.info watchping: adding ifup trigger for wan
```
```
root@LiMe-WDR3600:~# logread | grep -i watchping
Sat Dec 30 19:35:02 2017 user.info watchping: starting all tasks
Sat Dec 30 19:35:02 2017 user.info watchping: started task (interface=eth0.2;timeout=120;pinghosts=8.8.8.8;pinginterval=20;hookname=wan)
Sat Dec 30 19:35:02 2017 user.info watchping: all tasks started
Sat Dec 30 19:35:02 2017 user.info watchping: adding ifup trigger for wan
```
```
root@LiMe-WDR3600:~# ping -c5 8.8.8.8
connect: Network unreachable
```
```
root@LiMe-WDR3600:~# ping 10.13.42.209
PING 10.13.42.209 (10.13.42.209) 56(84) bytes of data.
64 bytes from 10.13.42.209: icmp_req=1 ttl=64 time=15.4 ms
64 bytes from 10.13.42.209: icmp_req=2 ttl=64 time=1.37 ms
64 bytes from 10.13.42.209: icmp_req=3 ttl=64 time=1.38 ms
64 bytes from 10.13.42.209: icmp_req=4 ttl=64 time=1.63 ms
^C
--- 10.13.42.209 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3004ms
rtt min/avg/max/mdev = 1.379/4.959/15.433/6.048 ms

root@LiMe-WDR3600:~#  ping6 LiMe-WDR3500.mesh
PING LiMe-WDR3500.mesh(LiMe-WDR3500.mesh) 56 data bytes
64 bytes from LiMe-WDR3500.mesh: icmp_seq=1 ttl=64 time=3.05 ms
64 bytes from LiMe-WDR3500.mesh: icmp_seq=2 ttl=64 time=1.37 ms
64 bytes from LiMe-WDR3500.mesh: icmp_seq=3 ttl=64 time=1.32 ms
64 bytes from LiMe-WDR3500.mesh: icmp_seq=4 ttl=64 time=1.31 ms
^C
--- LiMe-WDR3500.mesh ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3005ms
rtt min/avg/max/mdev = 1.313/1.768/3.055/0.743 ms

```

modante, 2017-12-31:
Last night I tried to flash dev snapshot on WDR3600 but it had no wireless drivers neither DHCP (I thought i was bricked the device!!!!). Then I flashed again lede-17.01.2-lime-default-ar71xx-generic-tl-wdr3600-v1-squashfs-factory.bin, changed the channel of wifi 2.4GHz, selected WDR-3500 like primary gateway and has been working without issues till now (more than 12 hours). The thing is that before flashing, the only mod I did was to upgrade dsnmasq.
I left WDR4300 without changes and ran this:
`ping -i 5 74.125.130.94 | while read pong; do echo "$(date): $pong"; done > ./pingt.log &`
Few minutes ago dropped the internet connectivity. The las pong was:
`Sun Dec 31 08:44:53 -03 2017: 64 bytes from 74.125.130.94: icmp_req=260 ttl=39 time=346 ms`
And the system log:
```
Sun Dec 31 08:44:00 2017 cron.info crond[1631]: USER root pid 3144 cmd /usr/bin/bmx6hosts_dnsmasq_update
Sun Dec 31 08:45:00 2017 cron.info crond[1631]: USER root pid 3176 cmd /usr/sbin/smonit
Sun Dec 31 08:45:00 2017 cron.info crond[1631]: USER root pid 3177 cmd ( for file in /etc/alfred/* ; do [ -x $file ] && $file ; done )
Sun Dec 31 08:45:01 2017 daemon.info dnsmasq[1873]: read /etc/hosts - 4 addresses
```
It is not clear because passed more than 53 seconds but don't you think that bmx6hosts_dnsmasq_update could be the key?
And what about watchping? Because if I run it in the gateway router it gives me nothing
```
root@LiMe-WDR3500:~# logread | grep -i watchping
root@LiMe-WDR3500:~# 
```
Thanks and regards :-)

modante, 2017-12-31:
The WDR3500 with the clean firmware installed last night is working ok. I had a drop in internet connectivity a few minutes ago that returned automatically 3 minutes after. When the WDR4300 losts inet connectivity only returns after rebooting the router.
I am thinking to clean and flash the gateway router WDR3500 and setup WAN to observe what happens with WDR4300 and the clean WDR3600. Remember that WDR4300 has dsnmasq updated.
Regards :-)

modante, 2018-01-01:
Last night I flashed a clean lede-17.01.2-lime-default-ar71xx-generic-tl-wdr3500-v1-squashfs-factory.bin in the gateway router and configured the WAN PPPoE connection. Since them, the 3 routers are working without one issue. I left the WDR4300 untouched with the dnsmasq updated (i didn't do that on WDR3600 neither WDR3500)
Now I think I was far from detecting the issue and cannot replicate it. The only 2 packages that were installed in the gateway router before the cleaning were QoS and SQM that now didn't install.
Sorry, I feel I made you waste your time :-/

dangowrt, 2018-01-01:
I think the problem you encountered is caused by watchping's interface triggers being registered too late. Try to change `START=99` into `START=19` (ie. just before network) in
https://github.com/libremesh/lime-packages/blob/develop/packages/watchping/files/etc/init.d/watchping#L6
That should fix it.

modante, 2018-01-02:
Hello. Here again :-)
I recently was running ping in the WDR3600 that has low connectivity to the mesh (about 40% of signal average both, 2.4 and 5 GHz). When ping started to fail (more than 5 min without a pong) I manually ran /etc/watchping/wan-fail.d/bmx6-gw and /etc/watchping/wan-ok.d/bmx6-gw in the same WDR3600. Nothing changed because is a non-gateway router.
Then I ran both scripts (first the fail one and after the ok one) in the WDR3500 (the gateway router) and suddenly the pongs returned to the WDR3600.
Casualty? or Causally?
I have made this test two time with the immediate return of the pongs.
Probably one of you can analyze and get a conclusion.
Thank you and regards :-)

modante, 2018-01-02:
I forgot to explain that actually the 3 routers forms an "L", I mean in one end is the WDR3500 (gateway), in the middle the WDR4300 and at the other end the WDR3600. When internet connectivity is lost in the WDR3600 (the case I mentioned the last post), the connectivity remains in the WDR4300 (the one in the middle). The WDR3500 and WDR3600 are not connected (or with very low signal cased by the trees) between them.
I hope it helps to clarify :-)
Regards :-)

modante, 2018-01-03:
Help Please!!! I am starting to be desperated! :-(
Now, WDR4300 (which is between WDR3500 and WDR3600) has lost internet connectivity.
Running 
```
/etc/watchping/wan-fail.d/bmx6-gw
/etc/watchping/wan-ok.d/bmx6-gw 
```
in the gateway router WDR3500 doesn't fix the internet connectivity on WDR4300.
But WDR3600 has internet connectivity!!!
```
root@LiMe-WDR4300:~# logread | grep -i watchping
root@LiMe-WDR4300:~# 
```
gives nothing
```
root@LiMe-WDR4300:~# bmx6 -cp
 plugin                 bmx6_config.so       
 plugin                 bmx6_json.so         
 plugin                 bmx6_sms.so          
 ipVersion              6                    
 syslog                 0                    
 dbgMuteTimeout         1000000              
 dev                    br-lan               
 dev                    wlan0-mesh_13        
    /rateMax            54000                
 dev                    wlan1-mesh_13        
    /rateMax            54000                
 dev                    eth0.2               
 configSync             0                    
 tunOutTimeout          100000               
 tunDev                 main                 
    /tun4Address        10.13.165.148/16     
    /tun6Address        2a00:1508:a0d:fe00::94a5:fa00/64 
 tunOut                 nodes                
    /network            172.16.0.0/12        
 tunOut                 clouds               
    /network            10.0.0.0/8           
 tunOut                 inet4                
    /network            0.0.0.0/0            
    /maxPrefixLen       0                    
 tunOut                 inet6                
    /network            ::/0                 
    /maxPrefixLen       0                    
 tunOut                 publicv6             
    /network            2000::/3             
    /maxPrefixLen       64                   
 tunOut                 inet4p               
    /network            0.0.0.0/0            
    /gwName             LiMe-WDR3500         
    /maxPrefixLen       0                    
    /rating             1000                 
 tunOut                 inet6p               
    /network            ::/0                 
    /gwName             LiMe-WDR3500         
    /maxPrefixLen       0                    
    /rating             1000 
```
What can I do to debug this issue?!?!
Sorry to insist but almost a month trying to make it work the setup and I am delaying the project too much.
Thank you very much for your patience and regards :-)

modante, 2018-01-03:
I don' know what to do to make this setup working.
Resuming, internet connectivity is lost each time (no more than 2 or 3 hours working without issues) in all the non-gateway routers (not simultaneously, sometimes one, sometimes the other).
Sometimes I can revive the internet connectivity restarting the bmxg-gw in the gateway router.
But some other times I cannot get it work unless I reboot the router.
Now I have one of the routers without connectivity and cannot get it work.
Is normal that there are running 2 dnsmasq processes?
When I run ps i get:
```
 1892 dnsmasq   1120 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg02411c -k -x /var/run/dnsmasq/dnsmasq.cfg02411c
 1895 root      1080 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg02411c -k -x /var/run/dnsmasq/dnsmasq.cfg02411c
```
Is this normal?
@ilario said that is a smonit issue related with [https://github.com/libremesh/lime-packages/issues/243](https://github.com/libremesh/lime-packages/issues/243) but the status and log of smonit always says that dnsmasq is working ok and that hadn't issues.
Updating the dnsmasq package didn't make it work better.

Please help, I don't know what to to! :-(
Thanks and regards

modante, 2018-01-03:
I have discovered something.
WDR4300 was not having internet connectivity since 3 hours ago. I tried to restart and reload dnsmasq in the 3 routers without succes. WDR3600 has been working during this time even when this one was connected to the gateway (WDR3500) through the WDR3400 (tha haven't connectivity).
Then I restarted the gateway and suddenly the pongs on the WDR4300 (not working by hours) returned.
That this means that the problem is the gateway router?
Could be the SQM scripts that are the only package that I installed in the gateway? (and luci-app-proto-pppoe) But was disabled! Now I have removed.
Thanks and regards :-)

modante, 2018-01-27:
Three weeks ago I reflashed the 3 devices with lede-17.01.2-lime and installed no any additional package (neither Luci-app-pppoe) and since them, except for this issue that appeared a few days ago (https://github.com/libremesh/lime-packages/issues/300) it has been working fine.
I suspect that either SQM scripts or QoS packages that are the only one I installed before, could led to the issue, but I didn't make more tests because I needed the setup working (I promised to my neighbors)
Thanks and regards :-)