libremesh/lime-packages #275: Migration to OpenWrt

ilario, 2018-01-08:
See libremesh/lime-sdk#48
Where LEDE is cited in this repository:
https://github.com/libremesh/lime-packages/search?q=lede

ilario, 2019-12-04:
In #418 I forgot to update this line:

https://github.com/libremesh/lime-packages/blob/7607aaacefdf13a4e232e9d7dd5d714e9e63d112/packages/lime-system/files/etc/uci-defaults/92_add-lime-repos#L13

spiccinini, 2019-12-04:
> In #418 I forgot to update this line:
> 
> https://github.com/libremesh/lime-packages/blob/7607aaacefdf13a4e232e9d7dd5d714e9e63d112/packages/lime-system/files/etc/uci-defaults/92_add-lime-repos#L13

No you didn't forget, you introduced a bug changing it  that I fixed in PR #598 . 
If this script is failing it is for something else. In my setup it does not fail (all the uci-defaults exit with 0 status)

closed 2019-12-04 by spiccinini

ilario, 2019-12-04:
LOL I didn't realize that #598 happened!
I tested that the variable was named OPENWRT_ARCH but my build is based on OpenWrt's `openwrt-19.07` branch, which as @aparcar noted [here](https://github.com/libremesh/lime-packages/pull/598#issuecomment-533261119) includes https://github.com/openwrt/openwrt/commit/8a34a54b6aa6e9a5e2983d554fd5b97bec97e891

@spiccinini can you confirm that next LibreMesh release will be based on OpenWrt 18.06?

dangowrt, 2019-12-04:
@spiccinini seriously? Make a release based on OpenWrt-18.06 **now** that OpenWrt-19.07 is about to be released (rc2 is already out, most likely we will not need another release candidate)? I don't think many things relevant for LibreMesh have changed which would prevent the same LibreMesh codebase to use OpenWrt-19.07 as a base for the release. What would be the advantage of using an outdated OpenWrt release?

spiccinini, 2019-12-14:
> LOL I didn't realize that #598 happened!
> I tested that the variable was named OPENWRT_ARCH but my build is based on OpenWrt's `openwrt-19.07` branch, which as @aparcar noted [here](https://github.com/libremesh/lime-packages/pull/598#issuecomment-533261119) includes [openwrt/openwrt@8a34a54](https://github.com/openwrt/openwrt/commit/8a34a54b6aa6e9a5e2983d554fd5b97bec97e891)

I've opened #667 to support 19.07

> @spiccinini can you confirm that next LibreMesh release will be based on OpenWrt 18.06?

I really don't know.

ilario, 2020-01-09:
Most of the reasons for sticking to OpenWrt 18.06 were related to LibreRouter (LibreRouter project has a highly optimized 18.06 branch fork, which includes fixes for LibreRouter and more useful stuff).

But now, the LibreRouter project has released a LibreMesh&OpenWrt18.06-based firmware "LibreRouterOS" see https://lists.libremesh.org/pipermail/lime-users/2020-January/001558.html
which means (I hope) that next LibreMesh release can be less tied to LibreRouter support and can benefit from an updated OpenWrt base.

In brief, I agree with @dangowrt about the fact that next LibreMesh release should be based on OpenWrt 19.07.x

PR #418 (2018-10-22): Change LEDE occurrences to OpenWrt
Fix #275