libremesh/lime-packages #276: dnsmasq-lease-share.lua is not properly executed

p4u, 2018-01-08:
Dnsmasq is configured with the following option: `option dhcpscript '/etc/alfred/dnsmasq-lease-share.lua'`

But according to dnsmasq documentation:

```
 --dhcp-script=<path>               Shell script to run on DHCP lease creation and destruction.
 --dhcp-luascript=path                Lua script to run on DHCP lease creation and destruction.
```
So if dhcp-script is used, dnsmasq executes the file as a shell script (sh), then logread shows the error:

`daemon.debug dnsmasq-script[11466]: /usr/lib/dnsmasq/dhcp-script.sh: /etc/alfred/dnsmasq-lease-share.lua: line 5: syntax error: unexpected word (expecting ")")
`
Obviously dhcp-luascript option should be used instead, but it is not implemented by OpenWRT UCI pharser, so it cannot be easily changed.

p4u, 2018-01-08:
The option should be added to /etc/dnsmasq.d/lime-blah

p4u, 2018-01-08:
Bad news...

`Mon Jan  8 16:54:11 2018 daemon.crit dnsmasq[12263]: recompile with HAVE_LUASCRIPT defined to enable Lua scripts at line 1 of /etc/dnsmasq.d/30-lime-dnsmasq-lease-share.conf`

p4u, 2018-01-08:
PR sent

closed 2018-01-10 by p4u

nordurljosahvida, 2019-03-25:
I'm running a MIR3G with firmware built with lime-sdk from the develop branch [latest commit: https://github.com/libremesh/lime-sdk/commit/943c4ad3d4f3cb8982c530e5d00ff7fa08cbf1d5] and I think this issue is still happening.

All of a sudden at least several times a day all wireless clients stop having internet connection and can't even ping the router itself, but stay connected to the wifi network.

```
Mon Mar 25 13:30:00 2019 daemon.warn dnsmasq[2216]: script process exited with status 2
Mon Mar 25 13:32:00 2019 cron.info crond[1249]: USER root pid 32398 cmd /usr/bin/bmx6hosts_dnsmasq_update
### This is when all wifi clients got cutoff
Mon Mar 25 13:33:10 2019 daemon.notice hostapd: wlan1-2: AP-STA-DISCONNECTED 9c:b6:d0:92:cf:5d
Mon Mar 25 13:33:24 2019 daemon.info hostapd: wlan1-2: STA 9c:b6:d0:92:cf:5d IEEE 802.11: authenticated
Mon Mar 25 13:33:24 2019 daemon.info hostapd: wlan1-2: STA 9c:b6:d0:92:cf:5d IEEE 802.11: associated (aid 1)
Mon Mar 25 13:33:24 2019 daemon.notice hostapd: wlan1-2: AP-STA-CONNECTED 9c:b6:d0:92:cf:5d
Mon Mar 25 13:33:24 2019 daemon.info hostapd: wlan1-2: STA 9c:b6:d0:92:cf:5d RADIUS: starting accounting session 5FC6DB52DA0C1125
Mon Mar 25 13:33:24 2019 daemon.info hostapd: wlan1-2: STA 9c:b6:d0:92:cf:5d WPA: pairwise key handshake completed (RSN)
Mon Mar 25 13:33:24 2019 daemon.info dnsmasq-dhcp[2216]: DHCPREQUEST(anygw) 10.42.29.49 9c:b6:d0:92:cf:5d
Mon Mar 25 13:33:24 2019 daemon.info dnsmasq-dhcp[2216]: DHCPACK(anygw) 10.42.29.49 9c:b6:d0:92:cf:5d computer-love
Mon Mar 25 13:33:24 2019 daemon.debug dnsmasq-script[2216]: /usr/lib/dnsmasq/dhcp-script.sh: /etc/alfred/dnsmasq-lease-share.lua: line 3: --[[: not found
Mon Mar 25 13:33:24 2019 daemon.debug dnsmasq-script[2216]: /usr/lib/dnsmasq/dhcp-script.sh: /etc/alfred/dnsmasq-lease-share.lua: line 5: syntax error: unexpected word (expecting ")")
Mon Mar 25 13:33:24 2019 daemon.warn dnsmasq[2216]: script process exited with status 2
```

Running the `dhcp-script` maually in `set -eu` mode I get this:

```
/usr/lib/dnsmasq/dhcp-script.sh 
/usr/lib/dnsmasq/dhcp-script.sh: line 5: USER_DHCPSCRIPT: parameter not set
```

Thanks

commit 0d9e277b (2018-01-10): Merge pull request #277 from libremesh/hotfix/dnmasq-lease-share-script

Fix #276 dnsmasq lease share script
commit 16bad061 (2018-01-10): Fix #276 dnsmasq lease share script