libremesh/lime-packages #281: DNSMasq fails to start

nicopace, 2018-01-16:
This is what @altergui has said about it:

```
Guido, [16.01.18 16:40]
otro tema q tengo colgado desde la hackaton en barcelona: aquel bug del init.d de dnsmasq

Guido, [16.01.18 16:40]
este fue mi mail original en marzo
"dnsmasq sometimes fails to start on 17.01.0"
http://lists.infradead.org/pipermail/lede-dev/2017-March/006828.html

Guido, [16.01.18 16:42]
hubo un par de respuestas "Score:5, Informative" pero no conclusivas
http://lists.infradead.org/pipermail/lede-dev/2017-March/006919.html

Guido, [16.01.18 16:44]
la verdad q no voy a encontrar tiempo/motivacion para darle seguimiento (como pense q tendria), asique si alguien quiere retomar ese hilo, adelante. si no, quedara en el olvido
```

ilario, 2018-01-17:
Related issues: #273 #242 #111

p4u, 2018-01-25:
Maybe it's time to start thinking about #294

aparcar, 2018-05-26:
The links to the mail doesn't work anymore, could you elaborate on the problem or close the issue?

ilario, 2018-05-28:
I think this refers to the problem that "smonit" fixes re-running DNSmasq when finds it dead.
So, the problem is that DNSmasq is very likely to die during the boot.
A lot of other issues are related to this one, for example the ones I mentioned in a previous comment and also #243

gmarcos87, 2019-05-19:
Using 18.06, even without smonit, I did not find this situation again. Maybe I didn't pay special attention to it, but it doesn't seem to be failing in the latest versions.

ilario, 2019-10-04:
Also in my experience, this looks automagically solved, and it's a really good thing. Closing.

closed 2019-10-04 by ilario