libremesh/lime-packages #287: IPv4 parametrization

p4u, 2018-01-23:
> option main_ipv4_address '10.%N1.0.0/16'	
> # Here you have 4 possibilities: set a static IP and the subnet, like '10.0.2.1/16'; parametrize with %Mn and %Nn, and set the subnet, like '10.%N1.%M5.%M6/16'; set a whole network address (not a specific IP) for getting the IP autocompleted in that network with bits from MAC address, this works also with netmasks other than /24 or /16, like '10.0.128.0/17' (but not valid network addresses, for example '192.0.128.0/16' or '192.0.129.0/17' won't get parametrized); set two different parameters, the first for subnet and the second for IP parametrization, like '10.0.128.0/16/17', this results in /16 subnet but IP parametrized in a /17 range (from 10.0.128.1 to 10.0.255.254).
> 	option main_ipv6_address '2a00:1508:0a%N1:%N200::/64'	# Parametrizable in the same way as main_ipv4_address. If used, the IP autocompletion will fill maximum the last 24 bits, so specifying an IP autocompletion range bigger than /104 is not useful.
>

So according to lime documentation, these are the possible configuration options:

1. `10.0.2.1/16` So the node will self configure 10.0.2.1/16
2. `10.%N1.%M5.%M6/16` The node will self configure an IP based on cloud ID (essid) and last MAC bytes, the result would be something like: 10.2.3.4/16
3. `10.0.128.0/16/24` So first is the Netmask (which will be 255.255.0.0) and second is the IP space to autoconfigure the IP. Then a random IP from 10.0.128.0/24 will be used but the LAN range will still be /16

Why something like this `10.2.%M5.1/27` does not work (netmask=255.255.255.255 after running lime-config)? Is it not the second case?

Also I'm missing a way to specify the same of the third case but the other side around, so:

10.2.0.0/27/16 -> meaning that the node should self configure a /27 of the biggest range /16, something like 10.2.5.33/27

Until now I though the third scenario was refering to this (I even used in some nodes and it worked fine!) but after reading the lime-example documentation and experiencing some strange behavior in some nodes, I think it is for some "random" reason that it worked for me.

Can we discuss about it? Am I right with this configuration description? @G10h4ck and @ilario where the two modifying these lines of code.

p4u, 2018-01-23:
This case did not work either  `10.5.1.1/24`, netmask is set to 255.255.255.255 and IP address changes to 10.5.1.RANDOM

p4u, 2018-01-26:
I found that (also without lime-proto-anygw installed) ipv4_main_address like 10.5.1.1/24 does not work but 10.5.1.2/24 does work. So probably there is some hardcoded stuff related to anygw in the network lua code.

ilario, 2018-01-28:
> Why something like this 10.2.%M5.1/27 does not work

This setting is identified as invalid by [this line](/libremesh/lime-packages/blob/develop/packages/lime-system/files/usr/lib/lua/lime/network.lua#L68) which avoids that IP to be the same of anygw's one, inserted by @G10h4ck:

```
--! Generated address is the one reserved for anygw like 192.0.2.1/24 ?
invalid = invalid or ipv4_template:equal(mc:minhost())
```

> So probably there is some hardcoded stuff related to anygw in the network lua code.

Yep, see above. If this is actually needed, I have no idea.

> (netmask=255.255.255.255 after running lime-config)?

Indeed this is a bug! [This line](/libremesh/lime-packages/blob/develop/packages/lime-system/files/usr/lib/lua/lime/network.lua#L71)

```
if invalid then ipv4_template = mc:maxhost() end
```

sets a single IP without the previously set prefix, so a line needs to be added (I couldn't find the elegant way to just change the IP without changing the prefix, please tell me if there's a better way):

```
ipv4_template:prefix(tonumber(ipv4_maskbits))
```

I'll do a PR on this. (wow, we took 2 years for noticing this bug!!! Thanks @p4u for reporting)

> Also I'm missing a way to specify the same of the third case but the other side around, so:
> 10.2.0.0/27/16 -> meaning that the node should self configure a /27 of the biggest range /16, something like 10.2.5.33/27

I did not included this case because I thought no one would want to have many randomly separated domains. Do we want to have this?

ilario, 2018-01-28:
> This case did not work either 10.5.1.1/24, netmask is set to 255.255.255.255 and IP address changes to 10.5.1.RANDOM

Are you sure it was assigned to random? Should have been .254 which is maxhost for that subnet. For subnet /27 that you were using above it would be .30

ilario, 2018-01-28:
See #301

p4u, 2018-01-29:
> Are you sure it was assigned to random? Should have been .254 which is maxhost for that subnet. For subnet /27 that you were using above it would be .30

You right, it look random to me but it's not.

> I did not included this case because I thought no one would want to have many randomly separated domains. Do we want to have this?

I see this case very useful. In Guifi.net, for instance, you can create go to the web page and Mesh Zone which will reserve a /16 (for example). This /16 network will be published by the BGP border nodes to the rest of the Guifi.net network.

Inside the mesh cloud a non-roaming network is deployed where every node manages its own /27 (assigned to br-lan for the AP/LAN clients). 

So the network-profile must be something like 10.5.0.0/27/16 meaning: _autocofigure a /27 from the whole /16_

If you can implement this case it would be great :)

ilario, 2018-01-29:
@p4u can you create two more issues for:

. the check for anygw being active and installed in network.lua
. the scenaio of randomly broken mesh

I would keep this issue just for the lack of netmask when invalid IP address is detected, which is fixed in #301

ilario, 2018-02-04:
@p4u can you create another issue for:

. the scenaio of randomly broken mesh

I already filed the one for "the check for anygw being active and installed in network.lua", discuss on #313

commit 42eed7f1 (2018-02-09): Merge pull request #301 from ilario/issue/287

network.lua: set again the prefix in case of invalid IPv4 address, fix #287
commit 920f6c6a (2018-02-09): set again the prefix in case of invalid IPv4 address, fix #287
closed 2018-02-09 by G10h4ck