libremesh/lime-packages #294: Replace dnsmasq by odhcpd

p4u, 2018-01-25:
Odhcpd seems a better option than dnsmasq, it's an DHCP, DHCPv6 and RA server designed and implemented by OpenWRT with a very good integration.

Furthermore dnsmasq is the originator of many issues, so ideally Libremesh will get rid of dnsmasq In the future.

Currently dnsmasq is used as DHCP server but also as DNS server. The aim of this issue is to replace only the DHCP task, but not yet the DNS.

This is the default configuration section of odhcpd in /etc/config/dhcp

```
config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
```

Just setting maindhcp to 1 enables Odhcpd and disables dnsmasq as DHCP server. So this is quite an easy modification.

The lease file is stored in '/tmp/hosts/odhcpd'. The current lease share subsystem (based on ALFRED) must be modified to spread and include the leases to/from the other nodes of the mesh Cloud.

The leasetrigger is an script which is executed once there is a new lease, in this case it send a SIGNAL to dnsmasq to refresh the host file, thus to include the new leases into the DNS system.

Am I forgeting something? It is just these two modifications required to change dnsmasq by odhcpd ?

Any volunteer to to this task?

p4u, 2018-01-25:
Also the IPv6 delegation and anygw should be adapted to odhcpd

aparcar, 2018-01-26:
```
config dhcp 'lan'              
        option interface 'lan'
        option start '100'    
        option limit '150'         
        option leasetime '12h'      
        option dhcpv4 'server'  # add to serve v4
        option dhcpv6 'server'
        option ra 'server'   
```

p4u, 2018-01-28:
See lime-proto-anygw

https://github.com/libremesh/lime-packages/blob/develop/packages/lime-proto-anygw/src/anygw.lua

Some options are added to `/etc/config/dhcp` that might not be compatible with odhcpd (need testing). Also there are some extra dnsmasq rules written to `/etc/dnsmasq.d/` that might be exported somehow to odhcp.

aparcar, 2018-01-31:
Looking at [odhcpd README](https://github.com/openwrt/odhcpd) it looks like some option are missing:

* [force](https://github.com/libremesh/lime-packages/blob/develop/packages/lime-proto-anygw/src/anygw.lua#L81) has no replacement in odhcpd, but maybe it's default?

* [custom mtu](https://github.com/libremesh/lime-packages/blob/develop/packages/lime-proto-anygw/src/anygw.lua#L84) seems unsupported but may works automatically?

>   a) automatic detection of prefixes, delegated prefix and default routes, MTU

The content of `/etc/dnsmasq.d/lime-proto-anygw-20-ipv6.conf`

    enable-ra

can be archived with `option ra 'server'` for each interface.

    dhcp-range=tag:anygw,2012::,ra-names,24h
    dhcp-option=tag:anygw,option6:domain-search,qmp

Looks dns related and so not related to dhcp.

@dangowrt can you help out here?

p4u, 2018-02-01:
Related issues #104 and #150

aparcar, 2018-02-10:
depend on https://github.com/libremesh/lime-packages/pull/323

ilario, 2019-01-27:
@aparcar sorry for the ignorant question, would #379 completely solve this or just partially?

ilario, 2019-10-04:
There was also a duplicate in #375 and a pending PR in #379.
Are we still interested in this? @G10h4ck @nicopace @spiccinini @gmarcos87 @altergui 
Currently dnsmasq seems to work just fine...

altergui, 2020-11-30:
>     * [custom mtu](https://github.com/libremesh/lime-packages/blob/develop/packages/lime-proto-anygw/src/anygw.lua#L84) seems unsupported but may works automatically?

according to https://openwrt.org/docs/techref/odhcpd custom mtu is now supported with option `ra_mtu`

altergui, 2020-12-02:
i've spent some time (re)evaluating the possibility of using odhcpd as dhcpv6 server in libremesh (instead of dnsmasq-dhcpv6) to converge with upstream and benefit from all the progress done into supporting native ipv6 uplinks.
this is feature request is still a blocker.
@spiccinini has read the relevant code and concluded he'd need a significant amount of time to solve this.
so i pinged @dangowrt upstream https://github.com/openwrt/odhcpd/issues/113

G10h4ck, 2020-12-11:
> > ```
> > * [custom mtu](https://github.com/libremesh/lime-packages/blob/develop/packages/lime-proto-anygw/src/anygw.lua#L84) seems unsupported but may works automatically?
> > ```
> 
> according to https://openwrt.org/docs/techref/odhcpd custom mtu is now supported with option `ra_mtu`

What about IPv4 MTU?

spiccinini, 2020-12-11:
The feature to add feed leases to odhcpcd is now in master https://git.openwrt.org/?p=project/odhcpd.git;a=commit;h=8f27697b9b82420890cedd429622052c8b67cea1

andybe, 2022-01-03:
option maindhcp '1'  - No more ipv4

I wann use unbound for dns, but revert do dnsmasq,
 because of no ipv4.
 
 Before I wrote this I disable dnsmasq (functional) change only the option above and ok no ipv4.
(yes there are no dns domains configured at this time)

config odhcpd 'odhcpd'                                                                                                                                                                                                                        
        option maindhcp '1'                                                                                                                                                                                                                   
        option leasefile '/tmp/hosts/odhcpd'                                                                                                                                                                                                  
        option leasetrigger '/usr/sbin/odhcpd-update'                                                                                                                                                                                         
        option loglevel '4' 

config dhcp 'lan' 
        option interface 'lan'                                                                                                                                                                                                                
        option domain 'lan'                                                                                                                                                                                                                            
        option start '100'                                                                                                                                                                                                                    
        option limit '150'                                                                                                                                                                                                                 
        option leasetime '12h'                                                                                                                                                                                                                
        option dhcpv4 'server'                                                                                                                                                                                                                
        option dhcpv6 'server'                                                                                                                                                                                                                
        option ra 'server'                                                                                                                                                                                                                                                                                                                                                                                                                                  
        list ra_flags 'managed-config'                                                                                                                                                                                                        
        list ra_flags 'other-config'                                                                                                                                                                                                          
        option ra_default '1'

howl, 2022-04-26:
> option maindhcp '1' - No more ipv4
>
> I wann use unbound for dns, but revert do dnsmasq, because of no ipv4.
>
> Before I wrote this I disable dnsmasq (functional) change only the option above and ok no [...]

By default odhcpd-ipv6only is installed so making it to serve ipv4 result in no ipv4 at all. You need to also remove odhcpd-ipv6only and install odhcpd.

ilario, 2023-05-11:
> What about IPv4 MTU?

To get closer to OpenWrt we have to use odhcpd-ipv6only for IPv6 and dnsmasq for IPv4.
So we do not care about MTU for IPv4.

@spiccinini how should we use the ubus call that you implemented?

fsa, 2024-07-10:
I don't have dnsmasq installed. I have odhcpd running as DHCP. I encountered a problem when trying to use ra_pref64 and DHCPv4 option 108. These options do not work on odhcpd.

ilario, 2024-07-10:
@dangowrt

ilario, 2024-11-07:
> The feature to add feed leases to odhcpcd is now in master https://git.openwrt.org/?p=project/odhcpd.git;a=commit;h=8f27697b9b82420890cedd429622052c8b67cea1

@dangowrt can odhcpcd continuously check a list of leases provided by other nodes (e.g. stored in a file generated by Alfred or shared-state)? What else is missing for being able to use odhcpcd instead of dnsmasq?

dangowrt, 2024-11-07:
At this point `odhcpd` expects external lease information to be fed via `ubus`. This is different from dnsmasq using a file and then subscribing to information about file changes using `inotify`. However, I'd recommend to add that kind of functionality to `shared-state` itself, eg. using `inotify-tools` and `diff` (or your own implementation in C).

ilario, 2024-11-07:
@G10h4ck @javierbrk can you have a look at this?