libremesh/lime-packages #300: wlan1-mesh disconects without reason

modante, 2018-01-27:
Setup:
Node 1: WDR3500 working as gateway. Connects to Node 2 in 5 GHz with wlan1-mesh
Node 2: WDR4300. Connects to Node 1 and 3 in 5 GHz with wlan1-mesh
Node 3: WDR3600. Connects to Node 2 in 5 GHz with wlan1-mesh. Also connects with Node 1 but with a very low signal then most of the times reads 0 mbps tx/rx
The 3 routers are flashed with lede-17.01.2 and with no any additional software installed.

The issue is with Node 2 (WDR4300). During 2 weeks has been working like a charm, but 5 days ago I discovered that loosed connectivity with the rest of the mesh. When I connect directly to this router I can see that wlan1-mesh has drop connectivity (0% signal in wireless menu). If I disconnect and reconnects this device it connects again and works in a normal way (in the last days it is failing 1 or 2 times at day). Cause the connection between the routers is 1 <-> 2 <-> 3, when the Node 2 disconnects the wlan1-mesh also Node 3 lost its connectivity to the mesh.
But what it is really strange for me is that if I connect directly to Node 1 when Node 2 reads 0% signal in wlan1-mesh, Node 1 appears like connected to Node 2 with the usual signal (about 60%)!!!! but in the associated stations list reads a constant Tx of 150.0 Mbit/s, 40MHz, MCS 7, Short GI and Rx of 121.5 Mbit/s, 40MHz, MCS 6 (usually this values oscillates a bit but not now). The same happens with Node 3 that shows connectivity with Node 2 but there is no internet connectivity neither in Node 2 and 3.
I have read the system log and I cannot find anything related with wlan1-mesh. Only in Kernel log I found that yesterday:
```
[34988.616570] batman_adv: The newly added mac address (64:66:b3:fa:a5:96) already exists on: wlan1-mesh_29
[34988.626293] batman_adv: It is strongly recommended to keep mac addresses unique to avoid problems!
[34988.676298] IPv6: ADDRCONF(NETDEV_UP): tmp.wlan1-mesh: link is not ready
```
I tested and is not true that the mac address is used by any other device in the mesh.
Last night I restarted the wlan1-mesh and worked and this morning when failed again I couldn't find any new event in the System and Kernel log related with wlan1-mesh.
Please, could you help me? Which logs would be needed in order to find the issue and its solution?
Thanks and regards :-)

nicoechaniz, 2018-01-27:
does the node recover from this state if you just run:
iw wlan1-mesh scan

?

modante, 2018-01-27:
I'll try the next time.
Thanks Nico :-)

modante, 2018-01-28:
This morning the issue was a little different. When I tried to connect to Node 2 (WDR4300) through the 2,4 Ghz I couldn't. But if connected to Node 1 I could access to the LuCi administration of Node 2 and observed that wlan1-mesh was working and indeed had connectivity also in Node 3. This time was the wlan0-ap and wlan0-apname the ones which failed.
_iw wlan1-mesh_ scan didn't change anything because now wlan1 (5 GHz) was working
_iw wlan0-ap scan_ and _iw wlan0-apname_ scan made the connectivity to the 2,4 GHz to return.
Could be the same issue related with the ath9x driver?
Is there any way to update the driver in order to test it?
The three routers have a very similar hardware, all of the uses ath9x driver. Is normal that the issue only affects to one of them (WDR4300)?
Thank you very much and regards :-)

modante, 2018-01-29:
This morning the issue came back, now in the 5 GHz radio.
I ran  iw wlan1-mesh scan and in 5 seconds the wireless mesh reconnected and worked fine.
Yesterday I manually updated kernel to the last one in openwrt trunk (kernel_4.9.77-1-f05c927e6709514cb789f9278695cb63_mips_24kc.ipk) in order to update ath9k to kmod-ath9k_4.9.77+2017-11-01-3_mips_24kc.ipk but when rebooted wireless was not working I don't know why. I searched in the kernel log but I couldn't see something that I (a novice) could understand in order to fix it. Then I returned to flash to lede-17.01.2 and restore the configuration and in less than 10 hours appeared the issue.
Could you help me please? I need to turn this setup stable.
Thanks and regards :-)

ilario, 2019-10-04:
11 months without updates, maybe the issue is already fixed?
Closing.

spiccinini, 2019-10-07:
I think this issue may still be valid. This is a ath9k driver issue that may be workarrounded with scans or other things. I have seen a similar problem (but not exactly the same) but was not able to debug it because I could not access both devices. In this case the bitrate on the device that I had access (the working device) was going to 1mbps after a restart of the wireless but the other device never responds, only sends beacons

ilario, 2019-10-08:
So this is a duplicate of #144, right?
We could close this and keep the discussion there.

ilario, 2020-01-09:
Closing as it sounds as a #144 duplicate to me and something that [cotonete](https://github.com/nicopace/cotonete) should be able to fix.

closed 2020-01-09 by ilario