libremesh/lime-packages #303: Implement bmx7 mesh-dns

p4u, 2018-01-28:
Following the work done in bmx6-mdns https://github.com/routek/qMp/tree/testing/packages/bmx6-mdns

It is a distributed DNS system where every node can publish domains and its resolution (to IPv4 or IPv6) using the bmx7 SMS plugin.

The domains published by all nodes in the network are saved into a file which dnsmasq will read (as it does with /etc/hosts).

To prevent possible MITM or Phishing attacks, a node can decide which TLD extensions are accepted. So ".com" or ".net" wont be accepted but ".mesh" or ".lime" would be.

IMO the implementation must:

1. store the own published domains to a UCI file 
2. do not depend on LUA, so use only shell script
3. be integrated and use procd to be executed as a daemon (do not use cron)
4. luci web app can be based on current bmx6-mdns work

p4u, 2018-01-28:
More info about mdns http://bmx6.net/projects/bmx6/wiki/Mdns

nicopace, 2018-03-22:
Are we also using the bmx6 version?
I guess a very cheap step could be to integrate the qmp implementation... and based on that generalize the bmx7 one... does it make sense?

dangowrt, 2018-06-23:
On Sat, Jun 23, 2018 at 12:27:07AM -0700, Paul Spooren wrote:
> The name `mdns` is somewhat [taken](https://git.openwrt.org/?p=project/mdnsd.git;a=summary) so I'd suggest call it `mesh-dns` instead

Imho mesh-dns can very well be realized be extending exactly this
`mdnsd` package, it already got most the things we need, no need
to reimplement all the DNS protocol, caching, etc. from scratch.

aparcar, 2018-06-23:
I think the idea of mesh-dns is to gather dns entries and put them into /tmp/hosts/ to let whatever DNS tool do the real DNS handling...?

nicopace, 2018-06-23:
@aparcar you are right.
The idea is for all mesh nodes to be aware of the names of the hosts in the network.
So each node needs to share with the rest those that were registered on them.

ilario, 2019-10-04:
As Babeld will replace BMX* in the next 19.something release, this issue became a low priority one.
I have no idea if a similar feature is already present or should be added using Babeld.
Clearly, LibreMesh firmware including BMX* can be generated, so any help on this is welcomed but I'll close as it's not a must-have right now.

closed 2019-10-04 by ilario