libremesh/lime-packages #306: separate Mesh and AP radio based on frequencies

aparcar, 2018-01-30:
Using an encrypted 11s mesh and encrypted AP on the same radio (2.4GHz or 5GHz) does not work as reported [here](https://github.com/libremesh/lime-packages/issues/208).

To archive this feature it's [possible](https://github.com/ilario/lime-packages/blob/dee5ba5da8d54b0fec744070269b4db33cba7de4/packages/lime-docs/files/lime-example#L76) to set individual configurations, but due to the usage of [foreach](https://github.com/libremesh/lime-packages/blob/6e152fc2f08838f722ac5240a1c2ec81c11b1994/packages/lime-system/files/usr/lib/lua/lime/config.lua#L48) in the [wireless.lua](https://github.com/libremesh/lime-packages/blob/6e543c79a08e41482f1fdd0214dbf93e97f5d312/packages/lime-system/files/usr/lib/lua/lime/wireless.lua#L78) module it's not possible to set that in `lime-defaults'.

p4u, 2018-02-01:
Follows #307

p4u, 2018-02-01:
My proposal is a module named (for instance) lime-smart-wifi-config

This module must run before lime-config (via uci-defaults script).

The module looks for the existing WiFi radios in the system (for such task, LUA funcions from lime-system might be used) and writes to /etc/config/lime the specific sections according to a configuracion file or a set of heuristics (for those which are not in the config file).

This is a proposal for the default heuristics (empty config file).

If only 1 radio
1. If 2.4GHz or 5Ghz -> Set it to the correct channel and AP + MESH
2. If Dual Band -> Set it to 2.4GHz and AP + Mesh

If more than 1 radio
1. If only one at 2.4GHz -> set it to only AP
2. If more than one at 2.4GHz -> set the first found to only AP
3. Set the rest of radios at 5GHz (prefered) or 2.4Ghz -> only Mesh 

Channels must not collide! So if two radios at 2.4Ghz, the first will get one channel and the second another. lime-defaults file can be used for getting the first channel, for instance "channel_2ghz = 11" the next will be 1 and the next 6 (always 1, 6 or 11).

Let's say config file is /etc/config/lime-smart-wifi-config

These are a set of options that might exist in this file

```
option all_to_mesh = 1/0
option all_to_ap = 1/0
option mesh_2ghz = 1/0
option mesh_5ghz = 1/0
option ap_2ghz = 1/0
option ap_5ghz=1/0
list channels_2ghz 1 11 6
list channels_5ghz 36 48 149
```

This is just a proposal!

p4u, 2018-02-01:
The options from smart-wifi-config can also live inside /etc/config/lime-defaults in a separated section.

aparcar, 2018-02-01:
> The options from smart-wifi-config can also live inside /etc/config/lime-defaults in a separated section.

I'd prefer that. I thinks @rogerpueyo did some work on that and I'll try to implement something like that.

ilario, 2018-02-04:
> If only 1 radio
    If 2.4GHz or 5Ghz -> Set it to the correct channel and AP + MESH
    If Dual Band -> Set it to 2.4GHz and AP + Mesh

> If more than 1 radio
    If only one at 2.4GHz -> set it to only AP
    If more than one at 2.4GHz -> set the first found to only AP
    Set the rest of radios at 5GHz (prefered) or 2.4Ghz -> only Mesh

I love this :D
In the common case of the TP-Link dual band routers would result in 2.4 GHz used just for AP and 5 GHz used just for mesh. I'm faithful that this will improve performances (still to be verified, see #109 and #262).

commit 1824a6c0 (2018-02-12): lime-smart-wifi: add package, fix #306 (#321)

fix separate Mesh and AP radio based on frequencies #306

1 radio with dual band is not supported due to missing device for
  testing, this will be added.

automatically swaps channels if multible radios on same freq exist

tested on tp link wdr 4300

also introduce new function to utils to check if an element is inside an
array
closed 2018-02-12 by aparcar