libremesh/lime-packages #308: convention to replace tabs with 4 white spaces

aparcar, 2018-01-30:
In many areas within the lime I find plenty of tabs to indent code. Is that intended? Is there a conventions to use tabs *or* whitespaces?

ilario, 2018-01-31:
Let's decide here the standard.
Right now in this repository there are 125 files with lines starting with tab and 52 files with lines starting with 4 spaces.
The files containing a mix of the two are 25.
I don't have any preference for one over the other, maybe 4 spaces is easier to respect.
We never fix the old indentation errors for not breaking the "git blame" function.

nicopace, 2018-01-31:
I would say to be consistent with the file you are editting.
Deciding one over the other will generate a long and boring process between those that have a preference for one over the other, that will not result in nothing productive.

Unless this is a serious issue about this, I would say to ignore the topic alltogether and get back to our lives (or to dnsmasq bugs #281, or @aparcar RFC #304 for example ;) ).

aparcar, 2018-01-31:
>  We never fix the old indentation errors for not breaking the "git blame" function.

Good point, did not think of that.

> Unless this is a serious issue about this

Nope it's not. But as I started to *[contribute](https://github.com/libremesh/lime-packages/pull/307/files#diff-192457068239fe711325c38fafa5f441)* some stuff I thought of asking before doing stuff [wrong](https://github.com/libremesh/lime-packages/pull/295#issuecomment-361078556). :)

p4u, 2018-02-01:
I see no point for not having this discussion in paralel with other important tasks.

I think it is a right thingo to get a consensus on this and DO NOT ACCEPT Pull requests not respecting this consensus.

Peronally, I like more spaces than tabs for identation (I've no point but it's what I've been doing in the past). However I would vote for **following the OpenWRT/LEDE approach which is use TABS**.

aparcar, 2018-02-04:
we can have the same fun discussion with camelCase or this_style for variable names. 
I like spaces (4) and this_style 🙄

p4u, 2018-02-05:
> we can have the same fun discussion with camelCase or this_style for variable names.
> I like spaces (4) and this_style roll_eyes

I agree on getting agreements here too. Code will look much unified and easy to read.

Another point: max line lenght before truncating

aparcar, 2018-02-05:
> Another point: max line lenght before truncating

80 💥

G10h4ck, 2018-02-05:
The convention I use is:

Max line length 80 unless it's ugly (I have reasons)
Use tab strictly for indentation only. Use space for alignment and other things (I have reasons)
about camel case or _ I usually prefer camel-case because names are sorter but it sometime looks ugly so i won't enforce this.

aparcar, 2018-06-19:
I adapted to @G10h4ck style but the camel case, I stick to `_`. 

Closing

closed 2018-06-19 by aparcar