libremesh/lime-packages #310: safe-reboot: Add notification to all console users that safe-reboot is running

nicopace, 2018-02-01:
If safe-reboot is running it is hard, even for the same user that is using it, to know when it will reboot the system.

It would be great if it can notify all users connected that it will reboot the system and how to prevent it or react to it.

I think it could be added here:
https://github.com/libremesh/lime-packages/blob/feb3b8996fe8e008ff391dd6feef9a6f11225dfa/packages/safe-reboot/files/usr/sbin/safe-reboot#L100

This is the function that it could use to notify:
```bash
function message_to_all_loggedin_users {
  message="$1";
  for pts in `ls /dev/pts/`; do
    echo "$message" > /dev/pts/$pts;
  done;
}
```

gmarcos87, 2018-02-01:
**SIDE NOTE**
With almost the same code we could notify users logged in on the node that another user entered.
This prevents users from changing the settings between them without knowing it.

ilario, 2018-02-02:
@nicopace did you see #238 ?

nicopace, 2018-02-02:
Haven't seen, sorry!
Closing duplicate!

closed 2018-02-02 by nicopace