libremesh/lime-packages #322: Spectrum analyzers

Benik3, 2018-02-08:
Hello.
I would like to ask on the spectral scan feature. Is anything working with LEDE?
First I found libremesh/spectrum-analyzer-packages. But half of the directories is empty and the second one I wasn't able to compile.
Then I looked into releases of libremesh FW. Under packages I found compiled pckg ubus-lime-openairview_2018-01-15-1516066917_mips_24kc.ipk. I installed this package, but I wasn't able to find how to use it.
Here in the packages I found openairview, I compiled it with LEDE 17.01.4 sources to target AR71xx and install it on WR841nd v9, but it didn't worked for me. It looks like it's outdated? (i will post logs and screens if you want)
Thank you :)

ilario, 2018-02-09:
@nicopace did his GSoC on this, check out this:
https://blog.freifunk.net/2017/08/29/gsoc-libremesh-spectrum-analyzer-summary/
@nicopace how is the current status? Is this included in current LibreMesh release?

Benik3, 2018-02-09:
E.g. folder spectrum-scan-decoder contains only ReadMe.md.
BTW I found a project Wlanslovenija, which also has luci-app-openairview included.
But it also doesn't work for me. I get error with lua script - attempt to index local 'net' (a nil value).
Also for some reason the website where should be spectrum the analyzer is "empty" (only plain texts).
I tried to take html from this repository but I got similar error with lua and nil value.
https://github.com/wlanslovenija/firmware-packages-opkg/tree/master/luci/luci-app-openairview

FreifunkUFO, 2018-02-11:
we tested it some years ago 
https://twitter.com/freifunk_l/status/628616896267591681

Benik3, 2018-02-12:
Thanks for reply.
I tried for sure to compile it again, but still I'm getting this error:
`/usr/lib/lua/luci/dispatcher.lua:460: Failed to execute template dispatcher target for entry '/admin/openairview/spectral_scan'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'openairview/spectral_scan'.
A runtime error occured: [string "/usr/lib/lua/luci/view/openairview/spectral..."]:19: attempt to index local 'wdev' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:460: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>`
BTW there is also missing package fft-eval needed for the openairview. I used the one from wlanslovenija as I didn't find it elsewhere...
Here I attach the compiled packages for target AR7XXX/AR9XXX generic with source Lede 17.01.4
[openairview.zip](https://github.com/libremesh/lime-packages/files/1717797/openairview.zip)

ilario, 2018-02-12:
Regarding fft-eval, the is also [this](https://github.com/libremesh/FFT_eval) repository from @altergui which has some additional commits but looks outdated.

Benik3, 2018-02-12:
OK, the under line is the reason why I didn't find it by googling :) (fft_eval vs fft-eval)

gmarcos87, 2019-02-20:
Ready in https://github.com/libremesh/lime-packages/pull/470

closed 2019-02-20 by gmarcos87

Benik3, 2019-02-23:
Great!
But I tried to install it on WR841 v9 on OpenWRT 18.06 (package from daily build and even my own build on OWRT sources) and it doesn't see any interfaces...
[image]

I wasn't able to test the luci-app-openairview because is has too much of dependencies and takes too much of space :/

gmarcos87, 2019-02-23:
Try using wlan0 as device
lime-openairview gets ifaces names from something like this **uci show wireless.@wifi-iface[0]=wifi-iface**

Benik3, 2019-02-23:
I tried, but it doesn't work at all :/
[image]

gmarcos87, 2019-02-24:
You need to compile openwrt whit this options:

> CONFIG_PACKAGE_ATH_DEBUG=y
> CONFIG_PACKAGE_ATH_SPECTRAL=y

Otherwise fft-eval doesn't work

Benik3, 2019-02-24:
Thanks, I didn't know that!
I compiled my own 18.06 and there is spectral scan working OK :)
So only thing which doesn't work for me is the mentioned get_stations. I tested other commands with success!