libremesh/lime-packages #325: Attended Sysupgrade inclusion

ilario, 2018-02-11:
[Attended Sysupgrade](https://blog.freifunk.net/2017/08/25/attended-sysupgrade-final-evaluation-update/) system helps the user to upgrade the router with one click. Developed by @aparcar 
What's missing:

* subdomain upgrade.libremesh.org for upgrade server (already accessible [here](https://ledeupdate.planetexpress.cc/))
* upgrade button in lime-app
* ?

aparcar, 2018-02-11:
@gmarcos87 can you help with the lime-app?

ilario, 2018-12-08:
@aparcar updates? Is this planned for inclusion in 18.06 release?

aparcar, 2019-01-14:
@ilario actually @gmarcos87 must do the magic, I'm not working on lime-app

ilario, 2019-10-04:
@aparcar @gmarcos87 what is missing?

ilario, 2019-10-04:
[luci-app-attendedsysupgrade](https://github.com/openwrt/luci/tree/openwrt-18.06/applications/luci-app-attendedsysupgrade) is already present in OpenWrt 18.06 and there's also stuff related to ubus that I don't understand in the [attendedsysupgrade-common](https://github.com/openwrt/packages/tree/openwrt-18.06/utils/attendedsysupgrade-common) package which is also included in OpenWrt 18.06.

ilario, 2023-03-03:
@spiccinini is there any planned automatic updating method?
In this still the one that is going to be used at some point in the future?